Koenkk/zigbee2mqtt #1915: Not starting on Synology Docker 18.09.0-0505

Kat-CeDe, 2019-08-31:
Hi,
on Monday I made an update of my Synology DS415+ and got the Docker version 18.09.0-0505 and after that Zigbee2MQTT did not start anymore. 

I always get

`8/31/2019, 9:00:10 AM - info: Logging to directory: '/app/data/log/2019-08-31.09-00-10'
8/31/2019, 9:00:11 AM - info: Starting zigbee2mqtt version 1.4.0 (commit #unknown)
8/31/2019, 9:00:11 AM - info: Starting zigbee-shepherd
8/31/2019, 9:00:23 AM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
8/31/2019, 9:01:23 AM - info: Starting zigbee-shepherd
8/31/2019, 9:01:35 AM - error: Error while starting zigbee-shepherd! (Error: request timeout)
`

I tried USB sticks with March and June firmware which already worked the last weeks. Today I got a 3rd new stick with same error. I pressed reset and did an update of the docker image today but always the same:-(

I do not think that all 3 sticks died at the same time so it must have something to do with new Docker on Synology. Change.log states that daemon is now 18.09.6 and compose 1.24.0.

Even though the log above states 1.4 I have installed docker 1.51 at least it says so.

Any ideas how to get it running again?

Ralf

f-bader, 2019-09-01:
Did maybe the device ID change? Check your docker config file and the available devices in your Synology 
```
 ls /dev/ttyACM*
```

Kat-CeDe, 2019-09-01:
Hi,
I cannot check in the console of the docker itself as it does not start. I use high priority on my Synology which means that all ports (complete /dev/) are passed through to the docker image and I see /dev/ACM0 in the console of the Synology so the port is there.

Ralf

hackenfu, 2019-09-02:
> Hi,
> I cannot check in the console of the docker itself as it does not start. I use high priority on my Synology which means that all ports (complete /dev/) are passed through to the docker image and I see /dev/ACM0 in the console of the Synology so the port is there.
> 
> What about the 1.4 instead of 1.51 in the log is this normal?
> 
> Edit: I am a big step further. I build Zigbee2MQTT from scratch on my Ryzen desktop. To my surprise the stick with old firmware started here. I put it in my Synology and now it starts there also:-) The other 2 sticks refused to work also on my PC so it is not a docker problem as I thought. I still is curious. Next weekend I will check with CC Debugger.
> 
> Thanks for making me digging deeper.
> 
> Ralf

it is synology's problem, back to the 17.05.0-0401 version.

Kat-CeDe, 2019-09-02:
Hi,
"it is synology's problem, back to the 17.05.0-0401 version."
do you have the same problems or did you read about problems somewhere?

I am not so sure anymore because I had the same error on my Antergos Desktop PC running Zigbee2MQTT without docker.

Ralf

hackenfu, 2019-09-02:
> Hi,
> "it is synology's problem, back to the 17.05.0-0401 version."
> do you have the same problems or did you read about problems somewhere?
> 
> I am not so sure anymore because I had the same error on my Antergos Desktop PC running Zigbee2MQTT without docker.
> 
> Ralf

the reason why i saw your post is, I google "18.09.0-0505" and there was your post in the search result list. i have synology 218+ and install docker to run ttrss and qbittorrent. in the past few days i found these two container didn't work. firstly i dount if there is something wrong with these containers, but then i found that  my 218+ automatically updated the docker to 19.09.0-0505, so i uninstalled it and reinstalled the previous 17.05.0-0401 version and everything works very well.
there are problems with this error posted somewhere: https://www.reddit.com/r/synology/comments/cw75ck/dockercompose_dont_pass_anymore_env_to_docker/

closed 2019-11-08 by stale[bot]