Koenkk/zigbee2mqtt #1918: Aqara: crash on pairing

dirstel, 2019-09-01:
# Bug Report
On trying to pair an Aqara Sensor, Zigbee2Mqtt crashes

> zigbee2mqtt:debug 9/1/2019, 4:59:51 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"LUMI","modelId":"lumi.sensor_magnet.aq2","powerSource":3}}' of device 'lumi.sensor_magnet.aq2' (0x00158d00########) of endpoint 1
>   zigbee2mqtt:info 9/1/2019, 4:59:51 PM New device 'lumi.sensor_magnet.aq2' with address 0x00158d00######## connected!
> /app/node_modules/js-yaml/lib/js-yaml/dumper.js:756
>       throw new YAMLException('unacceptable kind of an object to dump ' + type);
>       ^
> YAMLException: unacceptable kind of an object to dump [object Undefined]
>     at writeNode (/app/node_modules/js-yaml/lib/js-yaml/dumper.js:756:13)
>     at dump (/app/node_modules/js-yaml/lib/js-yaml/dumper.js:817:7)
>     at Object.safeDump (/app/node_modules/js-yaml/lib/js-yaml/dumper.js:823:10)
>     at Object.writeYaml (/app/lib/util/fs.js:13:33)
>     at write (/app/lib/util/settings.js:149:12)
>     at writeRead (/app/lib/util/settings.js:132:5)
>     at addDevice (/app/lib/util/settings.js:216:5)
>     at Object.addDevice (/app/lib/util/settings.js:400:30)
>     at DeviceReceive.onZigbeeMessage (/app/lib/extension/deviceReceive.js:68:22)
>     at extensions.filter.forEach (/app/lib/controller.js:143:31)
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! zigbee2mqtt@1.5.1 start: 'node index.js'
> npm ERR! Exit status 1
> npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.

## How to reproduce it (minimal and precise)
pair again ;-). Do not really know, why. My Zigbee Network is build from 51 Devices (1 Coordinator, 19 Router, Sensors). I did not manage to pair an Aqara Sensor since before migrating to 1.5.1. Most of the time I get the ""-Error, but sometimes z2m crashes like this.

## Debug Info
zigbee2mqtt version: Starting zigbee2mqtt version 1.5.1 (commit #4f7e3a1) latest dev
CC253X firmware version: Coordinator firmware version: '20190619'

Godhart, 2019-09-02:
Same here:

# Log
zigbee2mqtt:info 9/2/2019, 4:07:44 AM New device 'lumi.remote.b186acn01' with address 0xXXXXXXXXXXXXXXXX connected!
/zigbee2mqtt-1.5.1/node_modules/js-yaml/lib/js-yaml/dumper.js:756
      throw new YAMLException('unacceptable kind of an object to dump ' + type);
      ^
YAMLException: unacceptable kind of an object to dump [object Undefined]
    at writeNode (/zigbee2mqtt-1.5.1/node_modules/js-yaml/lib/js-yaml/dumper.js:756:13)
    at dump (/zigbee2mqtt-1.5.1/node_modules/js-yaml/lib/js-yaml/dumper.js:817:7)
    at Object.safeDump (/zigbee2mqtt-1.5.1/node_modules/js-yaml/lib/js-yaml/dumper.js:823:10)
    at Object.writeYaml (/zigbee2mqtt-1.5.1/lib/util/fs.js:13:33)
    at write (/zigbee2mqtt-1.5.1/lib/util/settings.js:139:12)
    at writeRead (/zigbee2mqtt-1.5.1/lib/util/settings.js:122:5)
    at addDevice (/zigbee2mqtt-1.5.1/lib/util/settings.js:206:5)
    at Object.addDevice (/zigbee2mqtt-1.5.1/lib/util/settings.js:390:30)
    at DeviceReceive.onZigbeeMessage (/zigbee2mqtt-1.5.1/lib/extension/deviceReceive.js:68:22)
    at extensions.filter.forEach (/zigbee2mqtt-1.5.1/lib/controller.js:143:31)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-09-02T01_07_46_775Z-debug.log

# Hardware:
* Raspberry PI v1 512Mb
* CC2531. Coordinator firmware version: '20190608'

# Software:
* Host system HassOS 2.12
* Hass.io supervisor version 184
* zigbee2mqtt version 1.5.1 (commit #unknown) (installed as add-on from hassio)

# P.S.
Along with this I'm experiencing problems with zigbee2mqtt connection to mqtt server but don't think it's the case

TLBM, 2019-09-02:
Hi!

I expirienced similar errors yesterday after I tried the option to put devices in a separate file like mentioned at the bottom of [https://www.zigbee2mqtt.io/configuration/configuration.html](https://www.zigbee2mqtt.io/configuration/configuration.html)

configuration.yaml
```
devices: devices.yaml
groups: groups.yaml
```

I couldn't pair an Aqara Cube and an Aqara Vibration Sensor any more. Most of the time the device address was added, but only as unknown device. It also crashed when I tried to remove them via `zigbee2mqtt/bridge/config/remove`.
I lost a lot of time because I thought it had something to do with the also before added Aqara LED Bulb and that they used it as router. But after I put the devices back in configuration.yaml it all worked again.

zigbee2mqtt version: version 1.5.1 (commit #ac3b924)
CC253x firmware version: Coordinator CC2530_CC2592_DEFAULT_20190608

dirstel, 2019-09-02:
> ```
> devices: devices.yaml
> groups: groups.yaml
> ```

Hmm, did that too, will try to push devices back to configuration.yaml (right now I'm not using groups).

dirstel, 2019-09-02:
When moving devices and (empty) groups section back to configuration.yaml pairing works fine, without crash. I even managed to avoid the "Cannot get node descriptor for..."-Error. So there may be some sort of hickup in the code which allows seperate yamls for devices and groups?!

Koenkk, 2019-09-03:
@dirstel yes probably, I will investigate it after #1888

james-fry, 2019-09-05:
> When moving devices and (empty) groups section back to configuration.yaml pairing works fine, without crash. I even managed to avoid the "Cannot get node descriptor for..."-Error. So there may be some sort of hickup in the code which allows seperate yamls for devices and groups?!

Thanks for this. 
I have been tearing my hair out with pairing on 1.5.1 (hassio).
Will try this ASAP.

Nihvel, 2019-09-06:
Same here, same log, sometimes same devices: Aqara Vibration sensor and Temperature sensor. The vibration doesn't work at all, errors and crashes. The temperature one let the network crash every some minutes and all the other devices (3 motion sensor and a door sensor) stop reporting the status.

Koenkk, 2019-09-13:
Can you share your `database.db` and `DEBUG=* npm start` log when pairing this motion sensor?

dirstel, 2019-09-22:
Sorry for not posting details... I had no time for checking. But I read another issue and retried - now pairing is working for the motion sensors.

closed 2019-09-22 by dirstel