Koenkk/zigbee2mqtt #1919: zigbee2mqtt crashing

iz3man, 2019-09-01:
Can't really explain when and why this happened, but now I get this in an endless loop:

```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.16.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.5.1~prestart: zigbee2mqtt@1.5.1
6 info lifecycle zigbee2mqtt@1.5.1~start: zigbee2mqtt@1.5.1
7 verbose lifecycle zigbee2mqtt@1.5.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.5.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.5.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.5.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.5.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.5.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.5.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.5.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.66-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.3
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.5.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.5.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

I don't know if this is caused by the CC2531 stick disconnecting, or if the above issue causes the disconnect

```
[  633.212329] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[  693.435641] usb 1-1.4: USB disconnect, device number 17
[  693.732021] usb 1-1.4: new full-speed USB device number 18 using dwc_otg
[  693.879764] usb 1-1.4: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[  693.879774] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  693.879779] usb 1-1.4: Product: TI CC2531 USB CDC
[  693.879784] usb 1-1.4: Manufacturer: Texas Instruments
[  693.879789] usb 1-1.4: SerialNumber: __0X00124B001938A6DE
[  693.882095] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[  754.107740] usb 1-1.4: USB disconnect, device number 18
[  754.412012] usb 1-1.4: new full-speed USB device number 19 using dwc_otg
[  754.561279] usb 1-1.4: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[  754.561290] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  754.561295] usb 1-1.4: Product: TI CC2531 USB CDC
[  754.561299] usb 1-1.4: Manufacturer: Texas Instruments
[  754.561304] usb 1-1.4: SerialNumber: __0X00124B001938A6DE
[  754.563515] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```

This is the syslog:

```
Sep  1 21:03:47 hassbian systemd[1]: Stopped Landroid Bridge.
Sep  1 21:03:47 hassbian systemd[1]: Started Landroid Bridge.
Sep  1 21:03:48 hassbian npm[6924]: > zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
Sep  1 21:03:48 hassbian npm[6924]: > node index.js
Sep  1 21:03:48 hassbian landroid-bridge[6932]: internal/modules/cjs/loader.js:638
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     throw err;
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     ^
Sep  1 21:03:48 hassbian landroid-bridge[6932]: Error: Cannot find module '/home/homeassistant/landroid-bridge/node_modules/sqlite3/lib/binding/node-v64-linux-arm/node_sqlite3.node'
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     at Function.Module._load (internal/modules/cjs/loader.js:562:25)
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     at Module.require (internal/modules/cjs/loader.js:692:17)
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     at require (internal/modules/cjs/helpers.js:25:18)
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     at Object.<anonymous> (/home/homeassistant/landroid-bridge/node_modules/sqlite3/lib/sqlite3.js:4:15)
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     at Module._compile (internal/modules/cjs/loader.js:778:30)
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     at Module.load (internal/modules/cjs/loader.js:653:32)
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
Sep  1 21:03:48 hassbian landroid-bridge[6932]:     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
Sep  1 21:03:48 hassbian systemd[1]: landroid-bridge.service: Main process exited, code=exited, status=1/FAILURE
Sep  1 21:03:48 hassbian systemd[1]: landroid-bridge.service: Unit entered failed state.
Sep  1 21:03:48 hassbian systemd[1]: landroid-bridge.service: Failed with result 'exit-code'.
Sep  1 21:03:49 hassbian npm[6924]: /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:121
Sep  1 21:03:49 hassbian npm[6924]:         throw e;
Sep  1 21:03:49 hassbian npm[6924]:         ^
Sep  1 21:03:49 hassbian npm[6924]: Error: The module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node'
Sep  1 21:03:49 hassbian npm[6924]: was compiled against a different Node.js version using
Sep  1 21:03:49 hassbian npm[6924]: NODE_MODULE_VERSION 67. This version of Node.js requires
Sep  1 21:03:49 hassbian npm[6924]: NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
Sep  1 21:03:49 hassbian npm[6924]: the module (for instance, using `npm rebuild` or `npm install`).
Sep  1 21:03:49 hassbian npm[6924]:     at Object.Module._extensions..node (internal/modules/cjs/loader.js:807:18)
Sep  1 21:03:49 hassbian npm[6924]:     at Module.load (internal/modules/cjs/loader.js:653:32)
Sep  1 21:03:49 hassbian npm[6924]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
Sep  1 21:03:49 hassbian npm[6924]:     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
Sep  1 21:03:49 hassbian npm[6924]:     at Module.require (internal/modules/cjs/loader.js:692:17)
Sep  1 21:03:49 hassbian npm[6924]:     at require (internal/modules/cjs/helpers.js:25:18)
Sep  1 21:03:49 hassbian npm[6924]:     at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:112:48)
Sep  1 21:03:49 hassbian npm[6924]:     at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:1:36)
Sep  1 21:03:49 hassbian npm[6924]:     at Module._compile (internal/modules/cjs/loader.js:778:30)
Sep  1 21:03:49 hassbian npm[6924]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
Sep  1 21:03:49 hassbian npm[6924]: npm ERR! code ELIFECYCLE
Sep  1 21:03:49 hassbian npm[6924]: npm ERR! errno 1
Sep  1 21:03:49 hassbian npm[6924]: npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
Sep  1 21:03:49 hassbian npm[6924]: npm ERR! Exit status 1
Sep  1 21:03:49 hassbian npm[6924]: npm ERR!
Sep  1 21:03:49 hassbian npm[6924]: npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
Sep  1 21:03:49 hassbian npm[6924]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Sep  1 21:03:49 hassbian npm[6924]: npm ERR! A complete log of this run can be found in:
Sep  1 21:03:49 hassbian npm[6924]: npm ERR!     /home/pi/.npm/_logs/2019-09-01T19_03_49_873Z-debug.log
Sep  1 21:03:49 hassbian systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Sep  1 21:03:49 hassbian systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Sep  1 21:03:49 hassbian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Sep  1 21:03:50 hassbian systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Sep  1 21:03:50 hassbian systemd[1]: Stopped zigbee2mqtt.
```

I know this shows an error about landroid-bridge not finding a file, but I don't know if this is connected to the problem I see here.

All this is running on an update system, running latest version of Hassbian.

iz3man, 2019-09-02:
#1892

Koenkk, 2019-09-03:
What is landroid bridge? Did you follow https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html?

iz3man, 2019-09-03:
Landroid Bridge is an MQTT bridge for my robot mower. As is up/downgraded node because zigbee2mqtt complained about node v11 I also had to rebuild modules for landroid bridge. That's probably we it showed those errors. This has been solved.
I'm now quite sure that my problem is version 1.5.1 as several other posts here show.
Zigbee2mqtt was running fine for the last few month, but there problems occured last week after the update.

Koenkk, 2019-09-03:
It seems they are conflicting with each other (see serialport warning in log).

If you execute in the `/opt/git/zigbee2mqtt` `rm -rf node_modules && npm install`, does it work again?

iz3man, 2019-09-03:
No it doesn't work. And there's no way they could interfere in principal. The landroid app doesn't access the USB port in any way. It's wifi only. 
The serialport issue seem to be caused by cc2531 reconnecting.

Koenkk, 2019-09-03:
So the first issue has been solved, for the other one lets continue in #1892

closed 2019-09-03 by Koenkk