Koenkk/zigbee2mqtt #1923: Osram Smart+ Motion Sensor support

mabi, 2019-09-02:
# Bug Report
## What happened
With a newly installed Osram Smart+ Motion Sensor, I just get one temperature message and then nothing more. Log shows failed configuration attempt.
```
8/29/2019, 7:10:52 AM - info: Successfully reenabled joining
8/29/2019, 7:12:05 AM - info: New device 'Motion Sensor-A' with address 0x00redacted000000 connected!
8/29/2019, 7:12:06 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00redacted000000","meta":{"modelID":"Motion Sensor-A"}}'
8/29/2019, 7:12:10 AM - warn: Failed to configure 0x00redacted000000 (0x00redacted000000) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
8/29/2019, 7:12:10 AM - warn: This can be ignored if the device is working properly
8/29/2019, 7:12:19 AM - warn: Failed to configure 0x00redacted000000 (0x00redacted000000) ('Error: request unsuccess: 112') (attempt #2)
8/29/2019, 7:12:19 AM - warn: This can be ignored if the device is working properly
8/29/2019, 7:13:05 AM - warn: Failed to configure 0x00redacted000000 (0x00redacted000000) ('Error: request unsuccess: 112') (attempt #3)
8/29/2019, 7:13:05 AM - warn: This can be ignored if the device is working properly
8/29/2019, 7:13:07 AM - info: MQTT publish: topic 'zigbee2mqtt/0x00redacted000000', payload '{"temperature":24.68,"linkquality":42}'
8/29/2019, 7:13:32 AM - info: Successfully reenabled joining

```

## What did you expect to happen
The sensor should send occupation and temperature messages without errors in log.
I read in #507 that other people had similar issues and pairing is difficult. Is there some trick to pair correctly? Long pressing the button seems to do nothing.

## How to reproduce it (minimal and precise)

Switch on Motion sensor close to the coordinator and check logs.

## Debug Info
zigbee2mqtt version:  1.5.1 (commit #ac3b924)
CC253X firmware version: '20190315' max_stability

dreimer1986, 2019-09-10:
Small deja vu... It's a PAIN to add this sensor to z2m right now. I have one too and sometimes need 10 tries to get it pairing properly. I just updated to new port v1.6 edge version and will try my luck with zigbee-herdsman later today. Maybe not only Xiaomi devices now work better regarding pairing. If you wanna try right now. Stop the addin, edit the db file in /share/zigbee2mqtt removing all sections of the sensor and put it in pairing mode, then start the addin again. Redo UNTIL it finally works. But better wait for me or try yourself with the most recent edge versions.

dreimer1986, 2019-09-10:
OK, try new Edge versions. I connected to the (right now unused sensor due to the problems with repairing and replaced by Hue one) Smart+ sensor FIRST TRY. It ... just WORKS 💃

mabi, 2019-09-11:
Thank you @dreimer1986 I will try the dev version this weekend :)

mabi, 2019-09-20:
Confirmed working at first try with version 1.6.0 d4f01e6

closed 2019-09-20 by mabi