Koenkk/zigbee2mqtt #1925: Please help to add Yale YRD210-HA-605.

alexsahka, 2019-09-03:
https://github.com/Koenkk/zigbee2mqtt/issues/1871

Any links to guide how to properly add this or similar, not supported locks?

This guide: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
not explaining how to work with locks...

I still experiencing problems, 
Changing/adjusting the devices.js file, only giving me two options:
a lot of messages and partially working lock,
or nothing.

alexsahka, 2019-09-28:
Switched to zigbee2mqtt 1.6-dev and now device stopped to reporting but controllable with lock/unlock command. 
Getting error: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000dxxxxxx' failed: 'Error: Timeout - 62507 - 1 - 4 - 11 after 15000ms'","meta":{"friendly_name":"0x000dxxxxxx"}}'

Koenkk, 2019-09-28:
- You can probably re-use and existing entry (e.g. https://github.com/Koenkk/zigbee-shepherd-converters/blob/e24dca879c22a5cd9351161da75f352daa549a79/devices.js#L3986), just update the `zigbeeModel` (`modelId` in `database.db`), `model` (only used for documentation purposes) accordingly.
- If you keep getting the 15000ms` timeout error, add the following to the definition https://github.com/Koenkk/zigbee-shepherd-converters/blob/e24dca879c22a5cd9351161da75f352daa549a79/devices.js#L4118 (`disableDefaultResponse`)

alexsahka, 2019-09-28:
Thank you for the response.

Using now with the same problems (lock not reporting, and still Error: Timeout)
 I tried to reuse before all YALE Locks existing entries without any good result. Here is the photo of the lock Zigbee Module. Maybe this is a universal module someone used somewhere?

[image]

{
        zigbeeModel: ['YRD210 PB DB'],
        model: 'YRD210-HA-605',
        vendor: 'Yale',
        description: 'Assure lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.generic_lock, fz.battery_200],
        toZigbee: [tz.generic_lock],
        meta: {options: {disableDefaultResponse: true}, configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },

Koenkk, 2019-09-29:
Do you see a successful or failed to configure log line when starting zigbee2mqtt?

alexsahka, 2019-09-29:
Here is the zigbee2mqtt log:

Using '/app/data' as data directory

> zigbee2mqtt@1.6.0 start /app

> node index.js

zigbee2mqtt:info  2019-09-29T21:48:23: Logging to directory: '/app/data/log/2019-09-29.14-48-23'

zigbee2mqtt:debug 2019-09-29T21:48:23: Loaded state from file /app/data/state.json

zigbee2mqtt:info  2019-09-29T21:48:23: Starting zigbee2mqtt version 1.6.0 (commit #a7e9ad7)

zigbee2mqtt:info  2019-09-29T21:48:23: Starting zigbee-herdsman...

zigbee2mqtt:debug 2019-09-29T21:48:23: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'

zigbee2mqtt:info  2019-09-29T21:48:25: zigbee-herdsman started

zigbee2mqtt:info  2019-09-29T21:48:25: Coordinator firmware version: '{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}}'

zigbee2mqtt:debug 2019-09-29T21:48:25: Zigbee network parameters: {"panID":6754,"extendedPanID":"0x00xxxxxxxxxxxxx","channel":11}

zigbee2mqtt:info  2019-09-29T21:48:25: Currently 1 devices are joined:

zigbee2mqtt:info  2019-09-29T21:48:25: 0x000xxxxxxxxxxx (0x000xxxxxxxxxxx): YRD210HA605 - Yale Assure lock (EndDevice)

zigbee2mqtt:warn  2019-09-29T21:48:25: `permit_join` set to  `true` in configuration.yaml.

zigbee2mqtt:warn  2019-09-29T21:48:25: Allowing new devices to join.

zigbee2mqtt:warn  2019-09-29T21:48:25: Set `permit_join` to `false` once you joined all devices.

zigbee2mqtt:info  2019-09-29T21:48:25: Zigbee: allowing new devices to join.

zigbee2mqtt:info  2019-09-29T21:48:25: Connecting to MQTT server at mqtt://10.x.x.xxx

zigbee2mqtt:info  2019-09-29T21:48:25: Connected to MQTT server

zigbee2mqtt:info  2019-09-29T21:48:25: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

zigbee2mqtt:info  2019-09-29T21:48:25: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"a7e9ad7","coordinator":{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}},"log_level":"debug","permit_join":false}'

zigbee2mqtt:debug 2019-09-29T21:48:25: Received MQTT message on 'zigbee2mqtt/0x000xxxxxxxxxxx/set' with data '{"state":"LOCK"}'

zigbee2mqtt:debug 2019-09-29T21:48:25: Publishing 'set' 'state' to '0x000xxxxxxxxxxx'

zigbee2mqtt:debug 2019-09-29T21:48:30: Received MQTT message on 'zigbee2mqtt/bridge/config/devices/get' with data ''

zigbee2mqtt:info  2019-09-29T21:48:30: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00xxxxxxxxxxxxx","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack30x","dateCode":"20190425","lastSeen":1569793710199},{"ieeeAddr":"0x000xxxxxxxxxxx","type":"EndDevice","networkAddress":14993,"model":"YRD210HA605","friendly_name":"0x000xxxxxxxxxxx","manufacturerID":43690,"manufacturerName":"Yale","powerSource":"Battery","modelID":"YRD210 PB DB","hardwareVersion":17,"lastSeen":1569793707580}]'

zigbee2mqtt:debug 2019-09-29T21:48:30: Received MQTT message on 'zigbee2mqtt/bridge/networkmap' with data 'raw'

zigbee2mqtt:info  2019-09-29T21:48:30: Starting network scan (includeRoutes 'false')

zigbee2mqtt:debug 2019-09-29T21:48:30: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '[{"ieeeAddr":"0x00xxxxxxxxxxxxx","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack30x","dateCode":"20190425","lastSeen":1569793710199},{"ieeeAddr":"0x000xxxxxxxxxxx","type":"EndDevice","networkAddress":14993,"model":"YRD210HA605","friendly_name":"0x000xxxxxxxxxxx","manufacturerID":43690,"manufacturerName":"Yale","powerSource":"Battery","modelID":"YRD210 PB DB","hardwareVersion":17,"lastSeen":1569793707580}]'

zigbee2mqtt:info  2019-09-29T21:48:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00xxxxxxxxxxxxx","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack30x","dateCode":"20190425","lastSeen":1569793710205},{"ieeeAddr":"0x000xxxxxxxxxxx","type":"EndDevice","networkAddress":14993,"model":"YRD210HA605","friendly_name":"0x000xxxxxxxxxxx","manufacturerID":43690,"manufacturerName":"Yale","powerSource":"Battery","modelID":"YRD210 PB DB","hardwareVersion":17,"lastSeen":1569793707580}]}'

zigbee2mqtt:debug 2019-09-29T21:48:30: LQI succeeded for 'Coordinator'

zigbee2mqtt:info  2019-09-29T21:48:30: Network scan finished

zigbee2mqtt:info  2019-09-29T21:48:30: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/raw', payload '{"nodes":[{"ieeeAddr":"0x00xxxxxxxxxxxxx","friendlyName":"Coordinator","type":"Coordinator","networkAddress":0,"failed":[],"lastSeen":null},{"ieeeAddr":"0x000xxxxxxxxxxx","friendlyName":"0x000xxxxxxxxxxx","type":"EndDevice","networkAddress":14993,"manufacturerName":"Yale","modelID":"YRD210 PB DB","lastSeen":1569793707580}],"links":[{"source":{"ieeeAddr":"0x000xxxxxxxxxxx","networkAddress":14993},"target":{"ieeeAddr":"0x00xxxxxxxxxxxxx","networkAddress":0},"linkquality":45,"depth":1,"routes":[],"sourceIeeeAddr":"0x000xxxxxxxxxxx","targetIeeeAddr":"0x00xxxxxxxxxxxxx","sourceNwkAddr":14993,"lqi":45,"relationship":1}]}'

zigbee2mqtt:error 2019-09-29T21:48:40: Publish 'set' 'state' to '0x000xxxxxxxxxxx' failed: 'Error: Timeout - 14993 - 1 - 2 - 11 after 15000ms'

zigbee2mqtt:info  2019-09-29T21:48:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000xxxxxxxxxxx' failed: 'Error: Timeout - 14993 - 1 - 2 - 11 after 15000ms'","meta":{"friendly_name":"0x000xxxxxxxxxxx"}}'

Koenkk, 2019-09-30:
I did some fixes, can you try again with the latest dev branch?

alexsahka, 2019-10-01:
Yes, the Error: Timeout - is gone, Thank you,
but the lock is still not reporting anything back ... zero.

Koenkk, 2019-10-02:
- Can you share your `database.db`?
- Can you provide the debug log when locking/unlocking the lock? To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

alexsahka, 2019-10-02:
Here is the database.db

{"id":1,"type":"Coordinator","ieeeAddr":"0x000xxxxxxxxxxx","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{}}},"interviewCompleted":false,"meta":{}}
{"id":2,"type":"EndDevice","ieeeAddr":"0x000xxxxxxxxxxx","nwkAddr":3775,"manufId":43690,"manufName":"Yale","powerSource":"Battery","modelId":"YRD210 PB DB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,9,10,257,32],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"YRD210 PB DB","manufacturerName":"Yale","powerSource":3,"zclVersion":1,"appVersion":132,"hwVersion":17}}}}},"appVersion":132,"hwVersion":17,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":1}}
{"id":3,"type":"EndDevice","ieeeAddr":"0x000xxxxxxxxxxx","nwkAddr":43516,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI wireless dimmer","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2064,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI wireless dimmer","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":30,"dateCode":"20160923","swBuildId":"1.1.1.1-5.7.2.0"}}}}},"appVersion":17,"stackVersion":87,"hwVersion":30,"dateCode":"20160923","swBuildId":"1.1.1.1-5.7.2.0","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1}}

alexsahka, 2019-10-02:
Here is the debug log:

zigbee2mqtt:debug 2019-10-02T15:23:43: Received MQTT message on 'zigbee2mqtt/bridge/config/log_level' with data 'Debug'

zigbee2mqtt:info  2019-10-02T15:23:43: Switching log level to 'debug'

zigbee2mqtt:info  2019-10-02T15:23:43: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"90c70e8","coordinator":{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}},"log_level":"debug","permit_join":true}'

zigbee2mqtt:debug 2019-10-02T15:24:05: Received MQTT message on 'zigbee2mqtt/main_door_lock/set' with data '{"state":"UNLOCK"}'

zigbee2mqtt:debug 2019-10-02T15:24:05: Publishing 'set' 'state' to 'main_door_lock'

zigbee2mqtt:error 2019-10-02T15:24:11: Publish 'set' 'state' to 'main_door_lock' failed: 'Error: SRSP - AF - dataRequest after 6000ms'

zigbee2mqtt:info  2019-10-02T15:24:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'main_door_lock' failed: 'Error: SRSP - AF - dataRequest after 6000ms'","meta":{"friendly_name":"main_door_lock"}}'

zigbee2mqtt:debug 2019-10-02T15:25:01: Received MQTT message on 'zigbee2mqtt/main_door_lock/set' with data '{"state":"LOCK"}'

zigbee2mqtt:debug 2019-10-02T15:25:01: Publishing 'set' 'state' to 'main_door_lock'

zigbee2mqtt:info  2019-10-02T15:25:02: MQTT publish: topic 'zigbee2mqtt/main_door_lock', payload '{"state":"LOCK"}'

zigbee2mqtt:debug 2019-10-02T15:25:11: Saving state to file /app/data/state.json

zigbee2mqtt:debug 2019-10-02T15:25:17: Received MQTT message on 'zigbee2mqtt/bridge/config/devices/get' with data ''

zigbee2mqtt:info  2019-10-02T15:25:17: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x000xxxxxxxxxxx","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack30x","dateCode":"20190425","lastSeen":1570029917506},{"ieeeAddr":"0x000xxxxxxxxxxx","type":"EndDevice","networkAddress":3775,"model":"YRD210-HA-605","friendly_name":"main_door_lock","manufacturerID":43690,"manufacturerName":"Yale","powerSource":"Battery","modelID":"YRD210 PB DB","hardwareVersion":17,"lastSeen":1570029902906},{"ieeeAddr":"0x000xxxxxxxxxxx","type":"EndDevice","networkAddress":43516,"model":"ICTC-G-1","friendly_name":"ikea_dimmer_01","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI wireless dimmer","hardwareVersion":30,"softwareBuildID":"1.1.1.1-5.7.2.0","dateCode":"20160923","lastSeen":1569998298580}]'

zigbee2mqtt:debug 2019-10-02T15:25:17: Received MQTT message on 'zigbee2mqtt/bridge/networkmap' with data 'raw'

zigbee2mqtt:info  2019-10-02T15:25:17: Starting network scan (includeRoutes 'false')

zigbee2mqtt:debug 2019-10-02T15:25:17: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '[{"ieeeAddr":"0x000xxxxxxxxxxx","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack30x","dateCode":"20190425","lastSeen":1570029917506},{"ieeeAddr":"0x000xxxxxxxxxxx","type":"EndDevice","networkAddress":3775,"model":"YRD210-HA-605","friendly_name":"main_door_lock","manufacturerID":43690,"manufacturerName":"Yale","powerSource":"Battery","modelID":"YRD210 PB DB","hardwareVersion":17,"lastSeen":1570029902906},{"ieeeAddr":"0x000xxxxxxxxxxx","type":"EndDevice","networkAddress":43516,"model":"ICTC-G-1","friendly_name":"ikea_dimmer_01","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI wireless dimmer","hardwareVersion":30,"softwareBuildID":"1.1.1.1-5.7.2.0","dateCode":"20160923","lastSeen":1569998298580}]'

zigbee2mqtt:info  2019-10-02T15:25:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x000xxxxxxxxxxx","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack30x","dateCode":"20190425","lastSeen":1570029917507},{"ieeeAddr":"0x000xxxxxxxxxxx","type":"EndDevice","networkAddress":3775,"model":"YRD210-HA-605","friendly_name":"main_door_lock","manufacturerID":43690,"manufacturerName":"Yale","powerSource":"Battery","modelID":"YRD210 PB DB","hardwareVersion":17,"lastSeen":1570029902906},{"ieeeAddr":"0x000xxxxxxxxxxx","type":"EndDevice","networkAddress":43516,"model":"ICTC-G-1","friendly_name":"ikea_dimmer_01","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI wireless dimmer","hardwareVersion":30,"softwareBuildID":"1.1.1.1-5.7.2.0","dateCode":"20160923","lastSeen":1569998298580}]}'

zigbee2mqtt:debug 2019-10-02T15:25:17: LQI succeeded for 'Coordinator'

zigbee2mqtt:info  2019-10-02T15:25:17: Network scan finished

zigbee2mqtt:info  2019-10-02T15:25:17: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/raw', payload '{"nodes":[{"ieeeAddr":"0x000xxxxxxxxxxx","friendlyName":"Coordinator","type":"Coordinator","networkAddress":0,"failed":[],"lastSeen":null},{"ieeeAddr":"0x000xxxxxxxxxxx","friendlyName":"main_door_lock","type":"EndDevice","networkAddress":3775,"manufacturerName":"Yale","modelID":"YRD210 PB DB","lastSeen":1570029902906},{"ieeeAddr":"0x000xxxxxxxxxxx","friendlyName":"ikea_dimmer_01","type":"EndDevice","networkAddress":43516,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI wireless dimmer","lastSeen":1569998298580}],"links":[{"source":{"ieeeAddr":"0x000xxxxxxxxxxx","networkAddress":3775},"target":{"ieeeAddr":"0x000xxxxxxxxxxx","networkAddress":0},"linkquality":51,"depth":1,"routes":[],"sourceIeeeAddr":"0x000xxxxxxxxxxx","targetIeeeAddr":"0x000xxxxxxxxxxx","sourceNwkAddr":3775,"lqi":51,"relationship":1},{"source":{"ieeeAddr":"0x000xxxxxxxxxxx","networkAddress":43516},"target":{"ieeeAddr":"0x000xxxxxxxxxxx","networkAddress":0},"linkquality":52,"depth":1,"routes":[],"sourceIeeeAddr":"0x000xxxxxxxxxxx","targetIeeeAddr":"0x000xxxxxxxxxxx","sourceNwkAddr":43516,"lqi":52,"relationship":1}]}'

Koenkk, 2019-10-03:
Can you try with:

```js

{
    zigbeeModel: ['YRD210 PB DB'],
    model: 'YRD210-HA-605',
    vendor: 'Yale',
    description: 'Assure lock',
    supports: 'lock/unlock, battery',
    fromZigbee: [fz.generic_lock, fz.battery_200],
    toZigbee: [tz.generic_lock],
    meta: {options: {disableDefaultResponse: true}, configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock]);
        await configureReporting.lockState(endpoint);
        await configureReporting.batteryPercentageRemaining(endpoint);
    },
},
```

If it doesn't work please try to re-pair the device, currently you see `"meta":{"configured":1}` for this device in `database.db`, this should be `"meta":{"configured":2}` if everything went ok.

alexsahka, 2019-10-04:
It doesn't work, the lock does not report anything back.
After re-pair, the .db file now contains this:   ,"interviewCompleted":false,"meta":{}}
is this normal? I tried to re-pair multiple times and ended up with the same result: ,"interviewCompleted":false,"meta":{}}

Koenkk, 2019-10-04:
No that's not normal, `interviewCompleted` should be `true`; can you post the log when running with `DEBUG=zigbee-herdsman* npm start` and pairing the device? (please don't post it here but on pastebin.com)

alexsahka, 2019-10-04:
How to start with docker DEBUG=zigbee-herdsman* npm start?

Koenkk, 2019-10-04:
How do you run zigbee2mqtt now? Docker or bare metal?

alexsahka, 2019-10-04:
Docker

Koenkk, 2019-10-04:
Run docker container with `-e DEBUG=zigbee-herdsman*`

alexsahka, 2019-10-05:
Thank you, the lock is working now, reporting everything except battery level.
I don't know why it starts working, so here are my steps:

1. Start from clean (latest) new docker container (dev branch) with -e DEBUG=zigbee-herdsman*
2. Paired lock with interviewCompleted":true,"meta":{"configured":2}
3. Lock start reporting state and linkquality
4. Added fz.generic_lock_operation_event
5. Lock start correctly reporting "state" "linkquality" "user" and "source"
6.  Started container without  -e DEBUG=zigbee-herdsman*
7. Everything working, lock still reporting "state" "linkquality" "user" and "source"

So the only thing is missing, the battery status for some reason.
Here is the device.js:
{
        zigbeeModel: ['YRD210 PB DB'],
        model: 'YRD210-HA-605',
        vendor: 'Yale',
        description: 'Assure lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.generic_lock, fz.generic_lock_operation_event, fz.battery_200],
        toZigbee: [tz.generic_lock],
        meta: {options: {disableDefaultResponse: true}, configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
			await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },

Here is the database.db:
{"id":2,"type":"EndDevice","ieeeAddr":"0x000xxxxxxxxxxx","nwkAddr":32574,"manufId":43690,"manufName":"Yale","powerSource":"Battery","modelId":"YRD210 PB DB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,9,10,257,32],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"YRD210 PB DB","manufacturerName":"Yale","powerSource":3,"zclVersion":1,"appVersion":132,"hwVersion":17}},"closuresDoorLock":{"attributes":{"lockState":2}}},"binds":[{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x000xxxxxxxxxxx","endpointID":1}]}},"appVersion":132,"hwVersion":17,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":2}}

Koenkk, 2019-10-05:
The battery can take up-to 24 hours before reporting it. Can you check again after that?

alexsahka, 2019-10-05:
Yes sure, thank you for your work.

alexsahka, 2019-10-06:
More than 48 hours and no status on the battery, everything else except this working good

Koenkk, 2019-10-07:
Can you try with:

```js
{
        zigbeeModel: ['YRD210 PB DB'],
        model: 'YRD210-HA-605',
        vendor: 'Yale',
        description: 'Assure lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.generic_lock, fz.battery_200],
        toZigbee: [tz.generic_lock],
        meta: {options: {disableDefaultResponse: true}, configureKey: 5},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);

            const payload = [{
                attribute: 'batteryPercentageRemaining',
                minimumReportInterval: 0,
                maximumReportInterval: 3,
                reportableChange: 0,
            }];
            await endpoint.configureReporting('genPowerCfg', payload);
        },
    },
```

It should report it's battery level every 3 seconds now. If not, make sure that is has been configured; you should see in `database.db` `"meta":{"configured":5}}`. If not try to re-pair.

alexsahka, 2019-10-07:
Yes, it has happened exctly as you described,
database.db  changed to "meta":{"configured":5}}.
and lock reports it's battery level x3 times per second

The question now is, how to stop lock to reporting 3 times per second (at least)?

Koenkk, 2019-10-07:
Can you try with

```js
{
        zigbeeModel: ['YRD210 PB DB'],
        model: 'YRD210-HA-605',
        vendor: 'Yale',
        description: 'Assure lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.generic_lock, fz.battery_200],
        toZigbee: [tz.generic_lock],
        meta: {options: {disableDefaultResponse: true}, configureKey: 6},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },
```

- Configured should be 6 in `database.db`
- It should report it's battery level max every 24 hours. (not every 3 seconds anymore)

alexsahka, 2019-10-07:
OK, Thank you for your help, I will watch for battery reporting.

alexsahka, 2019-10-07:
OK, it was quiet for 10 + minutes and then lock start sending messages 2+- per second again.
the "interviewCompleted":true,"meta":{"configured":5}} not changed to "interviewCompleted":true,"meta":{"configured":6}}
Do I need to chang it mannually?

Koenkk, 2019-10-08:
No that means configuring failed, (you should see this log line when starting zigbee2mqtt). Please try to re-pair the device.

alexsahka, 2019-10-08:
Can you explain how to re-pair the device properly?
Delete from configuration.yaml only?
 Or from every file (configuration.yaml, database.db, state.json, devices.js )

Thank you.

Koenkk, 2019-10-09:
Only from `database.db`

alexsahka, 2019-10-09:
OK thanks, I re-pair it (But I deleted lock from configuration.yaml, database.db and state.json, )
started from a clean container with:
	{
        zigbeeModel: ['YRD210 PB DB'],
        model: 'YRD210-HA-605',
        vendor: 'Yale',
        description: 'Assure lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.generic_lock, fz.generic_lock_operation_event, fz.battery_200],
        toZigbee: [tz.generic_lock],
        meta: {options: {disableDefaultResponse: true}, configureKey: 6},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },

The database.db changed to "meta":{"configured":6}
Lock is reporting everythig but battery is still missing, will see in 24 hours.

And I added  fz.generic_lock_operation_event,  without it lock is only reporting: "state" and "linkquality".

With fz.generic_lock_operation_event, I got: "state","linkquality","user" and "source"

alexsahka, 2019-10-09:
OK thank you, 
after 10-15 hours the lock is reporting battery status as well.
So, I don't know maybe this Yale YRD210 PB DB needs to be added permanently to devices.js?
This is your call.

closed 2019-10-09 by alexsahka

Koenkk, 2019-10-10:
Yes, could you make a pr or the current definition you use in `devices.js`?

farhanito, 2023-03-14:
Hi, 

So sorry for bringing up an old thread. I'm new to zigbee2mqtt. Using v1.30.2 with docker.

I have the same Yale lock as @alexsahka with the exact same zigbee module. I managed to get it paired, but the lock is only reporting "state", "battery", and "linkquality". 
What do i have to do to make it reports "user", "source", and "action" as well?

Thank you