Koenkk/zigbee2mqtt #1926: Issue starting with latest-dev build

aidbish, 2019-09-03:
## What happened
Unable to start latest-dev build

## What did you expect to happen
would start normally 
## How to reproduce it (minimal and precise)
pulled latest-dev docker build

## Debug Info
zigbee2mqtt version: latest-dev - version 1.5.1 (commit #b00d4d5)
CC253X firmware version: "coordinator": 20190523, 3.x

trying to test out the latest-dev build gives error upon startring and just loops through trying to start, reload the latest build and no issue starting

logfile: https://hastebin.com/raw/cemuditimi

Koenkk, 2019-09-03:
Clearly a bug, can you provide the `data/database.db` file?

aidbish, 2019-09-17:
Rebuild Z2M and issue no longer appears. Probably a issue on my installation

closed 2019-09-17 by aidbish