Koenkk/zigbee2mqtt #1928: error on mqtt password

pcfreak1, 2019-09-03:
hi, i have an issue, i get an error on my mqtt password. it would be a big hassle having to change it so how can i fix this?

the error log: 

pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:173
  throw generateError(state, message);
  ^
YAMLException: bad indentation of a mapping entry at line 15, column 13:
      password: ,W(/@6JJ>RCe5CRX
                ^
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:167:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:173:9)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1107:7)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1359:12)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1519:3)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1575:5)
    at load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1596:19)
    at Object.safeLoad (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1618:10)
    at Object.readYaml (/opt/zigbee2mqtt/lib/util/fs.js:5:17)
    at read (/opt/zigbee2mqtt/lib/util/settings.js:147:18)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-03T17_51_47_264Z-debug.log

Koenkk, 2019-09-04:
Your `configuration.yaml` is invalid, please use e.g. http://www.yamllint.com/ to find the problems and fix it.

pcfreak1, 2019-09-04:
i fixed my yaml but still getting this:

pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'zigbee-herdsman/dist/zcl-id'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/converters/toZigbee.js:5:15)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-04T16_48_26_726Z-debug.log
pi@raspberrypi:/opt/zigbee2mqtt $ nano /home/pi/.npm/_logs/2019-09-04T16_48_26_726Z-debug.log
     File: /home/pi/.npm/_logs/2019-09-04T16_48_26_726Z-debug.log

13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_pro$
14 verbose pkgid zigbee2mqtt@1.5.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.66-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.3
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.5.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.5.1 start script.
23 error This is probably not a problem with npm. There is likely additional lo$
24 verbose exit [ 1, true ]

Koenkk, 2019-09-04:
That's another common issue, please read https://github.com/Koenkk/zigbee2mqtt/issues/1892#issuecomment-527837064

As this issue is fixed  I will close it.

closed 2019-09-04 by Koenkk