Koenkk/zigbee2mqtt #1929: Heiman HS2WD-E

lassieee, 2019-09-03:
After reading that the Heiman HS2WD-E was supported, I ordered one to expand my zigbee DIY security system.
Pairing it was a breeze, but I'm getting warnings in my zigbee2mqtt logs:
```
zigbee2mqtt:warn 9/3/2019, 5:49:02 PM No converter available for 'HS2WD-E' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":61}}'
  zigbee2mqtt:warn 9/3/2019, 5:49:02 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```
The documentation of the siren states it has a backup battery, so it can keep making noise, even after being pulled from the wall outlet. In above exerpt of the log, it shows the battery currently at 61%.
Is it possible to add this attribute so it can be monitored, or at least doesn't give a warning?

zigbee2mqtt version: koenkk/zigbee2mqtt:1.5.1-arm32v6
CC253X firmware version: 20190315 (CC2531_SOURCE_ROUTING)

Please let me know if there is anything I can do to contribute.

Koenkk, 2019-09-05:
This has already been addressed in the dev branch, will be in zigbee2mqtt 1.6.

closed 2019-09-05 by Koenkk

lassieee, 2019-09-05:
Many thanks!

TomBa01, 2020-02-01:
> After reading that the Heiman HS2WD-E was supported, I ordered one to expand my zigbee DIY security system.
> Pairing it was a breeze, and I have it working in Home Assistant, but I'm getting warnings in my zigbee2mqtt logs:
> 
> ```
> zigbee2mqtt:warn 9/3/2019, 5:49:02 PM No converter available for 'HS2WD-E' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":61}}'
>   zigbee2mqtt:warn 9/3/2019, 5:49:02 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
> ```
> 
> The documentation of the siren states it has a backup battery, so it can keep making noise, even after being pulled from the wall outlet. In above exerpt of the log, it shows the battery currently at 61%.
> Is it possible to add this attribute so it can be monitored, or at least doesn't give a warning?
> 
> zigbee2mqtt version: koenkk/zigbee2mqtt:1.5.1-arm32v6
> CC253X firmware version: 20190315 (CC2531_SOURCE_ROUTING)
> 
> Please let me know if there is anything I can do to contribute.

Halfway there. - Pairing worked, I see the entities "Battery State" and "link quality" in my lovelace UI in Home assistant. - So far so good. Question: How did you manage to trigger the alarm? i know about "The alarm can be trigged by publishing to zigbee2mqtt/[FRIENDLY_NAME]/set message {"warning": {"duration": 10, "mode": "emergency", "strobe": false}}." but where exactly do you enter this code in home assistant? I do not find this anywhere described...

lassieee, 2020-02-02:
Hi, I'm using an automation to trigger an action (sound alarm) when a trigger (front door opened) gets triggered and both me and my wife (our phones) aren't home (if/else):

```
- service: mqtt.publish
      data_template:
        topic: "zigbee2mqtt/siren1/set"
        payload: >
            {% if is_state( 'device_tracker.me', 'not_home' ) and is_state( 'device_tracker.wife', 'not_home' ) %}
            "{\"warning\": {\"duration\": 5, \"mode\": \"emergency\", \"strobe\": true}}"
            {% else %}
            "{\"warning\": {\"duration\": 5, \"mode\": \"stop\", \"strobe\": false}}"
            {% endif %}
```

If you want the whole automation, send me a pm.
Good luck :-)