Koenkk/zigbee2mqtt #1930: unreliable - due to reporting state time

jaburges, 2019-09-04:
Hardware: 
CC2531 co-ordinator latest firmware
CC2531 router latest firmware (router)

6 x E26 Hue Ambiance (Bathroom Mirror)
2 x BR30 Hue Ambiance (Bedside lights)
2 x A60 Hue White (Guestroom Bedside lights)

Co-ordinator in understairs cupboard, bulbs all within 10m (router under bathroom lights cupboard)

## What happened - 2 related issues
Pairing was relatively painless after flashing CC2531s
WiFi Mesh router (EERO) is constantly picking best channel. Moved to Channel 26 and had results with higher and stable Zigbee mesh (Link Quality is >250) 

**Issue 1** - reporting all over the place
Lights are not reporting states correctly, and group is inconsistent. 
reporting = true
## What did you expect to happen
On sending MQTT command to group, all devices should turn on or off based on command. Only a few devices respond to the command.

**Issue 2 ** - light on/off not working, but brightness does
The bulb responds to brightness controls but doesn't respond to specific on/off requests as the state of the bulb is incorrect for quite a while. 

## How to reproduce it (minimal and precise)
turn on bulb (adjust brightness fine) check state. State still shows as off
unable to complete light.turn_off (as state incorrect)

This causes HAVOC with the 6 bulbs I have in a group. 
light.turn_on group I maybe get 3 or 4 of the 6 respond.
send the commands individually I have more luck, but due to the slow reporting state I'm unable to turn off again for a while.

## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: 20190608

jaburges, 2019-09-04:
Also the same when using group:
```
zigbee2mqtt/1/set
on
```
most bulbs turn on
then
```
zigbee2mqtt/1/set
off
```
bulbs don't respond as state is still `off`

jaburges, 2019-09-04:
moving to the docker `latest-dev` seems have improved reporting.

However some lights just don't take a response from HA (and the switch reverts back)

1. Turn switch off in HA by sliding slider
2. Switch bounces back to previous state
3. Bulb most times eventually responds moments later (but usually 1 or 2 bulbs do not respond)

Koenkk, 2019-09-05:
@jaburges all these symptoms indicate that you are having range or interference issues, https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html may help

Also a linkquality of 250 is very high, I wonder if this is really true because even when holding devices next to the CC2531 I don't get these high values.

jaburges, 2019-09-05:
looks like it is interference as the network map was changing regularly back to a basic 'all connecting to the co-ordinator'
A mesh router seems to make Zigbee a non-starter it seems

closed 2019-09-06 by jaburges