Koenkk/zigbee2mqtt #1933: zigbee2mqtt and Node red admin panel

acelauwc, 2019-09-05:
Not sure if this is the correct place, but i would like to highlight this in case others having the same problem.

Just upgrade to latest version of zigbee2mqt 1.5.1 and latest source routing firmware on my zigbee network of about 40 end devices. The zigbee nework has been previously running stable for many months. However after the upgrade I see frequent disconnection and timeout on router errors. 

After troubleshooting, I found that its the node-red admin panel that is sending out network map request every minute. This cause stability issue with the zigbee network. I have since disable the every min polling and the problem seems to be revolved. 

I would suggest putting a note on the documentation linking to the node-red admin web panel  cautioning this behavior.

Koenkk, 2019-09-05:
@ben423423n32j14e could this be disabled by default? This will cause many problems for users and they may not realize where it comes from.

james-fry, 2019-09-05:
The same applies to an extent with zigbee2mqtt assistant. Both use the raw map query to get data to populate their UIs

dirstel, 2019-09-06:
At least my zigbee2mqtt-admin panel "only" polls the list of devices. This is triggered by an inject-node. I replaced this by a button to refresh - less convinient, but less trafic. I'd love smth. configurable too.

closed 2019-11-12 by stale[bot]