Koenkk/zigbee2mqtt #1934: Niko Smart Switch - Friends of Hue (Zigbee 3.0?)

WouterJN, 2019-09-05:
# Bug Report
## What happened
I know the "Niko Smart Switch" is not a supported device, my plan was to buy the switch and add the support for it to the zigbee-shepherd-converters, as per the published how-to.
Unfortunately, I’m already stuck at the first step. I cannot manage to connect the switch with my CC2530 running the “Z-Stack_Home_1.2”. I can connect the switch with my Hue bridge, so the switch itself is ok.  
I’m 100% sure that permit_join: true and that I do not get any INFO or WARN messages about a new device or not supported device.
The datasheet of the switch only mention “zigbee” as protocol, they do not mention the version of the zigbee protocol. Can it be that the switch is only compatible with zigbee 3.0? If yes, are there any plan to release a Z-stack for the CC2530?

## What did you expect to happen

see a "new device" in the log.

## How to reproduce it (minimal and precise)

See "what happened"

## Debug Info
zigbee2mqtt version:
Latest release
CC253X firmware version:
latest release

sjorge, 2019-09-05:
Looks like it is zigbee 3 form the spec:
https://www.niko.eu/nl-be/producten/niko-home-control/producten-voor-traditionele-bekabeling/geconnecteerde-schakelbare-wandcontactdoos-voor-niko-home-control-met-penaarde-en-productmodel-niko-aa84a653-36c4-55de-b6d1-d2ddfc6326dc

These also interest me, as I am currently experimenting with both zigbee2mqtt and zwave (openzwave) and zigbee2mqtt has be a 10x more pleasant to work with aside from not finding a good socket to use. So this one also ahs my interest.

Koenkk, 2019-09-05:
Probably the same as https://github.com/Koenkk/zigbee-herdsman/issues/22

WouterJN, 2019-09-05:
I think it is indeed the same issue. I would not be surprised if both switches use the same EnOcean (dolphin) modules. The Niko switch is just an EnOcean module with a very cheaply made plastic casing around it.

WouterJN, 2019-09-05:
I just found out that even the Hue Tap and many other friend of hue switchen are using this EnOcean module.

sjorge, 2019-10-04:
@Koenkk these should still not work right? (The EnOcean based modules)
I'm considering one of the Niko Outlets from the same series. As I am not happy with my zwave based outlets. (I moved everythign else to zigbee already)

closed 2019-12-10 by stale[bot]

outkastm, 2020-04-18:
I'm also intrested on this to have it work with zigbee2mqtt

Koenkk, 2020-04-19:
These are supported now, check #3322