Koenkk/zigbee2mqtt #1938: Errors during the installation of Koenkk zigbee2mqtt dongle on openhabian

shnyr, 2019-09-06:
### Hi all,
### I have Raspberry PI with actual openhabian + openhab 2.4 running on it. I bought CC2531 and flashed the dongel using my other Raspberry. It worked. Dongel is flashed. On my openhabian Raspberry is also running mosquitto Mqtt Brocker and it working fine.

### In Youtube I found a video (https://www.youtube.com/watch?v=8u9jMRod5XI&t=1s) how to set up Zigbee2mqtt dongel for Openhab 2.4. I followed this instruction.  The procedure discribed in the video is in my opinion the same like on https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

### But when I execute the comand "npm start", I get error. Look below:

 zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
 node index.js

/opt/zigbee2mqtt/lib/extension/deviceGroupMembership.js:4
const topicRegex = new RegExp(`^${settings.get().mqtt.base_topic}/bridge/device/.+/get_group_membership$`);
                                                  
TypeError: Cannot read property 'base_topic' of null
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/extension/deviceGroupMembership.js:4:54)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/controller.js:17:40)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!  /home/openhabian/.npm/_logs/2019-09-06T18_51_27_132Z-debug.log

### My " /opt/zigbee2mqtt/data/configuration.yaml " configuration file looks like:

#Required: Home Assistant integration (MQTT discovery)
homeassistant: false
#Required: allow new devices to join.
#WARNING: Disable this after all devices have been paired!
permit_join: true
#Required: MQTT settings
mqtt:
#Required: MQTT base topic for zigbee2mqtt MQTT messages
base_topic: zigbee2mqtt
#Required: MQTT server URL
server: 'mqtt://localhost:1883'
#Optional: MQTT server authentication user
#user: my_user
#Optional: MQTT server authentication password
#password: my_passwd
#Optional: MQTT client ID
client_id: 'Zigbee'
#Disable self-signed SSL certificates
reject_unauthorized: true
#Optional: Include device information to mqtt messages (default: false)
include_device_information: true
#Required: serial settings
serial:
#Required: location of CC2531 USB sniffer
port: /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001936A92B-if00
#port: /dev/ttyACM0
#Optional: disable LED of CC2531 USB sniffer
disable_led: false

### Please help me! What is the reason for this failure? How can I fix it?
### My knowledge about Linux is limited. 

### Best Regards. Shnyr

cron0mat, 2019-09-27:
Please use [http://www.yamllint.com/](http://www.yamllint.com/) to check your configuration.yaml !
I had similar problems with same setup and yaml had bad format !

Jolly-Roger82, 2019-10-28:
i have the same problem, followed steezy's tutorial but everytime it fails to install/launch

`[09:20:36] openhabian@openhab:/opt/zigbee2mqtt$ npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/lib/extension/deviceGroupMembership.js:4
const topicRegex = new RegExp(`^${settings.get().mqtt.base_topic}/bridge/device/.+/get_group_membership$`);
                                                     ^

TypeError: Cannot read property 'base_topic' of null
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/extension/deviceGroupMembership.js:4:54)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/controller.js:17:40)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/openhabian/.npm/_logs/2019-10-28T08_20_45_210Z-debug.log
`

closed 2020-01-03 by stale[bot]