Koenkk/zigbee2mqtt #1939: Xiaomi WXKG12LM - Device with modelID 'undefined' is not supported

adamtal3, 2019-09-07:
After a smooth pairing of an Aqara switch, I'm having trouble pairing a 2nd switch (of the same type). On the first few tries I got a timeout:

```
  zigbee2mqtt:warn 9/7/2019, 3:48:23 AM Message without device!
  zigbee2mqtt:error 9/7/2019, 3:48:47 AM Cannot get the Node Descriptor of the Device: 0x00XXXXXXYYYYYYYY (Error: Timed out after 10000 ms)
```

After that, I tried a trick that worked with a door sensor I had trouble pairing - I held the reset button until the led blinked and after that pressed switch button every two seconds for a while - it got it to add but as an undefined device and the messages from the switch were either empty or "release" (and sometimes "hold" even though I didn't hold it). The blue led on the switch also blinks most of the time even when I don't touch it:
```
  zigbee2mqtt:warn 9/7/2019, 3:36:58 AM Message without device!
  zigbee2mqtt:warn 9/7/2019, 3:37:08 AM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 9/7/2019, 3:37:08 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 9/7/2019, 3:37:09 AM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 9/7/2019, 3:37:09 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 9/7/2019, 3:37:18 AM Device incoming...
  zigbee2mqtt:info 9/7/2019, 3:37:18 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 9/7/2019, 3:37:20 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":105,"battery":100,"voltage":3055,"action":"hold"}'
  zigbee2mqtt:info 9/7/2019, 3:37:20 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":105,"battery":100,"voltage":3055,"action":""}'
  zigbee2mqtt:info 9/7/2019, 3:37:20 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":107,"battery":100,"voltage":3055,"action":"release"}'
  zigbee2mqtt:info 9/7/2019, 3:37:20 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":107,"battery":100,"voltage":3055,"action":""}'
  zigbee2mqtt:info 9/7/2019, 3:37:21 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":107,"battery":100,"voltage":3055,"action":"release"}'
  zigbee2mqtt:info 9/7/2019, 3:37:21 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":107,"battery":100,"voltage":3055,"action":""}'
  zigbee2mqtt:info 9/7/2019, 3:37:23 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":105,"battery":100,"voltage":3055,"action":"release"}'
  zigbee2mqtt:info 9/7/2019, 3:37:23 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":105,"battery":100,"voltage":3055,"action":""}'
  zigbee2mqtt:info 9/7/2019, 3:37:24 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":107,"battery":100,"voltage":3055,"action":"release"}'
  zigbee2mqtt:info 9/7/2019, 3:37:24 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":107,"battery":100,"voltage":3055,"action":""}'
  zigbee2mqtt:info 9/7/2019, 3:37:25 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":105,"battery":100,"voltage":3055,"action":"release"}'
  zigbee2mqtt:info 9/7/2019, 3:37:25 AM MQTT publish: topic 'zigbee2mqtt/0x00XXXXXXYYYYYYYY', payload '{"linkquality":105,"battery":100,"voltage":3055,"action":""}'
```

The other switch shows a "single" / "double" / "hold" actions when it should so the "release" and "" actions make no sense.
Any ideas?

zigbee2mqtt version: `1.5.1.2`
CC253X firmware version: `20190608`

Koenkk, 2019-09-08:
The logging relating to pairing of Xiaomi devices is crap (and will be fixed by #1888). 

However when zigbee2mqtt publishes messages to MQTT of this device, it means that is has successfully been paired (which seems to be the case). 

Could you post the debug log when pressing the button?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

adamtal3, 2019-09-08:
I added the log_level and tried clicking the switch - nothing happened.. So I paired it again, pressed the switch, double pressed it, shaked it, held it and here is the log for everything that I tried:

```
  zigbee2mqtt:debug 9/8/2019, 2:29:03 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_switch.aq3"}}' of device 'lumi.sensor_switch.aq3' (0x00158d00025dab20) of endpoint 1
  zigbee2mqtt:info 9/8/2019, 2:29:03 PM MQTT publish: topic 'homeassistant/sensor/0x00XXXXXXYYYYYYYY/click/config', payload '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","state_topic":"zigbee2mqtt/0x00XXXXXXYYYYYYYY","json_attributes_topic":"zigbee2mqtt/0x00XXXXXXYYYYYYYY","name":"0x00XXXXXXYYYYYYYY_click","unique_id":"0x00XXXXXXYYYYYYYY_click_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00XXXXXXYYYYYYYY"],"name":"0x00XXXXXXYYYYYYYY","sw_version":"Zigbee2mqtt 1.5.1","model":"Aqara wireless switch (with gyroscope) (WXKG12LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 9/8/2019, 2:29:03 PM MQTT publish: topic 'homeassistant/sensor/0x00XXXXXXYYYYYYYY/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00XXXXXXYYYYYYYY","json_attributes_topic":"zigbee2mqtt/0x00XXXXXXYYYYYYYY","name":"0x00XXXXXXYYYYYYYY_battery","unique_id":"0x00XXXXXXYYYYYYYY_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00XXXXXXYYYYYYYY"],"name":"0x00XXXXXXYYYYYYYY","sw_version":"Zigbee2mqtt 1.5.1","model":"Aqara wireless switch (with gyroscope) (WXKG12LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 9/8/2019, 2:29:03 PM MQTT publish: topic 'homeassistant/sensor/0x00XXXXXXYYYYYYYY/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00XXXXXXYYYYYYYY","json_attributes_topic":"zigbee2mqtt/0x00XXXXXXYYYYYYYY","name":"0x00XXXXXXYYYYYYYY_linkquality","unique_id":"0x00XXXXXXYYYYYYYY_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00XXXXXXYYYYYYYY"],"name":"0x00XXXXXXYYYYYYYY","sw_version":"Zigbee2mqtt 1.5.1","model":"Aqara wireless switch (with gyroscope) (WXKG12LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:debug 9/8/2019, 2:29:11 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_switch.aq3"}}' of device 'lumi.sensor_switch.aq3' (0x00XXXXXXYYYYYYYY) of endpoint 1
  zigbee2mqtt:debug 9/8/2019, 2:29:11 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"LUMI","modelId":"lumi.sensor_swit","powerSource":3}}' of device 'lumi.sensor_switch.aq3' (0x00XXXXXXYYYYYYYY) of endpoint 1
  zigbee2mqtt:debug 9/8/2019, 2:29:18 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_switch.aq3"}}' of device 'lumi.sensor_swit' (0x00XXXXXXYYYYYYYY) of endpoint 1
  zigbee2mqtt:debug 9/8/2019, 2:29:19 PM Received zigbee message of type 'devInterview' with data '"0x00XXXXXXYYYYYYYY"'
  zigbee2mqtt:debug 9/8/2019, 2:29:24 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_switch.aq3"}}' of device 'lumi.sensor_swit' (0x00XXXXXXYYYYYYYY) of endpoint 1
  zigbee2mqtt:debug 9/8/2019, 2:29:24 PM Received zigbee message of type 'devInterview' with data '"0x00XXXXXXYYYYYYYY"'
  zigbee2mqtt:debug 9/8/2019, 2:29:29 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_switch.aq3"}}' of device 'lumi.sensor_swit' (0x00XXXXXXYYYYYYYY) of endpoint 1
  zigbee2mqtt:debug 9/8/2019, 2:29:30 PM Received zigbee message of type 'devInterview' with data '"0x00XXXXXXYYYYYYYY"'
  zigbee2mqtt:debug 9/8/2019, 2:29:34 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_switch.aq3"}}' of device 'lumi.sensor_swit' (0x00XXXXXXYYYYYYYY) of endpoint 1
  zigbee2mqtt:debug 9/8/2019, 2:29:34 PM Received zigbee message of type 'devInterview' with data '"0x00XXXXXXYYYYYYYY"'
  zigbee2mqtt:debug 9/8/2019, 2:29:34 PM Accept device incoming with ieeeAddr '0x00XXXXXXYYYYYYYY' permit join is 'true'
  zigbee2mqtt:debug 9/8/2019, 2:29:34 PM Allowing device '0x00XXXXXXYYYYYYYY' to join
  zigbee2mqtt:debug 9/8/2019, 2:29:34 PM Received zigbee message of type 'devIncoming' with data '"0x00XXXXXXYYYYYYYY"' of device 'lumi.sensor_swit' (0x00XXXXXXYYYYYYYY) of endpoint 1
  zigbee2mqtt:info 9/8/2019, 2:29:34 PM Device incoming...
  zigbee2mqtt:info 9/8/2019, 2:29:34 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 9/8/2019, 2:29:34 PM Received zigbee message of type 'devStatus' with data '"online"' of device 'lumi.sensor_swit' (0x00XXXXXXYYYYYYYY) of endpoint 1
  zigbee2mqtt:debug 9/8/2019, 2:29:43 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_switch.aq3"}}' of device 'lumi.sensor_swit' (0x00XXXXXXYYYYYYYY) of endpoint 1
  zigbee2mqtt:debug 9/8/2019, 2:29:43 PM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_switch.aq3"}}' of device 'lumi.sensor_swit' (0x00XXXXXXYYYYYYYY) of endpoint 1
[... cut ...]
  zigbee2mqtt:debug 9/8/2019, 2:31:25 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_switch.aq3"}}' of device 'lumi.sensor_swit' (0x00XXXXXXYYYYYYYY) of endpoint 1
```

I just changed the actual entity ID with 0x00XXXXXXYYYYYYYY and copied everything since I started pairing. I'm expecting actions similar to the other switch of the same type I have ("single", "double", "hold", "shake") but all I get is an empty action and a "release" action right after it.
Any chance the switch it-self is defective? I don't have a different hub to test it.

james-fry, 2019-09-08:
From the version number you posted I guess you are using hassio addon.
Maybe this advice will help the pairing?

https://github.com/Koenkk/zigbee2mqtt/issues/1918#issuecomment-527235404

I am also having issues pairing on 1.5.1(.2) and this seemed to work for me.

closed 2019-11-14 by stale[bot]