Koenkk/zigbee2mqtt #1941: Wrong payload length in packet via publish

alexmorbo, 2019-09-08:
# Bug Report
## What happened
Got wrong packet length byte when publish packet include device information (`include_device_information: true`)

## What did you expect to happen

When include_device_information: false in z2m logs:
```
9/8/2019, 10:03:37 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device 'lumi.sensor_swit' (0x00158d0001eb90f6) of endpoint 1
9/8/2019, 10:03:37 PM - info: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"linkquality":52,"battery":100,"voltage":3005,"click":"single"}'
```
At my backend (php) i got:
```
0T\x00\x12zigbee2mqtt/Button{"linkquality":57,"battery":100,"voltage":3005,"click":"single"}
```
Byte `T` means payload length is 84, thats correct

Now i turn on include_device_information to true
z2m logs:
```
9/8/2019, 10:08:40 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device 'lumi.sensor_swit' (0x00158d0001eb90f6) of endpoint 1
9/8/2019, 10:08:40 PM - info: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"linkquality":52,"battery":100,"voltage":3005,"click":"single","device":{"ieeeAddr":"0x00158d0001eb90f6","friendlyName":"Button","type":"EndDevice","nwkAddr":5,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_swit","hwVersion":1,"swBuildId":"3000-0001","dateCode":"20180629","status":"online"}}'
```
At my backend (php):
```
b"0Û\x02\x00\x12zigbee2mqtt/Button{"linkquality":52,"battery":100,"voltage":3005,"click":"single","device":{"ieeeAddr":"0x00158d0001eb90f6","friendlyName":"Button","type":"EndDevice","nwkAddr":5,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_swit","hwVersion":1,"swBuildId":"3000-0001","dateCode":"20180629","status":"online"}}"
```
Byte `Û` means payload length is 195, but real length is more than 300 symbols

## Debug Info
zigbee2mqtt version: 1.5.1 (#ac3b924)
CC253X firmware version:

Koenkk, 2019-09-09:
zigbee2mqtt uses the following MQTT library: https://www.npmjs.com/package/mqtt. 

Could you check, by creating a small example by yourself, if the issue comes from that library?

alexmorbo, 2019-09-11:
It's strange

In WireShark i got correct **payload length**
[image]

And this is full **payload** with device info
[image]

Koenkk, 2019-09-11:
ok, I will close this as it is not a zigbee2mqtt issue.

closed 2019-09-11 by Koenkk