Koenkk/zigbee2mqtt #1944: Battery Level eCozy 1TST-EU

sti0, 2019-09-09:
# Bug Report
## What happened
I paired a eCozy 1TST-EU 4 days ago and it works as expected but I miss the battery level information. I only got the voltage. I'm using the HASSIO plugin and zigbee2mqtt published a battery sensor but the current level is missing. Does this device don't deliver battery level information?

This is the JSON payload that I get:
  zigbee2mqtt:info 9/9/2019, 2:45:36 PM MQTT publish: topic 'zigbee2mqtt/clim_buero', payload '{"voltage":0.29,"linkquality":89,"local_temperature":19.31,"occupied_heating_setpoint":8,"setpoint_change_amount":-8,"setpoint_change_source":2,"setpoint_change_source_timestamp":93125}'

## What did you expect to happen
Get the battery level in JSON payload like for example Xiaomi and Hue devices.

## How to reproduce it (minimal and precise)
Pair the device and wait for MQTT messages.

## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: 20190425
hassio plugin: 1.5.1.2

Koenkk, 2019-09-10:
What type of battery does this device contain?

sti0, 2019-09-10:
It contains 4 AA batteries.
https://en.wikipedia.org/wiki/AA_battery

Koenkk, 2019-09-11:
Knowing that it has 4AA batteries, we need a function to convert the voltage to %, could you do some investigation on this?

sti0, 2019-09-11:
Ok. Now I got it ;)

Actually I placed 4 new AA batteries and voltage shows 0.29.

I will do further investigation and come back to you.

sti0, 2019-09-21:
I'm not using this device anymore. Battery consuming is way to much...

closed 2019-09-21 by sti0