Koenkk/zigbee2mqtt #1947: Please help with adding support for icasa zigbee 3.0 switch ICZB0IW11SW

rutgerhendriks, 2019-09-09:
# Bug Report
## What happened
While trying to subscribe the icasa ICZB-IW11SW it shows the device is not supported. Some time later it crashes, but that seems to be the same issue as issue #1946 
I'm attaching the debug logging of the pairing attempt. I'm able to edit the devices.js so try to add support for the device, but I am at a loss as to what I need to do. If someone would assist that would be great!
## What did you expect to happen
Support should be added
## How to reproduce it (minimal and precise)
Press the 'reset' button on the ICZB-IW11SW 5 times in a row, then the device will try to repair resulting in the attached logging
[icasa switch subscribe crash.txt](https://github.com/Koenkk/zigbee2mqtt/files/3592197/icasa.switch.subscribe.crash.txt)

## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: 20190619

james-fry, 2019-09-09:
Do you know what is the difference between the ICZB-IW11SW and the I[CZB-IW11D, which is supported](https://www.zigbee2mqtt.io/devices/ICZB-IW11D.html)?

In devices.js you could try to duplicate the entry for the ICZB-IW11D:

```
    {
        zigbeeModel: ['ICZB-IW11D'],
        model: 'ICZB-IW11D',
        vendor: 'iCasa',
        description: 'Zigbee 3.0 Dimmer',
        extend: generic.light_onoff_brightness,
    },
```

rutgerhendriks, 2019-09-09:
added to const generic:

> 
    light_onoff: {
        supports: 'on/off',
        fromZigbee: [
            fz.ignore_light_brightness_change, fz.state_change, fz.state, fz.brightness_report,
        ].concat(defaultIgnoreConverters),
        toZigbee: [tz.on_off, tz.ignore_transition, tz.light_alert],
    },

added after ICZB-IW11D:
>
    {                                                               
        zigbeeModel: ['ICZB-IW11SW'],
        model: 'ICZB-IW11SW',
        vendor: 'iCasa',               
        description: 'Zigbee 3.0 Switch',
        extend: generic.light_onoff,                                                           
    },  

This results in the following debug logging:
>
  zigbee2mqtt:debug 9/9/2019, 10:42:46 PM Received MQTT message on 'zigbee2mqtt/bridge/config/log_level' with data 'Debug'
  zigbee2mqtt:info 9/9/2019, 10:42:46 PM Switching log level to 'debug'
  zigbee2mqtt:info 9/9/2019, 10:42:46 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"bd88016","coordinator":20190619,"log_level":"debug","permit_join":false}'
  zigbee2mqtt:debug 9/9/2019, 10:42:47 PM Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
  zigbee2mqtt:info 9/9/2019, 10:42:47 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 9/9/2019, 10:42:47 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"bd88016","coordinator":20190619,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 9/9/2019, 10:42:54 PM Received zigbee message of type 'devLeaving' with data '"0x14b457fffec4d994"' of endpoint undefined
  zigbee2mqtt:warn 9/9/2019, 10:42:54 PM Message without device!
  zigbee2mqtt:info 9/9/2019, 10:42:58 PM Successfully configured Woonkamer TV Nis links Hue motion (0x00178801032aba3e)
  zigbee2mqtt:debug 9/9/2019, 10:42:58 PM Configuring Trap Hue motion (0x00178801032ab91e) ...
  zigbee2mqtt:debug 9/9/2019, 10:43:00 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"icasa","modelId":"ICZB-IW11SW","powerSource":1}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:00 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.5.3_r40"}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:01 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":0,"manufacturerName":"icasa"}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:01 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"9":255,"modelId":"ICZB-IW11SW","dateCode":"NULL","powerSource":1,"appProfileVersion":255}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:01 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"10":{"type":"Buffer","data":[48,48]},"11":"NULL","16389":0,"30721":49246,"swBuildId":"2.5.3_r40"}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:01 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"30722":528,"30723":255,"30724":1,"34819":0,"34820":0}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:01 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"36864":255,"36865":1000,"36866":{"type":"Buffer","data":[92,147,221,104,137,42,50,65,181,244,163,66,198,67,238,2]},"36867":4644,"65533":1}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:01 PM Received zigbee message of type 'devInterview' with data '"0x14b457fffec4d994"'
  zigbee2mqtt:debug 9/9/2019, 10:43:02 PM Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:02 PM Received zigbee message of type 'devInterview' with data '"0x14b457fffec4d994"'
  zigbee2mqtt:debug 9/9/2019, 10:43:02 PM Received zigbee message of type 'readRsp' with data '{"cid":"genGroups","data":{"65533":1,"nameSupport":0}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:02 PM Received zigbee message of type 'devInterview' with data '"0x14b457fffec4d994"'
  zigbee2mqtt:debug 9/9/2019, 10:43:02 PM Received zigbee message of type 'readRsp' with data '{"cid":"genScenes","data":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:02 PM Received zigbee message of type 'readRsp' with data '{"cid":"genScenes","data":{"65533":1}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
  zigbee2mqtt:debug 9/9/2019, 10:43:02 PM Received zigbee message of type 'devInterview' with data '"0x14b457fffec4d994"'
  zigbee2mqtt:debug 9/9/2019, 10:43:03 PM Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"16387":255,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}}' of device 'ICZB-IW11SW' (0x14b457fffec4d994) of endpoint 1
[... cut ...]
    at Controller.emit (events.js:182:13)
    at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/dist/lib/components/af.js:640:23)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-09-09T20_43_04_444Z-debug.log

james-fry, 2019-09-09:
It got you a bit further :) 
Im sorry but the error is outside my limited expertise and seems to be same issue as you got with the dimmer version in the other issue you posted - https://github.com/Koenkk/zigbee2mqtt/issues/1946

Since the dimmer is supported my guess is that this is a bug from the change to zigbee-herdsman

rutgerhendriks, 2019-09-19:
Support is added to latest-dev

closed 2019-09-19 by rutgerhendriks