Koenkk/zigbee2mqtt #1948: Transition of group

copystring, 2019-09-10:
# Bug Report
## What happened
Tried out the new default transition time feature.

It works perfectly when switching indiviual lights. Better than I actually expected. However with groups it does not work at all. I have added the transition to the group itself and to the lights. Switching the group does not transition.
## What did you expect to happen
Transition on/off via group
## How to reproduce it (minimal and precise)
add transition: 0.5 to the lights and group of your choice. Switch the group: no transition. Switch the individual lights: transitions.
## Debug Info
zigbee2mqtt version:  of 1.6 (actually using dev version 1.6.0 (commit #411f1fa))
CC253X firmware version: source routing firmware 20190706.

Koenkk, 2019-09-10:
Implemented in the dev branch!

closed 2019-09-10 by Koenkk

copystring, 2019-09-10:
Thanks. Works great!