Koenkk/zigbee2mqtt #1950: new device in the dev version

zhapwey, 2019-09-10:
# Bug Report
## What happened
I am trying to include a new device in the version of (1.6), when I have the ID model I add the device in devices.js(from_zigbee and to_zigbee without anything).

I only receive this message PM Device '0x000d6ffffe51fffa' announced itself.

If I delete it from devices.js I receive the message that the device is not supported and does not send messages again.

## What did you expect to happen

to receive the messages that the device sends to add them to devices.js

## How to reproduce it (minimal and precise)

add device not supported in devices.js empty.

## Debug Info

 zigbee2mqtt:debug 2019-9-10 4:05:04 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:05:34 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:06:04 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:06:34 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:07:04 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:07:34 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:08:04 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:08:34 PM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 2019-9-10 4:08:34 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:09:04 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:09:34 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:10:04 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:10:34 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:11:04 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:11:34 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:12:04 PM Device '0x000d6ffffe51fffa' announced itself
  zigbee2mqtt:debug 2019-9-10 4:12:34 PM Device '0x000d6ffffe51fffa' announced itself

zigbee2mqtt version: 
CC253X firmware version:
{"version":"1.6.0","commit":"411f1fa","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}

Koenkk, 2019-09-13:
What device is it and could you share your `database.db`?

zhapwey, 2019-09-13:
hello:
The device is a TVR, that the compatible ones that are compatible are expensive (eurotronic) or are not manufactured anymore (ecozy).
I copy link: https://spanish.alibaba.com/product-detail/wifi-smart-home-tuya-zigbee-thermostatic-radiator-valve-head-62079929222.html?spm=a2700.8443308.0.0.27543e5fbcOFoL

this is mi database.db
https://pastebin.com/2uaU6zmG

the device is _TYST11_2dpplnsn.

Thanks

Koenkk, 2019-09-15:
Can you try with 

```
{
        zigbeeModel: ['dpplnsn\u0000'],
        model: 'TODO',
        vendor: 'Alibaba',
        description: 'Thermostat',
        supports: 'temperature, heating/cooling system control',
        fromZigbee: [
            fz.generic_battery_voltage,
            fz.thermostat_att_report,
        ],
        toZigbee: [
            tz.factory_reset, tz.thermostat_local_temperature, tz.thermostat_local_temperature_calibration,
            tz.thermostat_occupancy, tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_occupied_cooling_setpoint,
            tz.thermostat_unoccupied_heating_setpoint, tz.thermostat_setpoint_raise_lower,
            tz.thermostat_remote_sensing, tz.thermostat_control_sequence_of_operation, tz.thermostat_system_mode,
            tz.thermostat_weekly_schedule, tz.thermostat_clear_weekly_schedule, tz.thermostat_weekly_schedule_rsp,
            tz.thermostat_relay_status_log, tz.thermostat_relay_status_log_rsp,
        ],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(3);
            const binds = [
                'genBasic', 'genIdentify', 'genPowerCfg', 'genTime', 'hvacThermostat', 'hvacUserInterfaceCfg',
            ];
            await bind(endpoint, coordinatorEndpoint, binds);
            await configureReporting.thermostatTemperature(endpoint);
        },
    },
```

and try to execute commands? The commands can be found here: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html (try e.g. `occupied_heating_setpoint`)

zhapwey, 2019-09-16:
hello:

Thanks for you help.
With you device, now recive two types messages:
1. https://pastebin.com/JTPGMufi
2. If execute commands:

 zigbee2mqtt:debug 2019-9-16 10:03:31 AM Received MQTT message on 'zigbee2mqtt/0x000d6ffffe51fffa/set' with data '{"unoccupied_heating_setpoint": 16}'
  zigbee2mqtt:info 2019-9-16 10:03:31 AM Zigbee publish to device '0x000d6ffffe51fffa', hvacThermostat - write - [{"attrId":20,"dataType":41,"attrData":1600}] - {"manufSpec":0,"disDefaultRsp":0} - 1
  zigbee2mqtt:debug 2019-9-16 10:03:34 AM Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6ffffe51fffa"' of device 'dpplnsn' (0x000d6ffffe51fffa) of endpoint 1

BR

zhapwey, 2019-09-19:
hello:

no idea?

BR

Koenkk, 2019-09-20:
Do you see the heating setpoint change on the device after executing the command?

zhapwey, 2019-09-23:
I have made a totally new installation. I have synchronized the device just to see what it does.

Just sync now do this.

https://pastebin.com/sMndGhdL

And I receive these messages do or don't do things with the device.

Do you have a possible solution?

BR

Koenkk, 2019-09-23:
I see that the configure fails; can you try with below and execute commands again or press buttons on the thermostat itself?

```
{
        zigbeeModel: ['dpplnsn\u0000'],
        model: 'TODO',
        vendor: 'Alibaba',
        description: 'Thermostat',
        supports: 'temperature, heating/cooling system control',
        fromZigbee: [
            fz.generic_battery_voltage,
            fz.thermostat_att_report,
        ],
        toZigbee: [
            tz.factory_reset, tz.thermostat_local_temperature, tz.thermostat_local_temperature_calibration,
            tz.thermostat_occupancy, tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_occupied_cooling_setpoint,
            tz.thermostat_unoccupied_heating_setpoint, tz.thermostat_setpoint_raise_lower,
            tz.thermostat_remote_sensing, tz.thermostat_control_sequence_of_operation, tz.thermostat_system_mode,
            tz.thermostat_weekly_schedule, tz.thermostat_clear_weekly_schedule, tz.thermostat_weekly_schedule_rsp,
            tz.thermostat_relay_status_log, tz.thermostat_relay_status_log_rsp,
        ],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            const binds = [
                'genBasic', 'genIdentify', 'genPowerCfg', 'genTime', 'hvacThermostat', 'hvacUserInterfaceCfg',
            ];
            await bind(endpoint, coordinatorEndpoint, binds);
            await configureReporting.thermostatTemperature(endpoint);
        },
    },
```

zhapwey, 2019-09-25:
hello:

I constantly receive these messages, press buttons on the device or not

https://pastebin.com/p4wQbGtK

thanks for your help

closed 2019-12-01 by stale[bot]