Koenkk/zigbee2mqtt #1952: UnhandledPromiseRejectionWarning

bochelork, 2019-09-10:
Running the latest dev:

```
zigbee2mqtt version 1.6.0 (commit #411f1fa)
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190706}}'
```

 I'm receiving besides the MQTT-message some frequent warning:

(node:5306) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 37): Error: Data request failed with error: 'No network route' (205)

Rejection ID is counting, I receive between 0-3 messages/minute

```
  zigbee2mqtt:debug 2019-9-10 19:36:08 Received Zigbee message from 'SOCKET1' of type 'attributeReport' with data '{"65281":{"3":39,"5":53,"7":"0x0000000000000000","8":4894,"9":1280,"100":1,"149":86.5730209350586,"152":0.7599999904632568,"154":0}}' from endpoint 1 with groupID 0
  zigbee2mqtt:info 2019-9-10 19:36:08 MQTT publish: topic 'zigbee2mqtt/SOCKET1', payload '{"state":"ON","linkquality":52,"power":0.76,"voltage":null,"consumption":86.57,"temperature":39,"last_seen":"2019-09-10T17:36:08.393Z"}'
  zigbee2mqtt:debug 2019-9-10 19:36:09 Saving state to file /volumeUSB1/usbshare/zigbee2mqtt/data/state.json
(node:5306) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 44): Error: Data request failed with error: 'No network route' (205)
(node:5306) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 45): Error: Data request failed with error: 'No network route' (205)
(node:5306) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 46): Error: Data request failed with error: 'No network route' (205)
  zigbee2mqtt:debug 2019-9-10 19:38:53 Received Zigbee message from 'SOCKET2' of type 'attributeReport' with data '{"65281":{"3":29,"5":24,"7":"0x0000000000000000","8":4894,"9":3080,"100":1,"149":7.40863561630249,"152":2.5899999141693115,"154":0}}' from endpoint 1 with groupID 0
  zigbee2mqtt:info 2019-9-10 19:38:53 MQTT publish: topic 'zigbee2mqtt/SOCKET2', payload '{"state":"ON","linkquality":94,"power":2.59,"voltage":null,"consumption":7.41,"temperature":29,"last_seen":"2019-09-10T17:38:53.061Z"}'
```

I've some difficulties with pairing, but I'm not sure if this is related to the errors messages.

Koenkk, 2019-09-10:
Can you provide the logging when running with `DEBUG=*`.

bochelork, 2019-09-10:
https://pastebin.com/Z8e4fZBP

bochelork, 2019-09-13:
Installed the update, but all my last-seen values in state.json are reset to 9-9-2019 or older (in the backup from before the update the dates are from today). They are not being updated anymore. Not sure if it's related, I can start a new issue if you want?

Koenkk, 2019-09-13:
This is expected, lastSeen is now being tracked in zigbee-herdsman instead of state.json.

closed 2019-09-13 by Koenkk