Koenkk/zigbee2mqtt #1953: can't re-pair Tradfri LED

iz3man, 2019-09-10:
I had this LED paired with my CC2531. Then I had to remove the device, and later added it. Most devices paired fine, but the Tradfri LED doesn't want to pair.
This happens when I reset the LED:

```
Sep 10 21:00:34 hassbian npm[2282]: #033[32m  zigbee2mqtt:info#033[39m 9/10/2019, 9:00:34 PM Device '0x086bd7fffe0202ae' joined
Sep 10 21:00:34 hassbian npm[2282]: #033[32m  zigbee2mqtt:info#033[39m 9/10/2019, 9:00:34 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x086bd7fffe0202ae"}'
Sep 10 21:00:34 hassbian npm[2282]:     at Object.addDevice (/opt/zigbee2mqtt/lib/util/settings.js:258:15)
Sep 10 21:00:34 hassbian npm[2282]:     at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:166:40)
Sep 10 21:00:53 hassbian npm[2282]: #033[31m  zigbee2mqtt:error#033[39m 9/10/2019, 9:00:53 PM Failed to interview '0x086bd7fffe0202ae', device has not succesfully been paired
Sep 10 21:00:53 hassbian npm[2282]: #033[32m  zigbee2mqtt:info#033[39m 9/10/2019, 9:00:53 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendlyName":"0x086bd7fffe0202ae"}}'
```
Pairing didn't work with 1.5.x, so I upgraded to 1.6, and the above is the log.

Koenkk, 2019-09-11:
Can you try with the latest dev? Pairing has been improved here (#1888)

iz3man, 2019-09-13:
This is the console out put with DEBUG=* enabled:

```
root@hassbian:/opt/zigbee2mqtt # DEBUG=* npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:info 9/13/2019, 5:05:20 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-09-13.17-05-19'
  zigbee2mqtt:debug 9/13/2019, 5:05:20 PM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-09-04.10-52-11'
  zigbee2mqtt:debug 9/13/2019, 5:05:20 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 9/13/2019, 5:05:20 PM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 9/13/2019, 5:05:20 PM Starting zigbee2mqtt version 1.6.0 (commit #798a4ee)
  zigbee2mqtt:info 9/13/2019, 5:05:20 PM Starting zigbee-herdsman...
  zigbee2mqtt:debug 9/13/2019, 5:05:20 PM Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:znp:log Opening with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  serialport/stream opening path: /dev/ttyACM0 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +4ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyACM0 +2ms
  serialport/stream _read reading +1ms
  serialport/binding-abstract read +7ms
  zigbee-herdsman:znp:log Serialport opened +16ms
  zigbee-herdsman:znp:log Writing skip bootloader payload +1ms
  zigbee-herdsman:unpi:writer --> buffer [239] +0ms
  serialport/stream _write 1 bytes of data +6ms
  serialport/binding-abstract write 1 bytes +6ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +5ms
  serialport/bindings/poller Polling for "readable" +18ms
  serialport/bindings/unixWrite write returned null 1 +5ms
  serialport/bindings/unixWrite wrote 1 bytes +1ms
  serialport/bindings/unixWrite Finished writing 1 bytes +0ms
  serialport/stream binding.write write finished +13ms
  zigbee-herdsman:znp:SREQ --> SYS - version - {} +0ms
  zigbee-herdsman:unpi:writer --> frame [254,0,33,2,35] +1s
  serialport/stream _write 5 bytes of data +1s
  serialport/binding-abstract write 5 bytes +1s
  serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +1s
  serialport/bindings/unixWrite write returned null 5 +1ms
  serialport/bindings/unixWrite wrote 5 bytes +0ms
  serialport/bindings/unixWrite Finished writing 5 bytes +1ms
  serialport/stream binding.write write finished +4ms
  serialport/bindings/poller received "readable" +1s
  serialport/binding-abstract read +8ms
  serialport/bindings/unixRead Starting read +1s
  serialport/bindings/unixRead Finished read 19 bytes +1ms
  serialport/stream binding.read finished +7ms
  zigbee-herdsman:unpi:parser <-- [254,14,97,2,2,0,2,6,3,155,21,52,1,2,0,0,0,0,209] +0ms
  zigbee-herdsman:unpi:parser --- parseNext [254,14,97,2,2,0,2,6,3,155,21,52,1,2,0,0,0,0,209] +2ms
  zigbee-herdsman:unpi:parser --> parsed 14 - 3 - 1 - 2 - [�4] - 209 +2ms
  zigbee-herdsman:znp:SRSP <-- SYS - version - {"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619} +0ms
  zigbee-herdsman:unpi:parser --- parseNext [] +12ms
  zigbee-herdsman:controller:zStack Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}) +0ms
  zigbee-herdsman:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +47ms
  zigbee-herdsman:unpi:writer --> frame [254,3,33,8,0,15,0,37] +44ms
  serialport/stream _write 8 bytes of data +32ms
  serialport/binding-abstract write 8 bytes +35ms
  serialport/bindings/unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +41ms
  serialport/stream _read reading +3ms
  serialport/binding-abstract read +4ms
  serialport/bindings/unixRead Starting read +38ms
  serialport/bindings/unixWrite write returned null 8 +3ms
  serialport/bindings/unixWrite wrote 8 bytes +1ms
  serialport/bindings/unixWrite Finished writing 8 bytes +0ms
  serialport/stream binding.write write finished +4ms
[... 364 lines cut ...]
  zigbee-herdsman:znp:AREQ <-- ZDO - permitJoinInd - {"duration":0} +1m
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  serialport/stream #flush +8ms
  serialport/binding-abstract flush +13ms
  serialport/stream _read reading +3ms
  serialport/binding-abstract read +2ms
  serialport/bindings/unixRead Starting read +13ms
  serialport/stream binding.flush finished +2ms
  serialport/stream #close +2ms
  serialport/binding-abstract close +5ms
  serialport/bindings/poller Stopping poller +21ms
[... cut ...]
  serialport/stream disconnected aborted because already closed Error: Port is not open
    at fs.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/unix-read.js:14:25)
    at FSReqWrap.wrapper [as oncomplete] (fs.js:467:17) +13ms
  serialport/stream _read queueing _read for after open +1ms
  serialport/stream binding.close finished +1ms
  zigbee-herdsman:znp:log Serialport closed +1m
  zigbee2mqtt:info 9/13/2019, 5:06:38 PM zigbee-herdsman stopped
```

Pasted on pastbin as well: https://pastebin.com/BJZpWxF8

iz3man, 2019-09-13:
Here again the short version from the syslog about pairing the LED bulb:
```

  zigbee2mqtt:info 9/13/2019, 5:19:49 PM Starting interview of '0x086bd7fffe0202ae'
  zigbee2mqtt:info 9/13/2019, 5:19:49 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendlyName":"0x086bd7fffe0202ae"}}'
  zigbee2mqtt:error 9/13/2019, 5:20:09 PM Failed to interview '0x086bd7fffe0202ae', device has not succesfully been paired
  zigbee2mqtt:info 9/13/2019, 5:20:09 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendlyName":"0x086bd7fffe0202ae"}}'

```

Koenkk, 2019-09-14:
After zigbee2mqtt started, did you try to re-pair by turning it on/off 5 times? I dont see that in the log

iz3man, 2019-09-15:
I thought it was 6 times?! But yes it did.

Koenkk, 2019-09-15:
Can you provide the DEBUG=* log when doing that?

iz3man, 2019-09-17:
It's kinda strange. When i start the daemon normally it shows all devices as online, but doesn't update the state.
If i stop it, and start it manually with "DEBUG=* npm start" it starts fine, throws no errors, shows the connected devices, but then just does NOTHING. It doesn't update the states, it doesn't show any connection messages. This is with DEV updated two days ago.

Koenkk, 2019-09-17:
Is `permit_join: true`?

iz3man, 2019-09-18:
Sure

closed 2019-11-24 by stale[bot]

FlexServ, 2019-11-30:
Having the same problem, also with an Ikea Led bulb.
Attached the debug log.

[debug.log](https://github.com/Koenkk/zigbee2mqtt/files/3907156/debug.log)