Koenkk/zigbee2mqtt #1955: Xiaomi sensors report battery as 100%

radinsky, 2019-09-11:


Koenkk, 2019-09-11:
What voltage do you get?

radinsky, 2019-09-11:
3055 or 3065 for motion sensors (RTCGQ11LM) - all of their batteries about 2 years old and in xiaomi app was showing about 40%..
for all door/window sensors (MCCGQ11LM) I got voltage value 3025 (they are also all about 2 years old)

Koenkk, 2019-09-11:
The batteries in these devices are 3 volts, based on a voltage of 3.055v (3055) I would assume it is full. However it seems that the Xiaomi bridge uses a different algorithm to calculate this. 

The current algorithm that zigbee2mqtt uses can be found here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L55. I'm open for improvements to this.

closed 2019-11-10 by Koenkk

radinsky, 2019-11-11:
The issue still exists, looks like it not reading the battery value correct, validated on most recent v.1.7.0