Koenkk/zigbee2mqtt #1956: npm start

akala57, 2019-09-11:
Hi guys,

I am trying to run Zigbee2mqtt on my rpi 3. I just flashed my CC2531 with CC Debugger.
I followed the steps in documentation but everytime I tried, the command `npm start` ends with errors. I cannot find an issue. I tried to clean npm cache, remove package-lock.json, remove and reinstall npm but it did not work.

Here is what comes when i run `npm start` : 

$ npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/lib/extension/deviceGroupMembership.js:4
const topicRegex = new RegExp(`^${settings.get().mqtt.base_topic}/bridge/device/.+/get_group_membership$`);
                                                     ^

TypeError: Cannot read property 'base_topic' of null
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/extension/deviceGroupMembership.js:4:54)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/controller.js:17:40)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/jeedom/.npm/_logs/2019-09-11T15_52_25_027Z-debug.log

I hope you can help me. 
Thank you in advance.

Koenkk, 2019-09-11:
Make sure that the `base_topic` is in the MQTT section (example: https://www.zigbee2mqtt.io/configuration/configuration.html).

closed 2019-09-11 by Koenkk

akala57, 2019-09-12:
I copy-paste this in the configuration.yalm when it was asked by the tutorial. So my file is 100% similar to this.

akala57, 2019-09-13:
What I pasted in configuration.yaml : https://pastebin.com/SU0avZ0X
Don't know how but it worked once. I was able to connect the Ikea Tradfri remote controller. Zigbee2mqtt is not really easy to start for me.
There were some errors "port not open" but it seems to be ok now (used `sudo`).
I have other errors.
https://pastebin.com/LUryNHnh
I will try "5. (Optional) Running as a daemon with systemctl".

I did not mention that I use SSH to connect to my rpi.

Thank you very much Koenkk.

Koenkk, 2019-09-14:
Please read the comments carefullyin configuration.yaml not all options are required and you should only use them when needed (otherwise it will cause errors if you use them incorrectly like now)

str3000, 2019-09-15:
Ich habe das gleiche Problem nachdem ich auf V1.6 upgedatet habe. Komme leider nicht weiter.
Und zuhause geht nichts mehr. Wer hat einen Tip für mich.

______________
npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'zigbee-herdsman/dist/deprecated/ziee'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/zapp/cie.js:6:14)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/str3000/.npm/_logs/2019-09-15T08_28_52_686Z-debug.log
str3000@nuc7:/opt/zigbee2mqtt$
__________________
aus der log datei:
__________________
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.11.3
3 info using node@v10.16.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.6.0~prestart: zigbee2mqtt@1.6.0
6 info lifecycle zigbee2mqtt@1.6.0~start: zigbee2mqtt@1.6.0
7 verbose lifecycle zigbee2mqtt@1.6.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.6.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
9 verbose lifecycle zigbee2mqtt@1.6.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.6.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.6.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.6.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.6.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.6.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.15.0-62-generic
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.3
19 verbose npm  v6.11.3
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.6.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.6.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

AxHa2111, 2019-09-24:
Hello Koenkk
today i try to install zigbee2mqtt on my Raspberry an I had the same error message then STR3000.
I am not sure that it is the same problem, because for me it is the first installation.

Did you have any idea?
=====================================
After npm start 

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js
internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'zigbee-herdsman/dist/deprecated/ziee'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/zapp/cie.js:6:14)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-24T18_30_20_741Z-debug.log
===================================================

0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.16.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.6.0~prestart: zigbee2mqtt@1.6.0
6 info lifecycle zigbee2mqtt@1.6.0~start: zigbee2mqtt@1.6.0
7 verbose lifecycle zigbee2mqtt@1.6.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.6.0~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/$
9 verbose lifecycle zigbee2mqtt@1.6.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.6.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.6.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.6.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.6.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/i$
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/l$
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.6.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.98-v7+
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 verbose node v10.16.3
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.6.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.6.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

johannesbec, 2019-09-27:
Hi,

I have exactly the same problem after update. I checked the configuration.yaml carefully but did not find any difference. I even put the original configuration.yaml from the repository and ended with the same error. I only added at the end 

devices: devices.yaml
groups: groups.yaml

whereas devices.yaml contains my devices. These were created automatically by the version of zigbee2mqtt I used before and just copied into this file.

thanks in advance for any hint / idea and help!

Johannes

Any idea / help?

zwergholdrio, 2019-10-11:
Hi

Same here. After updating according to the upgrade-section [here](https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html) I get the same errors as others mentioned above:

**from journalctl**

Oct 11 17:23:46 openhab npm[11896]: npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
Oct 11 17:23:46 openhab npm[11896]: npm ERR! Exit status 1
Oct 11 17:23:46 openhab npm[11896]: npm ERR!
Oct 11 17:23:46 openhab npm[11896]: npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
Oct 11 17:23:46 openhab npm[11896]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Oct 11 17:23:46 openhab npm[11896]: npm ERR! A complete log of this run can be found in:
Oct 11 17:23:46 openhab npm[11896]: npm ERR!     /home/drpi/.npm/_logs/2019-10-11T15_23_46_772Z-debug.log
Oct 11 17:23:46 openhab systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Oct 11 17:23:46 openhab systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Oct 11 17:23:46 openhab systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Oct 11 17:23:47 openhab systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Oct 11 17:23:47 openhab systemd[1]: Stopped zigbee2mqtt.
Oct 11 17:23:47 openhab systemd[1]: Started zigbee2mqtt.
Oct 11 17:23:48 openhab npm[11922]: > zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
Oct 11 17:23:48 openhab npm[11922]: > node index.js
Oct 11 17:23:48 openhab npm[11922]: internal/modules/cjs/loader.js:638
Oct 11 17:23:48 openhab npm[11922]:     throw err;
Oct 11 17:23:48 openhab npm[11922]:     ^
Oct 11 17:23:48 openhab npm[11922]: Error: Cannot find module 'semver'
Oct 11 17:23:48 openhab npm[11922]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
Oct 11 17:23:48 openhab npm[11922]:     at Function.Module._load (internal/modules/cjs/loader.js:562:25)
Oct 11 17:23:48 openhab npm[11922]:     at Module.require (internal/modules/cjs/loader.js:692:17)
Oct 11 17:23:48 openhab npm[11922]:     at require (internal/modules/cjs/helpers.js:25:18)
Oct 11 17:23:48 openhab npm[11922]:     at Object.<anonymous> (/opt/zigbee2mqtt/index.js:1:16)
Oct 11 17:23:48 openhab npm[11922]:     at Module._compile (internal/modules/cjs/loader.js:778:30)
Oct 11 17:23:48 openhab npm[11922]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
Oct 11 17:23:48 openhab npm[11922]:     at Module.load (internal/modules/cjs/loader.js:653:32)
Oct 11 17:23:48 openhab npm[11922]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
Oct 11 17:23:48 openhab npm[11922]:     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
Oct 11 17:23:48 openhab npm[11922]: npm ERR! code ELIFECYCLE
Oct 11 17:23:48 openhab npm[11922]: npm ERR! errno 1

**/home/drpi/.npm/_logs/2019-10-11T15_23_46_772Z-debug.log**

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.16.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.6.0 ~ prestart: zigbee2mqtt@1.6.0
6 info lifecycle zigbee2mqtt@1.6.0 ~ start: zigbee2mqtt@1.6.0
7 verbose lifecycle zigbee2mqtt@1.6.0 ~ start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.6.0 ~ start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.6.0 ~ start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.6.0 ~ start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.6.0 ~ start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.6.0 ~ start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.6.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.6.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.66-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.3
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.6.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.6.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

This issue was closed, but I couldn't find a solution... What needs to be done?

Cheers
Peter

johannesbec, 2019-10-11:
Hi, for me, flashing the Stick with latest firmware solved all issues. Before, I used zigbee2mqtt 1.2 and firmware from 02/2019. Now with both latest stable releases it works perfectly fine. Thanks for your excellent work, koenkk!!

Regards,
Johannes

zwergholdrio, 2019-10-11:
Hi Johannes.
Na, unfortuntely flashing the stick did not help... Tried it twice already...
Any other thoughts?
Cheers,
Peter

fefris, 2019-10-11:
Same issue i think for me after running upgrade, have not changed the config file.

`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.16.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.6.0~prestart: zigbee2mqtt@1.6.0
6 info lifecycle zigbee2mqtt@1.6.0~start: zigbee2mqtt@1.6.0
7 verbose lifecycle zigbee2mqtt@1.6.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.6.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.6.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.6.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.6.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.6.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.6.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.6.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.70-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.0
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.6.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.6.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

johannesbec, 2019-10-12:
Hi,

maybe have a look at Issue #1892. There are some hints.

Regards,

Johannes

zwergholdrio, 2019-10-12:
> […](#)

Also had no luck with each of the solutions in Issue #1892...

Will a clean uninstall/install solve the problem?

Koenkk, 2019-10-12:
@zwergholdrio did you check https://github.com/Koenkk/zigbee2mqtt/issues/1892#issuecomment-527837064? If that doesn't solve I would recommend switching to dev https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

zwergholdrio, 2019-10-28:
@Koenkk i finally found the time to switch to dev-branch and it works now. thanks for your work and help!

alexisayenko, 2019-11-04:
It doesn't work for me. I switched to dev and followed the mentioned steps.
My git status shows only configuration.yaml

I have raspberry pi 0 w.