Koenkk/zigbee2mqtt #1957: Heiman Remote (HS1RC-M) cannot pair

ascglennzarb, 2019-09-11:
**Issue**

I am getting the following error when pairing the Heiman Remote (HS1RC-M) device:

AM Failed to configure 0x005043c953279572 (0x005043c953279572) ('TypeError: Cannot read property 'status' of undefined')

It manages to identify the 3 sensors: action, battery and linkquality but no action is being recorded.

Any idea how to fix this issue or what steps can I do to identity the issue? I managed to connect some aqara sensors successfully.

Full debug log:
https://pastebin.com/5pCRxbq3

**Debug Info**
zigbee2mqtt version: 1.5.1.2 hassio addon
CC253X firmware version: 20190608

ascglennzarb, 2019-09-13:
Hi @Koenkk ,

Yes sure.

zigbee2mqtt:info 8/30/2019, 11:07:50 AM Successfully reenabled joining
zigbee2mqtt:info 8/30/2019, 11:08:02 AM New device 'RC_V14' with address 0x005043c953279572 connected!
zigbee2mqtt:info 8/30/2019, 11:08:02 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x005043c953279572","meta":{"modelID":"RC_V14"}}'
zigbee2mqtt:info 8/30/2019, 11:08:02 AM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_action","unique_id":"0x005043c953279572_action_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c953279572","sw_version":"Zigbee2mqtt 1.5.1","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info 8/30/2019, 11:08:02 AM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_battery","unique_id":"0x005043c953279572_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c953279572","sw_version":"Zigbee2mqtt 1.5.1","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info 8/30/2019, 11:08:02 AM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_linkquality","unique_id":"0x005043c953279572_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c953279572","sw_version":"Zigbee2mqtt 1.5.1","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:warn 8/30/2019, 11:08:04 AM Failed to configure 0x005043c953279572 (0x005043c953279572) ('TypeError: Cannot read property 'status' of undefined') (attempt #1)
zigbee2mqtt:warn 8/30/2019, 11:08:04 AM This can be ignored if the device is working properly
zigbee2mqtt:warn 8/30/2019, 11:08:07 AM Failed to configure 0x005043c953279572 (0x005043c953279572) ('TypeError: Cannot read property 'status' of undefined') (attempt #2)
zigbee2mqtt:warn 8/30/2019, 11:08:07 AM This can be ignored if the device is working properly
zigbee2mqtt:warn 8/30/2019, 11:08:10 AM Failed to configure 0x005043c953279572 (0x005043c953279572) ('TypeError: Cannot read property 'status' of undefined') (attempt #3)
zigbee2mqtt:warn 8/30/2019, 11:08:10 AM This can be ignored if the device is working properly
zigbee2mqtt:info 8/30/2019, 11:08:12 AM MQTT publish: topic 'zigbee2mqtt/0x005043c953279572', payload '{"battery":44.5,"linkquality":0}'
zigbee2mqtt:info 8/30/2019, 11:08:16 AM MQTT publish: topic 'zigbee2mqtt/0x005043c953279572', payload '{"battery":44.5,"linkquality":0,"battery_low":false}'
zigbee2mqtt:info 8/30/2019, 11:08:24 AM Device incoming...
zigbee2mqtt:info 8/30/2019, 11:08:24 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
zigbee2mqtt:info 8/30/2019, 11:10:30 AM Successfully reenabled joining

Koenkk, 2019-09-15:
Do you map a custom zigbee-shepherd-converters/devices.js, if so please remove it.

ascglennzarb, 2019-09-21:
Hi @Koenkk,

No I did not map anything myself. Today I updated to version  1.6 but the same issue still persists :/

Regards
Glenn

Koenkk, 2019-09-22:
Can you try again with the latest edge version? I've added extra logging which is necessary to debug this issue.

ascglennzarb, 2019-09-22:
Hi @Koenkk ,
This is the log for the latest edge version.

2019-09-22T21:51:06: PM2 log: Launching in no daemon mode
2019-09-22T21:51:06: PM2 log: App [npm:0] starting in -fork mode-
2019-09-22T21:51:06: PM2 log: App [npm:0] online
> zigbee2mqtt@1.6.0 start /app
> node index.js
  zigbee2mqtt:info 9/22/2019, 9:51:10 PM Logging to directory: '/share/zigbee2mqttedge/log/2019-09-22.21-51-09'
  zigbee2mqtt:debug 9/22/2019, 9:51:10 PM Can't load state from file /share/zigbee2mqttedge/state.json (doesn't exist)
  zigbee2mqtt:info 9/22/2019, 9:51:10 PM Starting zigbee2mqtt version 1.6.0 (commit #14a9ec9f11e12ab43401b8c4945158965e0167e0)
  zigbee2mqtt:info 9/22/2019, 9:51:10 PM Starting zigbee-herdsman...
  zigbee2mqtt:debug 9/22/2019, 9:51:10 PM Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqttedge/database.db","backupPath":"/share/zigbee2mqttedge/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
  zigbee2mqtt:info 9/22/2019, 9:51:12 PM zigbee-herdsman started
  zigbee2mqtt:info 9/22/2019, 9:51:12 PM Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
  zigbee2mqtt:debug 9/22/2019, 9:51:12 PM Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
  zigbee2mqtt:info 9/22/2019, 9:51:12 PM Currently 0 devices are joined:
  zigbee2mqtt:warn 9/22/2019, 9:51:12 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 9/22/2019, 9:51:12 PM Allowing new devices to join.
  zigbee2mqtt:warn 9/22/2019, 9:51:12 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 9/22/2019, 9:51:12 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 9/22/2019, 9:51:12 PM Connecting to MQTT server at mqtt://core-mosquitto:1883
  zigbee2mqtt:info 9/22/2019, 9:51:13 PM Connected to MQTT server
  zigbee2mqtt:info 9/22/2019, 9:51:13 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 9/22/2019, 9:51:13 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"14a9ec9f11e12ab43401b8c4945158965e0167e0","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
  zigbee2mqtt:info 9/22/2019, 9:51:59 PM Device '0x005043c953279572' joined
  zigbee2mqtt:info 9/22/2019, 9:51:59 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x005043c953279572"}}'
(node:37) UnhandledPromiseRejectionWarning: Error: Device '0x005043c953279572' already exists
    at Object.addDevice (/app/lib/util/settings.js:261:15)
    at Controller.onZigbeeEvent (/app/lib/controller.js:160:40)
(node:37) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:37) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
  zigbee2mqtt:debug 9/22/2019, 9:51:59 PM Device '0x005043c953279572' announced itself
  zigbee2mqtt:debug 9/22/2019, 9:52:04 PM Received Zigbee message from '0x005043c953279572' of type 'readResponse' with data '{"manufacturerName":"Heiman","modelId":"RC_V14","powerSource":3}' from endpoint 1 with groupID 0
  zigbee2mqtt:info 9/22/2019, 9:52:04 PM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_action","unique_id":"0x005043c953279572_action_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c953279572","sw_version":"Zigbee2mqtt 1.6.0","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 9/22/2019, 9:52:04 PM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_battery","unique_id":"0x005043c953279572_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c953279572","sw_version":"Zigbee2mqtt 1.6.0","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 9/22/2019, 9:52:04 PM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_linkquality","unique_id":"0x005043c953279572_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c [... cut ...]
  zigbee2mqtt:debug 9/22/2019, 9:52:10 PM Received Zigbee message from '0x005043c953279572' of type 'readResponse' with data '{"zclVersion":1,"appVersion":3,"stackVersion":2}' from endpoint 1 with groupID 0
  zigbee2mqtt:error 9/22/2019, 9:52:16 PM Failed to interview '0x005043c953279572', device has not succesfully been paired
  zigbee2mqtt:info 9/22/2019, 9:52:16 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x005043c953279572"}}'

Koenkk, 2019-09-23:
Can you update to the latest edge again? (one error in your log should be fixed know).

When pairing this device, please keep pressing a button on this device to keep it awake (until you see `interview_success` or `interview_failed`).

ascglennzarb, 2019-09-24:
Hi @Koenkk ,

After several attempts with the latest edge version I managed to get the interview_success message. I can see the battery percent and  linq quality but the action button is always empty. Should I be able to see something in the logs when I click on any button? Because for the time being nothing is being recorded in the logs.

[Info] Configuration file found in data path, but no backup file found in data path. Backing up existing configuration to /share/zigbee2mqttedge/.configuration.yaml.bk
2019-09-24T19:02:38: PM2 log: Launching in no daemon mode
2019-09-24T19:02:38: PM2 log: App [npm:0] starting in -fork mode-
2019-09-24T19:02:38: PM2 log: App [npm:0] online
> zigbee2mqtt@1.6.0 start /app
> node index.js
  zigbee2mqtt:info 9/24/2019, 7:02:43 PM Logging to directory: '/share/zigbee2mqttedge/log/2019-09-24.19-02-42'
  zigbee2mqtt:debug 9/24/2019, 7:02:43 PM Can't load state from file /share/zigbee2mqttedge/state.json (doesn't exist)
  zigbee2mqtt:info 9/24/2019, 7:02:43 PM Starting zigbee2mqtt version 1.6.0 (commit #14a9ec9f11e12ab43401b8c4945158965e0167e0)
  zigbee2mqtt:info 9/24/2019, 7:02:43 PM Starting zigbee-herdsman...
  zigbee2mqtt:debug 9/24/2019, 7:02:43 PM Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqttedge/database.db","backupPath":"/share/zigbee2mqttedge/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
  zigbee2mqtt:info 9/24/2019, 7:02:44 PM zigbee-herdsman started
  zigbee2mqtt:info 9/24/2019, 7:02:44 PM Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
  zigbee2mqtt:debug 9/24/2019, 7:02:44 PM Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
  zigbee2mqtt:info 9/24/2019, 7:02:44 PM Currently 0 devices are joined:
  zigbee2mqtt:warn 9/24/2019, 7:02:44 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 9/24/2019, 7:02:44 PM Allowing new devices to join.
  zigbee2mqtt:warn 9/24/2019, 7:02:44 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 9/24/2019, 7:02:44 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 9/24/2019, 7:02:44 PM Connecting to MQTT server at mqtt://core-mosquitto:1883
  zigbee2mqtt:info 9/24/2019, 7:02:45 PM Connected to MQTT server
  zigbee2mqtt:info 9/24/2019, 7:02:45 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 9/24/2019, 7:02:45 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"14a9ec9f11e12ab43401b8c4945158965e0167e0","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
  zigbee2mqtt:info 9/24/2019, 7:03:08 PM Device '0x005043c953279572' joined
  zigbee2mqtt:info 9/24/2019, 7:03:08 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x005043c953279572"}}'
  zigbee2mqtt:info 9/24/2019, 7:03:08 PM Starting interview of '0x005043c953279572'
  zigbee2mqtt:info 9/24/2019, 7:03:08 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x005043c953279572"}}'
  zigbee2mqtt:debug 9/24/2019, 7:03:10 PM Device '0x005043c953279572' announced itself
  zigbee2mqtt:debug 9/24/2019, 7:03:18 PM Device '0x005043c953279572' announced itself
  zigbee2mqtt:debug 9/24/2019, 7:03:27 PM Received Zigbee message from '0x005043c953279572' of type 'readResponse' with data '{"modelId":"RC_V14","manufacturerName":"Heiman","powerSource":3}' from endpoint 1 with groupID 0
  zigbee2mqtt:info 9/24/2019, 7:03:27 PM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_action","unique_id":"0x005043c953279572_action_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c953279572","sw_version":"Zigbee2mqtt 1.6.0","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 9/24/2019, 7:03:27 PM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_battery","unique_id":"0x005043c953279572_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c953279572","sw_version":"Zigbee2mqtt 1.6.0","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 9/24/2019, 7:03:27 PM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c9532795 [... cut ...]
  zigbee2mqtt:info 9/24/2019, 7:03:31 PM Successfully interviewed '0x005043c953279572', device has succesfully been paired
  zigbee2mqtt:info 9/24/2019, 7:03:31 PM Device '0x005043c953279572' is supported, identified as: HEIMAN Smart remote controller (HS1RC-M)
  zigbee2mqtt:info 9/24/2019, 7:03:31 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x005043c953279572","model":"HS1RC-M","vendor":"HEIMAN","description":"Smart remote controller","supported":true}}'

Koenkk, 2019-09-25:
@Fabiancrg initially added support for this device, do you still use it and does it work for you?

Fabiancrg, 2019-09-25:
@Koenkk  It stopped working a few months ago after an upgrade, cannot tell which one exactly.
It stopped working so I tried to pair it again but could not anymore.

Koenkk, 2019-09-25:
@Fabiancrg thanks for letting us know

@ascglennzarb can you share the log when starting with `DEBUG=* npm start`; please indicate at which moments you press the buttons.

ascglennzarb, 2019-09-28:
Hi @Koenkk ,

How do I start it with  DEBUG=* npm start in Hassio Home Assistant?

I set it to use log level debug in the config. Is that the same?

Thanks

Koenkk, 2019-09-28:
@ascglennzarb I just added a fix for that, please try in an hour by updating to the latest edge and setting the `zigbee_shepherd_debug` for the addon.

ascglennzarb, 2019-09-29:
Hi @Koenkk ,

Where do I set zigbee_shepherd_debug?

Koenkk, 2019-09-29:
I don't run hassio, but it should be close to where you can set e.g. the serial port.

victorsohk, 2019-10-02:
Hi! I'm having similar problem on the Heiman remote.  As my model is the same as the one mentioned in #1581, I followed @bambouzel's configurations but still no luck.

My log is long and contains lot of line starting of serialport/bindings etc so I don't know to post the relevant lines here.  Here are some lines I think related to the problem:  I would appreciate it if you can offer some helps.  Many thanks!

My firmware version is 20190608
.....
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":27514,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":15,"securityuse":0,"timestamp":1953507,"transseqnumber":0,"len":9,"data":{"0":24,"1":17,"2":1,"3":253,"4":255,"5":0,"6":33,"7":1,"8":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":17,"cmdId":"readRsp","payload":[{"attrId":65533,"status":0,"dataType":33,"attrData":1}]}} +49ms
  zigbee2mqtt:debug 10/2/2019, 8:46:51 PM Received zigbee message of type 'readRsp' with data '{"cid":"ssIasZone","data":{"65533":1}}' of device 'RC-EM' (0x000d6f00118c3ae1) of endpoint 1
  zigbee2mqtt:debug 10/2/2019, 8:46:51 PM Received zigbee message of type 'devInterview' with data '"0x000d6f00118c3ae1"'
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 18 +2s
...
  zigbee2mqtt:debug 10/2/2019, 8:46:53 PM Received zigbee message of type 'devInterview' with data '"0x000d6f00118c3ae1"'
  zigbee2mqtt:debug 10/2/2019, 8:46:53 PM Accept device incoming with ieeeAddr '0x000d6f00118c3ae1' permit join is 'true'
  zigbee2mqtt:debug 10/2/2019, 8:46:53 PM Allowing device '0x000d6f00118c3ae1' to join
  zigbee-shepherd Device: 0x000d6f00118c3ae1 join the network. +29s
  zigbee2mqtt:debug 10/2/2019, 8:46:53 PM Received zigbee message of type 'devIncoming' with data '"0x000d6f00118c3ae1"' of device 'RC-EM' (0x000d6f00118c3ae1) of endpoint 1
  zigbee2mqtt:info 10/2/2019, 8:46:53 PM Device incoming...
  zigbee2mqtt:info 10/2/2019, 8:46:53 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 10/2/2019, 8:46:53 PM Received zigbee message of type 'devStatus' with data '"online"' of device 'RC-EM' (0x000d6f00118c3ae1) of endpoint 1
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +160ms
  serialport/bindings/poller Polling for "readable" +220ms
  zigbee2mqtt:debug 10/2/2019, 8:47:12 PM Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'false'

Koenkk, 2019-10-03:
@victorsohk 

- Please first switch to the latest zigbee2mqtt dev branch (debugging is easier here) https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html
- Start zigbee2mqtt with `DEBUG=zigbee-herdsman* npm start`
- Re-pair the device
- After pairing, press some buttons on the device
- Share the log

Fabiancrg, 2019-10-03:
@Koenkk  I tried what you asked, here is the log --> https://pastebin.com/DJviw5bE

Apparently the pairing is not working for me.
Button did not generate any messgage, I suppose it's normal as it is not paired.

Hope this can help.

Koenkk, 2019-10-04:
@Fabiancrg the interview fails probably because the device goes to sleep mode during pairing. Can you try pairing it again while keep pressing buttons on the device? This should keep it awake.

victorsohk, 2019-10-04:
@Koenkk Thank you very much for your help! I've switched to the dev branch.  The log is very long and I'm trying some lines here.  (I kept pressing the pair hole or action button during the pairing but it is still not success)

zigbee-herdsman:controller:log Joining already permitted +59s
zigbee2mqtt:info  2019-10-04T12:13:25: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"732a97b","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
......
  zigbee-herdsman:controller:log New device '0x000d6f00118c3ae1' joined +1ms
  zigbee-herdsman:controller:log Creating device '0x000d6f00118c3ae1' +2ms
zigbee2mqtt:info  2019-10-04T12:13:54: Device '0x000d6f00118c3ae1' joined
zigbee2mqtt:info  2019-10-04T12:13:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f00118c3ae1"}}'
  zigbee-herdsman:controller:log Interview '0x000d6f00118c3ae1' start +56ms
zigbee2mqtt:info  2019-10-04T12:13:54: Starting interview of '0x000d6f00118c3ae1'
zigbee2mqtt:info  2019-10-04T12:13:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f00118c3ae1"}}'
  zigbee-herdsman:controller:device Interview - start device '0x000d6f00118c3ae1' +0ms
  zigbee-herdsman:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":21123,"nwkaddrofinterest":21123} +1m
.....
 zigbee-herdsman:controller:log Device joined '0x000d6f00118c3ae1' +363ms
  zigbee-herdsman:controller:log Not interviewing '0x000d6f00118c3ae1', completed 'false', in progress 'true' +2ms
.....
zigbee-herdsman:controller:device Interview - got node descriptor for device '0x000d6f00118c3ae1' +365ms
  zigbee-herdsman:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":21123,"nwkaddrofinterest":21123} +371ms
.....
  zigbee-herdsman:controller:device Interview - got active endpoints for device '0x000d6f00118c3ae1' +1s
  zigbee-herdsman:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":21123,"nwkaddrofinterest":21123,"endpoint":1} +1s
....
  zigbee-herdsman:controller:device Interview - got simple descriptor for endpoint '1' device '0x000d6f00118c3ae1' +995ms
  zigbee-herdsman:znp:SREQ --> AF - dataRequest - {"dstaddr":21123,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":1,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,2,0,5,0,4,0,7,0]}} +1s
......
 zigbee-herdsman:unpi:parser --- parseNext [254,49,68,129,0,0,0,0,131,82,1,1,0,0,0,37,150,64,0,0,29,24,2,1,5,0,0,66,5,82,67,45,69,77,4,0,0,66,6,72,69,73,77,65,78,7,0,0,48,3,131,82,29,24] +3ms
  zigbee-herdsman:unpi:parser --> parsed 49 - 2 - 4 - 129 - [�R%�@BRC-EMBHEIMAN0�R] - 24 +2ms
  zigbee-herdsman:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":21123,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":4232741,"transseqnumber":0,"len":29,"data":{"type":"Buffer","data":[24,2,1,5,0,0,66,5,82,67,45,69,77,4,0,0,66,6,72,69,73,77,65,78,7,0,0,48,3]}} +61ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":2,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":5,"status":0,"dataType":66,"attrData":"RC-EM"},{"attrId":4,"status":0,"dataType":66,"attrData":"HEIMAN"},{"attrId":7,"status":0,"dataType":48,"attrData":3}],"Cluster":{"ID":0,"attributes":{"zclVersion":{"ID":0,"type":32,"name":"zclVersion"},"appVersion":{"ID":1,"type":32,"name":"appVersion"},"stackVersion":{"ID":2,"type":32,"name":"stackVersion"},"hwVersion":{"ID":3,"type":32,"name":"hwVersion"},"manufacturerName":{"ID":4,"type":66,"name":"manufacturerName"},"modelId":{"ID":5,"type":66,"name":"modelId"},"dateCode":{"ID":6,"type":66,"name":"dateCode"},"powerSource":{"ID":7,"type":48,"name":"powerSource"},"appProfileVersion":{"ID":8,"type":48,"name":"appProfileVersion"},"swBuildId":{"ID":16384,"type":66,"name":"swBuildId"},"locationDesc":{"ID":16,"type":66,"name":"locationDesc"},"physicalEnv":{"ID":17,"type":48,"name":"physicalEnv"},"deviceEnabled":{"ID":18,"type":16,"name":"deviceEnabled"},"alarmMask":{"ID":19,"type":24,"name":"alarmMask"},"disableLocalConfig":{"ID":20,"type":24,"name":"disableLocalConfig"}},"name":"genBasic","commands":{"resetFactDefault":{"ID":0,"parameters":[],"name":"resetFactDefault"}},"commandsResponse":{}}},"networkAddress":21123,"endpoint":1,"linkquality":0,"groupID":0}' +7s
zigbee2mqtt:debug 2019-10-04T12:14:07: Received Zigbee message from '0x000d6f00118c3ae1', type 'readResponse', cluster 'genBasic', data '{"modelId":"RC-EM","manufacturerName":"HEIMAN","powerSource":3}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2019-10-04T12:14:07: Received message from unsupported device with Zigbee model 'RC-EM'
zigbee2mqtt:warn  2019-10-04T12:14:07: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee-herdsman:unpi:parser --- parseNext [] +91ms
  zigbee-herdsman:controller:device Interview - got 'modelId,manufacturerName,powerSource' for device '0x000d6f00118c3ae1' +1s
  zigbee-herdsman:znp:SREQ --> AF - dataRequest - {"dstaddr":21123,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":2,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,3,0,0,0,1,0,2,0]}} +1s
[... cut ...]
  zigbee-herdsman:unpi:parser --- parseNext [] +9ms
  zigbee-herdsman:controller:error Interview failed for '0x000d6f00118c3ae1 with error 'Error: Timeout - 21123 - 1 - 3 - 1 after 10000ms' +0ms
zigbee2mqtt:error 2019-10-04T12:14:17: Failed to interview '0x000d6f00118c3ae1', device has not successfully been paired
zigbee2mqtt:info  2019-10-04T12:14:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000d6f00118c3ae1"}}'

Koenkk, 2019-10-04:
The log your provided is exactly what I need. Please try to replace the following file `node_modules/zigbee-herdsman/dist/controller/model/devices.js` with this one: https://pastebin.com/Cnukd5Zi and provide me the log again when interviewing fails.

victorsohk, 2019-10-05:
@Koenkk , Thanks a lot.  Here are some lines from the log after using the device.js file you provided (I found there is no 's' on the original file name so I assume device_s_.js is a typo or it doesn't matter.)

  zigbee-herdsman:controller:log Device joined '0x000d6f00118c3ae1' +52s
  zigbee-herdsman:controller:log New device '0x000d6f00118c3ae1' joined +2ms
  zigbee-herdsman:controller:log Creating device '0x000d6f00118c3ae1' +2ms
zigbee2mqtt:info  2019-10-05T04:12:54: Device '0x000d6f00118c3ae1' joined
zigbee2mqtt:info  2019-10-05T04:12:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f00118c3ae1"}}'
  zigbee-herdsman:controller:log Interview '0x000d6f00118c3ae1' start +68ms
zigbee2mqtt:info  2019-10-05T04:12:54: Starting interview of '0x000d6f00118c3ae1'
zigbee2mqtt:info  2019-10-05T04:12:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f00118c3ae1"}}'
  zigbee-herdsman:controller:device Interview - start device '0x000d6f00118c3ae1' +0ms
  zigbee-herdsman:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":58044,"nwkaddrofinterest":58044} +42s
  zigbee-herdsman:unpi:writer --> frame [254,4,37,2,188,226,188,226,35] +42s
.......
  zigbee-herdsman:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":58044,"nwkaddr":58044,"ieeeaddr":"0x000d6f00118c3ae1","capabilities":128} +467ms
  zigbee-herdsman:controller:log Device announce '0x000d6f00118c3ae1' +392ms
zigbee2mqtt:debug 2019-10-05T04:12:55: Device '0x000d6f00118c3ae1' announced itself
......
 zigbee-herdsman:znp:AREQ <-- ZDO - nodeDescRsp - {"srcaddr":58044,"status":0,"nwkaddr":58044,"logicaltype_cmplxdescavai_userdescavai":2,"apsflags_freqband":64,"maccapflags":128,"manufacturercode":4619,"maxbuffersize":82,"maxintransfersize":82,"servermask":0,"maxouttransfersize":82,"descriptorcap":0} +990ms
  zigbee-herdsman:unpi:parser --- parseNext [] +8ms
  zigbee-herdsman:controller:device Interview - got node descriptor for device '0x000d6f00118c3ae1' +1s
  zigbee-herdsman:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":58044,"nwkaddrofinterest":58044} +1s
  zigbee-herdsman:unpi:writer --> frame [254,4,37,5,188,226,188,226,36] +1s
......
  zigbee-herdsman:znp:AREQ <-- ZDO - activeEpRsp - {"srcaddr":58044,"status":0,"nwkaddr":58044,"activeepcount":1,"activeeplist":[1]} +1s
  zigbee-herdsman:unpi:parser --- parseNext [] +7ms
  zigbee-herdsman:controller:device Interview - got active endpoints for device '0x000d6f00118c3ae1' +1s
  zigbee-herdsman:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":58044,"nwkaddrofinterest":58044,"endpoint":1} +1s
  zigbee-herdsman:unpi:writer --> frame [254,5,37,4,188,226,188,226,1,37] +1s
......
  zigbee-herdsman:controller:device Interview - got simple descriptor for endpoint '1' device '0x000d6f00118c3ae1' +995ms
  zigbee-herdsman:controller:device Interview - ssIasZone enrolling '0x000d6f00118c3ae1' endpoint '1' +12ms
  zigbee-herdsman:znp:SREQ --> AF - dataRequest - {"dstaddr":58044,"destendpoint":1,"srcendpoint":1,"clusterid":1280,"transid":1,"options":0,"radius":30,"len":14,"data":{"type":"Buffer","data":[16,2,2,16,0,240,87,113,54,25,0,75,18,0]}} +1s
......
zigbee-herdsman:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":58044,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":15,"securityuse":0,"timestamp":1227050,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[24,2,4,0]}} +34ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":2,"manufacturerCode":null,"commandIdentifier":4},"Payload":[{"status":0}],"Cluster":{"ID":1280,"attributes":{"zoneState":{"ID":0,"type":48,"name":"zoneState"},"zoneType":{"ID":1,"type":49,"name":"zoneType"},"zoneStatus":{"ID":2,"type":25,"name":"zoneStatus"},"iasCieAddr":{"ID":16,"type":240,"name":"iasCieAddr"},"zoneId":{"ID":17,"type":32,"name":"zoneId"},"numZoneSensitivityLevelsSupported":{"ID":18,"type":32,"name":"numZoneSensitivityLevelsSupported"},"currentZoneSensitivityLevel":{"ID":19,"type":32,"name":"currentZoneSensitivityLevel"}},"name":"ssIasZone","commands":{"enrollRsp":{"ID":0,"parameters":[{"name":"enrollrspcode","type":32},{"name":"zoneid","type":32}],"name":"enrollRsp"},"initNormalOpMode":{"ID":1,"parameters":[],"name":"initNormalOpMode"},"initTestMode":{"ID":2,"parameters":[],"name":"initTestMode"}},"commandsResponse":{"statusChangeNotification":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"},"enrollReq":{"ID":1,"parameters":[{"name":"zonetype","type":33},{"name":"manucode","type":33}],"name":"enrollReq"}}}},"networkAddress":58044,"endpoint":1,"linkquality":15,"groupID":0}' +4s
  zigbee-herdsman:unpi:parser --- parseNext [] +49ms
......
zigbee-herdsman:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":58044,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":10,"securityuse":0,"timestamp":1233251,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[9,118,1,15,1,11,18]}} +2s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false},"transactionSequenceNumber":118,"manufacturerCode":null,"commandIdentifier" [... cut ...]
  zigbee-herdsman:znp:SREQ --> AF - dataRequest - {"dstaddr":58044,"destendpoint":1,"srcendpoint":1,"clusterid":1280,"transid":2,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,118,11,1,0]}} +3s
......
  zigbee-herdsman:controller:error Interview failed for '0x000d6f00118c3ae1 with error 'Error: Timeout - 58044 - 1 - 3 - 11 after 15000ms' +0ms
zigbee2mqtt:error 2019-10-05T04:13:18: Failed to interview '0x000d6f00118c3ae1', device has not successfully been paired
zigbee2mqtt:info  2019-10-05T04:13:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000d6f00118c3ae1"}}'

Koenkk, 2019-10-05:
- Can you provide the complete logging? So without skipping some parts
- Did you check if the button works now? Do you see any messages appear when pressing the buttons on it (this might work even when interview failed)

victorsohk, 2019-10-06:
I checked and all the buttons don't work, i.e. nothing happened when pressing any button.

Here is the full log after the permit join - https://pastebin.com/1HWGgkU6.   
(I pressed the pair hole for 5s to remove the pervious pairing first so you can see the device leaving network message before I rejoin the device)
Thanks!

Koenkk, 2019-10-07:
Can you try with https://pastebin.com/jKTWqLep (small updated version of: https://github.com/Koenkk/zigbee2mqtt/issues/1957#issuecomment-538489129)

victorsohk, 2019-10-08:
Thanks!  Here is the log after the update:  https://pastebin.com/N9bjUq6W.
Unfortunately it still doesn't work!

Koenkk, 2019-10-09:
Can you try with: https://pastebin.com/Nt5fY8rT

victorsohk, 2019-10-09:
Thanks again.  Here is the log after the update: https://pastebin.com/GtYW4e62.
Still no luck!

Koenkk, 2019-10-11:
Honestly im lost how to continue at this point, perhaps somebody owning this device and having the original gateway could dive further into this.

kijuz, 2019-10-11:
maybe the problem is related to cc2531 fw version?

Fabiancrg, 2019-10-11:
Personnaly I think their device is just not working anymore, I have already tried to sniff the traffic when trying to join and I just can't see anything when pressing the buttons. Most of the time even the device LED is not blinking as it should when reseting or trying to pair the remote. I don't have the Heiman gateway and I don't think I will buy any other Heiman product anymore.

victorsohk, 2019-10-13:
@Koenkk Thanks anyway.  I think I need to spend some time to learn the Zigbee pairing flow and come back to solve this issue in the future.

@kijuz Thanks for the reminder.  You are right that it's possible for the it to work on the previous fm version but I don't want to try it as it will break all my existing devices.

Fabiancrg, 2019-11-20:
@Koenkk  I built a new lab environment with a CC2652R1 board and tried to repair the HS1RC-M.
It's not working but I have some info in the log now and see the packet when I run a trace.

Here is the debug log: https://pastebin.com/wqbHh6Jg

I also have a trace done with Wireshark I can provide in .pcapng format, if a TXT export is OK, here is the first part of it: https://pastebin.com/5EcFtzzp

I can see there the association request and the response (successful).
Then each time I press a button, I can see it in the trace but nothing in the Z2M log execpt the error.

Note that I was pressing a button every second during the pairing process. 

Something wierd, when I initially added the device, the model ID was RC_V14, now it's seen as RC-EM.

Let me know if you need something else.

bambouzel, 2019-11-20:
@Fabiancrg I have a HS1RC-E smart remote which uses 'RC-EM' as the model.  I have added this to the existing HS1RC-M device and all seems to be working fine.  I have a pull-request to get this into the main trunk.
The pairing process of the heimann is keep pushing the button a few seconds to join and a bit longer to leave the network.

Fabiancrg, 2019-11-20:
@bambouzel  I have two remotes, they look the same. Initially I purchased one on which I have the model **HS1RC-E** (written on the device), it was seen as  RC_V14 (Model-ID), it worked with Z2M during a few months but it suddenly stopped working and now I cannot pair it anymore on Z2M using a CC2531 with ZHA 1.2 (Source routing).

Using CC2652R1 with Zigbee 3.0 it looks better but the device is now seen as RC-EM so I updated the devices.js to be able to pair it but it's not working, I have the following error:
````
Interview - failed for device '0x000d6f00104f517e' with error 'Error: Data request failed with error: 'MAC transaction expired' (240)
````

The second remote I have is a **HS1RC-M** and its model ID is also RC-EM, for this one the pairing is working but I cannot use the buttons.
Z2M log shows:
````
info  2019-11-20T15:51:05: Device '0x005043c953287154' joined
info  2019-11-20T15:51:05: MQTT publish: topic 'z2mlab/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x005043c953287154"}}'
info  2019-11-20T15:51:05: Starting interview of '0x005043c953287154'
info  2019-11-20T15:51:05: MQTT publish: topic 'z2mlab/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x005043c953287154"}}'
info  2019-11-20T15:51:14: Successfully interviewed '0x005043c953287154', device has successfully been paired
info  2019-11-20T15:51:14: Device '0x005043c953287154' is supported, identified as: HEIMAN Smart remote controller (HS1RC-M)
info  2019-11-20T15:51:14: MQTT publish: topic 'z2mlab/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x005043c953287154","model":"HS1RC-M","vendor":"HEIMAN","description":"Smart remote controller","supported":true}}'
````

Wireshark trace can be found here --> https://pastebin.com/6WuPTngT

Unfortunately even with pairing being successful the buttons are not generating any message in the trace so I suppose there are two issues, maybe some additional config must be done on Z2M to support this model.
Maybe there is an issue with HS1RC-E and Heiman released the fixed version HS1RC-M... no idea, this vendor is not showing any release notes and support has never replied to my questions regarding these two models. maybe @Koenkk can make sense and find a way to have both working using the trace from this post and the one from the previous post I made.

Fabiancrg, 2019-11-20:
@bambouzel sorry I did not read correctly your post, so you are saying that it's working for you.
To clarify the pairing and leaving process:
1. To pair, press the button (hole on the top) until the green LED flashes quickly then stays on 1 sec.
2. To remove, press the button (hole on the top) until the green LED flashes slowly 6 times
Leaving is working fine, in debug log I can see this:
```` 
zigbee-herdsman:controller:log Device leave '0x000d6f00104f517e' +3m
zigbee-herdsman:controller:log Removing device from database '0x000d6f00104f517e' +0ms
````

But when you pair it; when did you release the button (the one in the hole) ? As soon as the LED stop flashing ? Did you do something else during the pairing ? Press any other button regularly maybe ?

bambouzel, 2019-11-20:
@Fabiancrg The pairing process you describe is correct and I indeed do not push on the other buttons while pairing.
The Z2M logfile (not in debug):

info  2019-11-18T17:40:57: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
info  2019-11-18T17:40:57: Starting zigbee-herdsman...
info  2019-11-18T17:40:58: zigbee-herdsman started
info  2019-11-18T17:40:58: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}'
info  2019-11-18T17:40:58: Currently 10 devices are joined:
info  2019-11-18T17:40:58: 0x000d6f00104f246e (0x000d6f00104f246e): **HS1RC-M** - HEIMAN Smart remote controller (EndDevice)
...
Clicking on the buttons gives me:
info  2019-11-19T09:10:39: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":44.5,"linkquality":39,"action":"**arm_all_zones**"}'
info  2019-11-19T09:10:39: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":44.5,"linkquality":39,"action":""}'
info  2019-11-19T09:10:48: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":44.5,"linkquality":42,"action":"**arm_partial_zones**"}
'
info  2019-11-19T09:10:48: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":44.5,"linkquality":42,"action":""}'
info  2019-11-19T09:10:56: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":44.5,"linkquality":52,"action":"**disarm**"}'
info  2019-11-19T09:10:56: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":44.5,"linkquality":52,"action":""}'
info  2019-11-19T09:11:02: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":44.5,"linkquality":0,"action":"**emergency**"}'
info  2019-11-19T09:11:02: MQTT publish: topic 'zigbee2mqtt/0x000d6f00104f246e', payload '{"battery":44.5,"linkquality":0,"action":""}'

Fabiancrg, 2019-11-20:
@bambouzel Thank you, I will flash a CC2531 with the same firmware as yours and retry the pairing.

kijuz, 2019-11-20:
> @Fabiancrg The pairing process you describe is correct and I indeed do not push on the other buttons while pairing.
> The Z2M logfile (not in debug):
>
> info 2019-11-18T17:40:57: Starting zigbee2mqtt [...]

can you provide the firmware you are using? because i can't find it.

thanks

kijuz

bambouzel, 2019-11-20:
@kijuz found a link to it: [Z-Stack-firmware](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin)
(last in the list)

kijuz, 2019-11-20:
thanks!! i'll try it.

Fabiancrg, 2019-11-20:
For me it's still the same problem with the CC2531 using ZHA 1.2 (20190425)

````
zigbee2mqtt:info  2019-11-20T19:19:18: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-20.19-19-16'
zigbee2mqtt:info  2019-11-20T19:19:18: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
zigbee2mqtt:info  2019-11-20T19:19:18: Starting zigbee-herdsman...
zigbee2mqtt:info  2019-11-20T19:19:20: zigbee-herdsman started
zigbee2mqtt:info  2019-11-20T19:19:20: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}'
zigbee2mqtt:info  2019-11-20T19:19:20: Currently 0 devices are joined:
zigbee2mqtt:warn  2019-11-20T19:19:20: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2019-11-20T19:19:20: Allowing new devices to join.
zigbee2mqtt:warn  2019-11-20T19:19:20: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2019-11-20T19:19:20: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2019-11-20T19:19:20: Connecting to MQTT server at mqtt://192.168.1.32:1883
zigbee2mqtt:info  2019-11-20T19:19:20: Connected to MQTT server
zigbee2mqtt:info  2019-11-20T19:19:20: MQTT publish: topic 'z2mlab/bridge/state', payload 'online'
zigbee2mqtt:info  2019-11-20T19:19:20: MQTT publish: topic 'z2mlab/bridge/config', payload '{"version":"1.7.1","commit":"b459c35","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}},"log_level":"info","permit_join":true}'
zigbee2mqtt:info  2019-11-20T19:19:24: Device '0x000d6f00104f517e' joined
zigbee2mqtt:info  2019-11-20T19:19:24: MQTT publish: topic 'z2mlab/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f00104f517e"}}'
zigbee2mqtt:info  2019-11-20T19:19:24: Starting interview of '0x000d6f00104f517e'
zigbee2mqtt:info  2019-11-20T19:19:24: MQTT publish: topic 'z2mlab/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f00104f517e"}}'
zigbee2mqtt:error 2019-11-20T19:19:53: Failed to interview '0x000d6f00104f517e', device has not successfully been paired
zigbee2mqtt:info  2019-11-20T19:19:53: MQTT publish: topic 'z2mlab/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000d6f00104f517e"}}'
``

bambouzel, 2019-11-20:
Very weird, no other messages?
Have you already tried pairing in debugging mode?
[Starting in full debug mode](https://github.com/zigbeer/zigbee-shepherd/wiki#8-debug-messages)

Fabiancrg, 2019-11-20:
Yes, I did, the output and trace are in this post:
https://github.com/Koenkk/zigbee2mqtt/issues/1957#issuecomment-556013630

It was with another baord and FW but the message look like to be exactly the same.

What's strange is that Z2M report as 'Failed interview" but the remote seems to work as I can see the Zigbee packet in the Wireshark trace when I press each button.

Koenkk, 2019-11-20:
@Fabiancrg can you share the wireshark trace?

Fabiancrg, 2019-11-20:
@Koenkk Here is the trace (Encryption key is the defualt one)

[HS1RC-E-Zigbee3.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/3871308/HS1RC-E-Zigbee3.pcapng.gz)

bambouzel, 2019-11-21:
@Fabiancrg looking in the debug logging I notice:
zigbee2mqtt:debug 2019-11-20T13:18:00: **No converter available for 'HS1RC-E'** with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"RC-EM","manufacturerName":"HEIMAN","powerSource":3}'
In the devices.js I just added the 'RC-EM' to the existing 'HS1RC-M' definition.
How does your devices.js look like for HS1-RC ?

Fabiancrg, 2019-11-21:
@Koenkk I added this to my devices.js:
````
    {
        zigbeeModel: ['RC-EM'],
        model: 'HS1RC-E',
        vendor: 'HEIMAN',
        description: 'Smart remote controller',
        supports: 'action',
        fromZigbee: [
            fz.battery_200,
            fz.heiman_smart_controller_armmode, fz.heiman_smart_controller_emergency,
        ],
        toZigbee: [],
    },
````

bambouzel, 2019-11-21:
@Fabiancrg I also used to add a new description for the 'HS1RC-E' and it looked like:
    {
        zigbeeModel: ['RC-EM'],
        model: 'HS1RC-E',
        vendor: 'HEIMAN',
        description: 'Smart remote controller',
        supports: 'action',
        fromZigbee: [
            fz.battery_200, fz.ignore_power_change, fz.ignore_iaszone_change,
            fz.HS1RC_E_arm, fz.heiman_smart_controller_emergency,
        ],
        toZigbee: [],
    },
and I made a new converter in the **fromZigbee.js**:
    HS1RC_E_arm: {
        cid: 'ssIasAce',
        type: 'cmdArm',
        convert: (model, msg, publish, options) => {
            const action = msg.data.data['armmode'];
            delete msg.data.data['armmode'];
            const modeLookup = {
                0: 'unlock',
                1: 'home',
                3: 'lock',
            };
            return {action: modeLookup[action]};
        },
    },

But with the newest release I just added 'RC-EM' to the existing 'HS1RC-M' definition. Nothing else required and everything stills seems to function (after changing the actions in my automation)

    {
        zigbeeModel: ['RC_V14', 'RC-EM'],
        model: 'HS1RC-M',
        vendor: 'HEIMAN',
        description: 'Smart remote controller',
        supports: 'action',
        fromZigbee: [
            fz.battery_200,
            fz.heiman_smart_controller_armmode, fz.heiman_smart_controller_emergency,
        ],
        toZigbee: [],
    },

Koenkk, 2019-11-21:
@Fabiancrg can you try if it pair correctly with the following CC2531 firmware? (https://drive.google.com/file/d/1-M6Q7Nx3f5cMba91tqbf5Rmf7Qck_r_B/view)

Fabiancrg, 2019-11-22:
@Koenkk I flashed my CC2531 with this FW and retried, I am still having the same error.
Here are the logs:

[HS1RC-E-CC2531ZNP-PROD_20191028.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/3881004/HS1RC-E-CC2531ZNP-PROD_20191028.pcapng.gz)
[Z2M-DEBUG-CC2531ZNP-Prod_20191028.txt](https://github.com/Koenkk/zigbee2mqtt/files/3881008/Z2M-DEBUG-CC2531ZNP-Prod_20191028.txt)

Koenkk, 2019-11-22:
Can you try replacing `node_modules/zigbee-herdsman/dist/controller/model/devices.js` with https://gist.github.com/Koenkk/33b4b9de314401f1c559501e93962765

Fabiancrg, 2019-11-22:
I did it and retry to pair, now it worked, I have this:
````
zigbee2mqtt:info  2019-11-22T20:57:19: Successfully interviewed '0x000d6f00104f517e', device has successfully been paired
zigbee2mqtt:info  2019-11-22T20:57:19: Device '0x000d6f00104f517e' is supported, identified as: HEIMAN Smart remote controller (HS1RC-M or HSRRC-E)
zigbee2mqtt:info  2019-11-22T20:57:19: MQTT publish: topic 'z2mlab/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6f00104f517e","model":"HS1RC-M or HSRRC-E","vendor":"HEIMAN","description":"Smart remote controller","supported":true}}'
````

But there is still a problem, when I press the buttons, I can see in the Wireshark trace the corresponding messages but nothing in Z2M debug log.
And after a couple of minutes, nothing is logged by the sniffer anymore when I press on the buttons.

Here are the log and trace:

[HS1RC-E-CC2531ZNP-PROD_20191028.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/3881424/HS1RC-E-CC2531ZNP-PROD_20191028.pcapng.gz)

[Z2M-DEBUG-CC2531ZNP-Prod_20191028.txt](https://github.com/Koenkk/zigbee2mqtt/files/3881425/Z2M-DEBUG-CC2531ZNP-Prod_20191028.txt)

Koenkk, 2019-11-25:
It looks that the device is correctly enrolled now but the problem is that the device send the messages to endpoint 0 which is an invalid endpoint for these kind of messages (should be endpoint 1). 

I also don't know why it stops sending, currently out of ideas on how to continue with this.

Koenkk, 2020-01-16:
FYI I will remove this as a supported device for now until somebody gets it working again: https://github.com/Koenkk/zigbee2mqtt/issues/2750#issuecomment-575374888

kisseler, 2020-01-18:
I can succefully pair the device:
`{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x005043c9a3332271","model":"HS1RC-M","vendor":"HEIMAN","description":"Smart remote controller","supported":true}}`
But afterwards zigbee2mqtt does not report any key action. I would really appreciate if that device was working. Does anyone know an alternative?

Fabiancrg, 2020-01-18:
Me too, it can be paired but afterwards it does not send anything over the zigbee network, it's not Z2M.
I tried to sniff the communication and it just stops sending anything when buttons are pressed.

There is another supported remote from another vendor (Swann) but I haven't tried it, apprently it is sold only in UK.

closed 2020-03-26 by stale[bot]