Koenkk/zigbee2mqtt #1963: CC2531 not found by raspberry pi

TheHackbraten, 2019-09-13:
Hello there,
i just bought me a CC2531 stick with preinstalled firmware. After i set up my raspberrry i could not find the stick via `ls -l /dev/serial/by-id`. I tried it on an older raspbian image (i think it was strech) but also on the latest version.
I am not sure where the problem is but my misgiving is that there is the wrong firmware preinstalled on the CC2531.
Is there any way to check the firmware of the stick without having a CC Debugger?
Btw I see that the stick is connected when I use `lsusb`
I'm sorry if this is a dump question or problem but i couldn't find a solution in the faq or the other issues.
If i forgott any information you need to help me please ask, i am a noob to zigbee stuff and only familiar with the basics of raspberry/linux.
Thanks in advance!

Koenkk, 2019-09-14:
Do you see it under `ls /dev`? (also check the output when it is not plugged).

TheHackbraten, 2019-09-14:
This is my `ls /dev`:
[image]
I think `serial1` is not the same as the `serial` i am looking for and there is nothing alike `ttyACM`.
Also there is no difference in the output between cc2531 plugged or not plugged.
My next Idea would be to flash the stick to the latest firmware via raspberry as explained on your website. I guess that can't make it worse.

Koenkk, 2019-09-14:
If it doesnt show you either need to reflash it or the stick is broken.

I will close this as it is not directly a zigbee2mqtt bug.

closed 2019-09-14 by Koenkk