Koenkk/zigbee2mqtt #1964: Add support for AOK Curtain Motor

Jinvis, 2019-09-13:
Hi, 
I am an user of zigbee2mqtt, I have a AOK Curtain Motor(zigbee) in my house, and i want to add it to homeassistant.
I have captured some data with wireshark from ovribo gateway, and try to control it as xiaomi Curtain Motor, but it does not work, Can someone help me with this? Thanks very much.

report attributes pictures:
[image]

this is close data:
[image]

zigbee2mqtt log:
  zigbee2mqtt:warn 9/13/2019, 6:01:39 PM No converter available for 'AOK_win_motor' with cid 'closuresWindowCovering', type 'attReport' and data '{"cid":"closuresWindowCovering","data":{"currentPositionLiftPercentage":0}}'

Jinvis, 2019-09-13:
I try to control as xiaomi motor, it does not work:
return [{
                    cid: 'genAnalogOutput',
                    cmd: 'write',
                    cmdType: 'foundation',
                    zclData: [{
                        attrId: 0x0008,
                        dataType: 0x20,
                        attrData: value,
                    }],
                    cfg: cfg.default,
                }];

Koenkk, 2019-09-16:
I would recommend adding support for this device in the zigbee2mqtt dev branch (as together with zigbee-herdsman-converters this should be easier). Here it for example shows how to do a write: https://github.com/Koenkk/zigbee-shepherd-converters/blob/zigbee-herdsman-converters/converters/toZigbee.js#L601

closed 2019-11-22 by stale[bot]