Koenkk/zigbee2mqtt #1967: Xiaomi SJCGQ11LM Water Sensor - False Positives

coolguymatt, 2019-09-13:
# Bug Report
## What happened
After updating to zigbee2mqtt to 1.6, I have received two false positives on two of my Xiaomi water leak sensors. For the first one, I could not get it to go back to Dry (Off). Removing and rejoining the device did not fix the issue so I completely removed it from the database manually and rejoined it. This seems to have fixed the first one. After about another hour, the second device had a false positive. It has the same behaviour - it does not go back to Dry (Off) when the device is dry.

## What did you expect to happen
Accurate water leak readings

## How to reproduce it (minimal and precise)
Random

## Debug Info
zigbee2mqtt version: 1.6
CC253X firmware version:  CC2531_DEFAULT_20190608.zip

mpuff, 2019-09-13:
i also get a false "true" notification yesterday from a Xiaomi water leak sensor, i am also on zigbee2mqtt 1.6

groenmarsmannetje, 2019-09-13:
I also have experienced this today. Sensor is still reporting leakage, but is dry. Even after removing battery and reconnect. I updated to version 1.6 just before this sensor turned on.

ptvoinfo, 2019-09-14:
I've added this update to the latest version but my sensors work well. It seems we should an option (disable or enable) interval reports for these sensors. @Koenkk, what do think?

Koenkk, 2019-09-14:
Im wondering if the converter is actually correct @ptvoinfo how do you know these are the correct values?

ptvoinfo, 2019-09-15:
@Koenkk I've tested it with the real device and simulate water leaks. I have 2 sensors, both works identically.

groenmarsmannetje, 2019-09-15:
But why has it changed? It used to work without any problems. My sensor is still reporting leakage and has become totally useless now.

docBliny, 2019-09-15:
I started having the exact same issue. I currently have ~15 sensors going off causing me to get alerts throughout the day. I'm using zigbee2mqtt on Home Assistant. I see lines similar to the following in the Docker log:
```
zigbee2mqtt:info 9/15/2019, 12:08:06 PM MQTT publish: topic 'zigbee2mqtt/water_leak_1', payload '{"water_leak":true,"linkquality":2,"battery":100,"voltage":3065}
```

This started immediately after updating to 1.6.

ptvoinfo, 2019-09-16:
The quick temporary workaround is to edit the following file:
node_modules\zigbee-shepherd-converters\devices.js
Find
`    {
        zigbeeModel: ['lumi.sensor_wleak.aq1'],
        model: 'SJCGQ11LM',
        vendor: 'Xiaomi',
        description: 'Aqara water leak sensor',
        supports: 'water leak true/false',
        fromZigbee: [
            fz.xiaomi_battery_3v, fz.SJCGQ11LM_water_leak_iaszone,
            fz.SJCGQ11LM_water_leak_interval, fz.ignore_basic_change,
        ],
        toZigbee: [],
    },`
and remove `fz.SJCGQ11LM_water_leak_interval, `.

Koenkk, 2019-09-16:
Fixed in the dev branch (removed converter).

closed 2019-09-16 by Koenkk

ptvoinfo, 2019-09-17:
Can I add this converter as a device option? Because it works for some devices?

Koenkk, 2019-09-17:
Yes

gecata93, 2019-10-08:
This issue is marked as fixed(closed) but i still experience this problem, is the new version deployed  with this fix? And if yes? What i need to do, to get these changes? 

I'm using: Zigbee2mqtt Hass.io Add-on
Aqara water leak sensor (SJCGQ11LM)
Firmware: Zigbee2mqtt 1.6.0

Koenkk, 2019-10-09:
@gecata93 it is not in zigbee2mqtt but in the latest dev (edge for hassio)

gecata93, 2019-10-09:
@Koenkk  oh okay, thanks. When will be available for the stable(regular) version?

Koenkk, 2019-10-09:
@gecata93 in zigbee2mqtt 1.7.0 (no eta)

VivantSenior, 2019-10-09:
In my case simple trick helped. @gecata93, just try to simulate leak (connect two screws at the bottom of sensor) using wire or just put it in the cup with water. After few seconds break the wire circuit / remove sensor from the cup. Now the sensor should detect real leakage.

apeeters, 2020-02-05:
I still experience this in 1.9.0 :(

docBliny, 2020-02-05:
> I still experience this in 1.9.0 :(

I had that too on some sensors, but then found out that the battery was dead in those... Worth a check.

groenmarsmannetje, 2020-02-06:
I only experience this after a restart of Home Assistant. I always have to restart Zigbee2Mqtt otherwise one water leak sensor will report that it detects water.

apeeters, 2020-03-05:
I triggered a wet and dry cycle once and seems stable now. I will keep a close look at it for a few more days. Didn't replace the battery.

apeeters, 2020-03-26:
As promised I report back: It is perfectly stable now: no more false positives.

bbrendon, 2022-04-09:
Apparently, my house is flooding right now. All 3 of my sensors are triggered. The place is dry.
It's broken again. version 1.25.0

mpuff, 2022-04-09:
I had the same yesterday on 2 of 3 sensors, also on 1.25.0

Koenkk, 2022-04-09:
See https://github.com/Koenkk/zigbee2mqtt/issues/12018