Koenkk/zigbee2mqtt #1970: Error update

papaeng89, 2019-09-14:
Hello,

I want update to 1.5.1 to 1.6.0 when "npm install" i have error

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: The value "false" is invalid for option "readableHighWaterMark"

npm ERR! code ERR_INVALID_OPT_VALUE
npm ERR! The value "false" is invalid for option "readableHighWaterMark"

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-14T09_30_40_403Z-debug.log

On Rasperry Zero

Koenkk, 2019-09-14:
What is your nodejs version?

papaeng89, 2019-09-14:
nodejs v10.16
npm 6.9.0

Koenkk, 2019-09-16:
This is an npm issue (https://stackoverflow.com/questions/49554848/how-to-solve-npm-err-code-err-invalid-opt-value). 

You may try to upgrade npm to the latest version (`sudo npm install -g npm`) and try again.

I will close this issue as it is not a zigbee2mqtt issue.

closed 2019-09-16 by Koenkk