Koenkk/zigbee2mqtt #1972: Waht is the meaning of not removing the ' ' in 'friendly_name'

alwashe, 2019-09-14:
In the past after editing configuration.yaml the devices e.x. '0x00158d0003f1cecb' to 'friendly_name' the ' ' where removed after a restart.

And in the past if the friendly_name was double it did not work. I had to delete the device in database and configuration.yaml and repair.

Now it happend again and nothing is working out. The devices are not working reliable and the ' ' are not removed. I tried to repair several times.

Only the devices where ' ' is not removed are not working reliable.

Koenkk, 2019-09-14:
Can you post your configutation.yaml?

alwashe, 2019-09-14:
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.178.204:1883'
  user: openhab
  password: 'xxxxxx'
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d0003468292':
    friendly_name: tür_1
    retain: false
  '0x00158d0003978c82':
    friendly_name: tür_2
    retain: false
  '0x00158d0003f1cecb':
    friendly_name: tür_3
    retain: false
  '0x00158d0003f1cf7f':
    friendly_name: tür_4
    retain: false
  '0x00158d000395638f':
    friendly_name: tür_5
    retain: false
  '0x00158d0002c15812':
    friendly_name: tür_6
    retain: false
  '0x00158d0002c15715':
    friendly_name: tür_7
    retain: false
  '0x00158d00035c0beb':
    friendly_name: tür_8
    retain: false
  '0x00158d00035c0842':
    friendly_name: tür_9
    retain: false
  '0x00158d000359d427':
    friendly_name: tür_10
    retain: false
  '0x00158d00034d8166':
    friendly_name: tür_11
    retain: false
  '0x00158d00032bd5b5':
    friendly_name: bewegung_1
    retain: false
  '0x00158d000302ec66':
    friendly_name: bewegung_2
    retain: false
  '0x00158d0002f3ca9a':
    friendly_name: bewegung_3
    retain: false
  '0x00158d0002fbd556':
    friendly_name: bewegung_4
    retain: false
  '0x00158d00032b71d2':
    friendly_name: bewegung_5
    retain: false
  '0x000d6ffffe41491f':
    friendly_name: bewegung_6
    retain: false
  '0x000d6ffffe4594fe':
    friendly_name: bewegung_7
    retain: false
  '0x00178801067232c9':
    friendly_name: hue_1
    retain: false
  '0x0017880106723191':
    friendly_name: hue_2
    retain: false
  '0x00158d0002b02be2':
    friendly_name: schalter_1
    retain: false
  '0x00158d0002b034c1':
    friendly_name: schalter_2
    retain: false
  '0x00158d0002c761fc':
    friendly_name: schalter_3
    retain: false
  '0x00158d00023e79bc':
    friendly_name: schalter_4
    retain: false
  '0x00158d000323dabb':
    friendly_name: schalter_6
    retain: false
  '0x00158d000323f9aa':
    friendly_name: schalter_7
    retain: false
  '0x00158d00032437d9':
    friendly_name: schalter_8
    retain: false
  '0x00158d0002f8ebf0':
    friendly_name: schalter_9
    retain: false
  '0x000d6ffffe3afcd6':
    friendly_name: tradfri_1
    retain: false
  '0x14b457fffe4f4583':
    friendly_name: tradfri_2
    retain: false
  '0x14b457fffe506e20':
    friendly_name: tradfri_3
    retain: false
  '0x0017880104a50a8f':
    friendly_name: esstisch_deckenlampe
    retain: false
  '0x000d6ffffe3f5381':
    friendly_name: sara_deckenlampe
    retain: false
  '0x00158d00036b7e3e':
    friendly_name: aussen_thermometer
    retain: false
  '0x00158d000358e2f2':
    friendly_name: kuehlschrank_temperatur
    retain: false
  '0x00158d00039269f8':
    friendly_name: temp_3
    retain: false
  '0x00158d0003588c81':
    friendly_name: temp_4
    retain: false
  '0x00158d000395b6a7':
    friendly_name: temp_5
    retain: false
  '0x00158d0003cbbf28':
    friendly_name: temp_6
    retain: false
  '0x00124b0018e24aaf':
    friendly_name: router_1
    retain: false
  '0x00124b001bd8199e':
    friendly_name: router_2
    retain: false
  '0x00124b001b7d96fd':
    friendly_name: router_3
    retain: false
  '0x00124b001b7d96df':
    friendly_name: router_4
    retain: false
  '0x00158d000357cd0f':
    friendly_name: water_1
    retain: false
  '0x00158d00031369ca':
    friendly_name: vibration_1
    retain: false
  '0x7cb03eaa00b26aef':
    friendly_name: steckdose_1
    retain: false
  '0x0017880104a529ba':
    friendly_name: 'hue_3_og_schreibtisch'
    retain: false
  '0x0017880104ca51be':
    friendly_name: 'hue_4_og_stehlampe'
    retain: false
  '0x0017880104ca5260':
    friendly_name: 'hue_5_og_bettlampe'
    retain: false
  '0x00124b001bd80d8b':
    friendly_name: 'router_5'
    retain: false
```

Koenkk, 2019-09-16:
I don't understand what the problem is, could you provide more information and steps on how to reproduce it?

alwashe, 2019-09-17:
i don't know what happend, but the apostrophes disapeared over night and now everything is woking.
Bevore it worked: if I turn on a light it sometimes didn't turn on, sometimes it turned on but there was no status update. sometimes i was not able to turn a light off.

closed 2019-11-23 by stale[bot]