Koenkk/zigbee2mqtt #1973: UnhandledPromiseRejectionWarning

ccorderod, 2019-09-14:
# Bug Report 
## What happened
UnhandledPromiseRejectionWarning TypeError: Cannot read property '0x000d6f0004d378e8' of null.
Z2M does not crash.
## What did you expect to happen

## How to reproduce it (minimal and precise)
Test bed running locally on a pi4 with lastest z2m dev build and only 1 device joined (attached database content).
Each time the contact door is opened or closed, z2m throw the error above.
Debug Info: https://pastebin.com/u9QQ98QR
zigbee2mqtt version: latest dev build
[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/3612660/database.txt)
CC253X firmware version: latest source routing firmware for cc2531

bochelork, 2019-09-14:
Still have the same error, Here it comes when a previously paired device which have been removed from database.db, sends a status update.

Koenkk, 2019-09-16:
@ccorderod your issue should be fixed in latest dev

@bochelork can you provide the log of yours?

closed 2019-11-15 by Koenkk