Koenkk/zigbee2mqtt #1975: Errors when adding "zigbee_shepherd_devices": true in hassio

TurboTronix, 2019-09-14:
My setup used to work but now for some reason it gives errors and none of my zigbee devices that are on devices.js work. I use a CC2531 with Rpi running latest Hassio and 1.6.0 zigbee2mqtt.

I noticed the errors are gone after removing:   "zigbee_shepherd_devices": true from my config.

The errors are:

```
> node index.js
/zigbee2mqtt-1.6.0/lib/util/utils.js:44
        .map((mappedModel) => mappedModel.zigbeeModel));
                                          ^
TypeError: Cannot read property 'zigbeeModel' of undefined
    at flatten.map.map (/zigbee2mqtt-1.6.0/lib/util/utils.js:44:43)
    at Array.map (<anonymous>)
    at Object.<anonymous> (/zigbee2mqtt-1.6.0/lib/util/utils.js:44:10)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR!
 Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-09-14T17_31_48_646Z-debug.log
```

TurboTronix, 2019-09-14:
It also goes offline when that line of code is added to config.

Koenkk, 2019-09-14:
Why do you want to add that? Do you want to add support for a new device?

TurboTronix, 2019-09-14:
Yes I have wall switches that require that in order to work.

Koenkk, 2019-09-15:
This device already seems to be supported in zigbee2mqtt 1.6 (https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L4384) so I expect you don't need it anymore.

closed 2019-11-21 by stale[bot]