Koenkk/zigbee2mqtt #1980: Upgrade 1.3.0 > 1.6.0 fails with firmware 20190109

tubededentifrice, 2019-09-15:
#1787 Bug Report
## What happened
After upgrading from 1.3.0 to 1.6.0, zigbee2mqtt wouldn't start (with `./upgrade.sh`).
Logs can be found at https://pastebin.com/Q2RuWwJt 
CC2531 is on firmware 20190109 and I had it running for months perfectly fine.

After downgrading to 1.3.0 with `git checkout 00ebd44; rm -rf node_modules; npm install`, everything starts just fine again: https://pastebin.com/2JUKP6Hm

I haven't tried to re-flash the CC2531 while on 1.6.0, so it could really be an incompatibility issue beweek 20190109 firmware and 1.6.0. However I haven't seen any such incompatibility disclosed anywhere on the releases page.

## What did you expect to happen
Painless, in-place upgrade

## How to reproduce it (minimal and precise)
- Get a CC2531 on 20190109
- Check that everything works fine on 1.3.0
- run `./upgrade.sh` to go to latest version
- Check that it doesn't start anymore

## Debug Info
zigbee2mqtt version: 1.3.0 to 1.6.0
CC253X firmware version: 20190109

tubededentifrice, 2019-09-15:
NB: Problem persists with CC2531 on 20190608 (tried on another CC I had though): https://pastebin.com/vcMeQKwN

v1.3.0 is working properly on that CC2531 with 20190608

str3000, 2019-09-15:
Hello,
only additional info because I had the same problem.
I go back to V1.5.1 and zigbee2mqtt starts now. Puhhh

Differents: 
- CC2531: CC2531_SOURCE_ROUTING_20190619
- nuc i3

[2019_09_15_V1.6-logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/3613853/2019_09_15_V1.6-logs.txt)

Koenkk, 2019-09-15:
@tubededentifrice can you provide the log when starting with `DEBUG=* npm start`

tubededentifrice, 2019-09-15:
@Koenkk sure thing, here is one: https://pastebin.com/Mc4gEZ8v

tubededentifrice, 2019-09-15:
@str3000 I've tried 1.5.1 (commit #ac3b924) but I've the same problems on my install

tubededentifrice, 2019-09-15:
Other tests:
1.3.0 => 1.3.1: Works fine
1.3.1 => 1.4.0: Works fine
1.4.0 => 1.5.0: Works fine
1.5.0 => 1.5.1: Doesn't work
So it looks like the bug was introduced in 1.5.1

str3000, 2019-09-15:
ac3b924 works in my environment. 
It seems that everything works fine.
mosquitto version 1.5.8

tubededentifrice, 2019-09-19:
@Koenkk indeed, the dev branch seems to be working fine, thanks a lot :) when is it "scheduled" to be GA?

Koenkk, 2019-09-19:
No ETA, first extensive testing needs to be done (as most of the zigbee2mqtt code base has been replaced/touched)

closed 2019-09-19 by Koenkk

jamesawebb, 2019-09-19:
FWIW confirmed that a 1.4.0 restore fixes this issue for me.
/ht @tubededentifrice

negmos, 2019-09-30:
I have the same problem and falling back to 1.4.0 or Dev version doesn't solve it.
My system is:
CC2531: CC2531_SOURCE_ROUTING_20190619
Device: PI3B
Using Docker container

thehaxxa, 2019-10-02:
I am also having this issue.

andreas-bell, 2019-10-11:
I also had the issue when upgrading from 1.3.1 to 1.6.0:
zigbee2mqtt did not start, rebooted, unplugged, hit reset, copied the config again, tried the previous firmware, the default 20190608 and the source routing 20190619 firmware, nothing worked with 1.6.0 (different errors).
I then upgraded to the dev branch: (in combination with the source routing 20190619 firmware)
payload '{"version":"1.6.0","commit":"ba92f73"
and then for the first time since the upgrade zigbee2mqtt started again and is processing messages.
I'm not sure if I should stay on the dev branch with my production system, or if I could downgrade to 1.6.0 again.

Koenkk, 2019-10-11:
Dont worry on staying on the dev branch, this uses zigbee-herdsman (#1888) which is much better than zigbee-shepherd (zigbee2mqtt 1.6). Also this will be close to what zigbee2mqtt 1.7 will be.