Koenkk/zigbee2mqtt #1982: [dev] Xiaomi Aqara failing if `no_occupancy_since` is set

juanjux, 2019-09-16:
# Bug Report
## What happened

On the current dev version, Xiaomi Aqara motion sensors fail with the traceback posted below if the `no_occupancy_since` option is set. They work again if the option is removed.

## What did you expect to happen

They should work.

## How to reproduce it (minimal and precise)

## Debug Info

```javascript
zigbee2mqtt:debug 9/16/2019, 12:15:00 AM Received Zigbee message from 'RecibidorMovimiento' of type 'attributeReport' with data '{"occupancy":1}' from endpoint 1 with groupID 0
  zigbee2mqtt:error 9/16/2019, 12:15:00 AM Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: options.no_occupancy_since.forEach is not a function
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:634:44)
    at converters.forEach (/app/lib/extension/deviceReceive.js:148:41)
    at Array.forEach (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:146:20)
    at Controller.callExtensionMethod (/app/lib/controller.js:310:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:221:14))
```
zigbee2mqtt version: dev as of 20190916.

CC253X firmware version: 20190619

Koenkk, 2019-09-16:
Is `no_occupancy_since` in your `configuration.yaml` an array like documented here: https://www.zigbee2mqtt.io/devices/RTCGQ01LM.html#device-type-specific-configuration?

juanjux, 2019-09-17:
Yes, that was it, sorry and thanks!

closed 2019-09-17 by juanjux