Koenkk/zigbee2mqtt #1984: Any tips for getting IKEA ICPSHC24-10EU-IL-1 to pair?

coolguymatt, 2019-09-17:
# Bug Report
## What happened
IKEA ICPSHC24-10EU-IL-1 fails to pair
## What did you expect to happen
IKEA ICPSHC24-10EU-IL-1 to pair
## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: latest
CC253X firmware version: latest

I am having a tough time getting the Ikea LED driver to pair. When I press the reset button nothing happens. Sometimes if I unplug and plug it back in, it will display this twice in a row:

Cannot get the Node Descriptor of the Device: 0x90fd9ffffe35d582 (Error: Timed out after 10000 ms)

I have the LED driver within 5 cm of the stick.

Does anyone have any tips on how to get it to connect? I looked at old issues and one person had to reflash the firmware, but I am scared of messing up my existing Xiaomi and Ikea devices.

alwashe, 2019-09-17:
Try pressing reset 5 times.

coolguymatt, 2019-09-17:
@alwashe Just tried, did not work :(

coolguymatt, 2019-09-17:
It seems like only unplug/plug will start the pairing. The reset button doesn't do anything. I might exchange for a different one in case this unit is not working.

coolguymatt, 2019-09-17:
dumbguymatt should be my username. I have purchased a ICPSLC24-10NA-IL-1 instead of a ICPSHC24-10EU-IL-1. The ICPSLC24 requires you to pair the driver with a "steering device" - one of their remotes. I am having troubles connecting the steering device to zjigbee2mqtt but I am at least on the right track.

closed 2019-11-23 by stale[bot]