Koenkk/zigbee2mqtt #1985: RangeError: Index out of range with network map

bochelork, 2019-09-17:
Bug Report

## What happened

```
  zigbee-herdsman:unpi:parser <-- [254,5,69,178,90,23,132,0,50,9] +49ms
  zigbee-herdsman:unpi:parser --- parseNext [254,5,69,178,90,23,132,0,50,9] +0ms
  zigbee-herdsman:unpi:parser --> parsed 5 - 2 - 5 - 178 - [Z2] - 9 +0ms
  zigbee-herdsman:znp:error Error while parsing to ZpiObject 'RangeError: Index out of range
  zigbee-herdsman:znp:error     at checkOffset (buffer.js:977:11)
  zigbee-herdsman:znp:error     at Buffer.readUInt8 (buffer.js:1015:5)
  zigbee-herdsman:znp:error     at BuffaloZnp.readUInt8 (/volumeUSB1/usbshare/zigbee2mqtt/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:34:35)
  zigbee-herdsman:znp:error     at BuffaloZnp.read (/volumeUSB1/usbshare/zigbee2mqtt/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:264:25)
  zigbee-herdsman:znp:error     at BuffaloZnp.read (/volumeUSB1/usbshare/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/buffaloZnp.js:102:26)
  zigbee-herdsman:znp:error     at Function.readParameters (/volumeUSB1/usbshare/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/zpiObject.js:79:46)
  zigbee-herdsman:znp:error     at Function.fromUnpiFrame (/volumeUSB1/usbshare/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/zpiObject.js:52:30)
  zigbee-herdsman:znp:error     at Znp.onUnpiParsed (/volumeUSB1/usbshare/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:69:48)
  zigbee-herdsman:znp:error     at emitOne (events.js:116:13)
  zigbee-herdsman:znp:error     at Parser.emit (events.js:211:7)' +5m
  zigbee-herdsman:unpi:parser --- parseNext [] +2ms

```

## What did you expect to happen

No errors at all! :)

## How to reproduce it (minimal and precise)

Mqtt message  graphviz to zigbee2mqtt/bridge/networkmap/routes

## Debug Info
zigbee2mqtt version: version 1.6.0 (commit #d24c41c)
CC253X firmware version: {"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190315}},"log_level":"debug","permit_join":true}

Full log at
https://pastebin.com/2CtJU9uj

Koenkk, 2019-09-18:
Can you sniff this message with wireshark? It looks that it is malformed and thus the parsing error is correct. Thus I'm wondering what wireshark gives, perhaps I misunderstood the API. (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

bochelork, 2019-09-18:
Here's some wireshark data during networkmap/routes. Tnx.
[networkmaproutes.zip](https://github.com/Koenkk/zigbee2mqtt/files/3628280/networkmaproutes.zip)

Koenkk, 2019-09-20:
Which message should I look at? Can you point me to the problematic one?

bochelork, 2019-09-22:
If you think the parsing error is correct you can close the issue. I was just hoping that you can find something which explains my network stability problems.

closed 2019-11-28 by stale[bot]