Koenkk/zigbee2mqtt #1986: Unstable Network / lost messages / Devices dropping out of the network

odx, 2019-09-17:
# Bug Report
## What happened
Messages are not delivered reliably. Sometimes with delay, sometimes they are completely lost. Sometimes devices appear even to drop out of the network completely.

## What did you expect to happen
Messages should have been transmitted correctly and in a timely manner.

## How to reproduce it (minimal and precise)
I am not sure if it is related, but this started to happen after I added a second Osram Smart+ Plug to the network. 

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190619 source routing

odx, 2019-09-17:
Here is the graphviz network map... What alerts me is that the two sensors that were irresponsive right before I captured this is are "pir2" and "terrassentuer_kueche". Both sensors are very close to "plug2" which is a Osram Smart+ Plug acting as router. "plug2" does not have any children according to that network map...

```
digraph G { node[shape=record]; "0x00124b001bb8620b" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{router2|0x0 |No model information available|online (2019-09-17T16:08:39.189Z)}"]; "0x00124b001bb8620b" -> "0x000b57fffe900c86" [penwidth=0.5, weight=0, color="#994444", label="80"] "0x00124b001bb8620b" -> "0x000b57fffe976159" [penwidth=0.5, weight=0, color="#994444", label="62"] "0x00124b001bb8620b" -> "0x000d6ffffe24cb50" [penwidth=0.5, weight=0, color="#994444", label="138"] "0x00124b001bb8620b" -> "0x7cb03eaa0a06a960" [penwidth=0.5, weight=0, color="#994444", label="24"] "0x00124b001bb8e9e9" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{router3|0xa310 no lqi|Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)|offline (unknown)}"]; "0x00124b001bb8620b" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{router2|0x0 |No model information available|online (2019-09-17T16:08:39.193Z)}"]; "0x00124b001bb8620b" -> "0x000b57fffe900c86" [penwidth=0.5, weight=0, color="#994444", label="80"] "0x00124b001bb8620b" -> "0x000b57fffe976159" [penwidth=0.5, weight=0, color="#994444", label="62"] "0x00124b001bb8620b" -> "0x000d6ffffe24cb50" [penwidth=0.5, weight=0, color="#994444", label="138"] "0x00124b001bb8620b" -> "0x7cb03eaa0a06a960" [penwidth=0.5, weight=0, color="#994444", label="24"] "0x00158d0002d48919" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{giebelfenster_rechts|0x9d9c |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T15:55:37.830Z)}"]; "0x00158d0002d481e9" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{giebelfenster_links|0x102b |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T15:55:41.289Z)}"]; "0x00158d0002d481e9" -> "0x000b57fffe976159" [penwidth=1, weight=0, color="#994444", label="255"] "0x00158d0002e20d06" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{dachfenster_galerie|0xf460 |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T15:55:27.456Z)}"]; "0x00158d0002e20d06" -> "0x000d6ffffe7aca4b" [penwidth=1, weight=0, color="#994444", label="214"] "0x00158d0002e25588" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{badfenster|0xdebc |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T15:59:25.558Z)}"]; "0x00158d0002ca2d76" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{dachfenster_foo|0x4b1f |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T15:59:35.084Z)}"]; "0x00158d0002ca2d76" -> "0x000d6ffffe7aca4b" [penwidth=1, weight=0, color="#994444", label="147"] "0x00158d000273e1a3" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{balkontuer|0x4c17 |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T15:52:12.747Z)}"]; "0x00158d000273e1a3" -> "0x000d6ffffe24cb50" [penwidth=1, weight=0, color="#994444", label="255"] "0x00158d0002d489e2" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{wc-fenster|0x56b3 |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T15:47:42.038Z)}"]; "0x00158d0002c6371f" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{pir2|0x4c2e |Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online (2019-09-17T16:05:17.444Z)}"]; "0x00158d000273e216" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{terrassentuer_kueche|0xd195 |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T16:06:31.758Z)}"]; "0x00158d0002d481c3" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{terrassentuer_rechts|0x441e |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T16:06:39.462Z)}"]; "0x00158d0002d481c3" -> "0x000b57fffe900c86" [penwidth=1, weight=0, color="#994444", label="255"] "0x00158d0002d48263" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{terrassentuer_links|0x9ae6 |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T16:06:45.171Z)}"]; "0x00158d0002e20b98" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{wohnzimmerfenster_rechts|0xc6e9 |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T16:06:51.800Z)}"]; "0x00158d0002e20b98" -> "0x000d6ffffe7aca4b" [penwidth=1, weight=0, color="#994444", label="85"] "0x00158d0002d489fb" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{wohnzimmerfenster_links|0xf2f1 |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T15:48:12.545Z)}"]; "0x00158d0002d48259" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{fenster_abstellraum|0xf448 |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (2019-09-17T15:47:52.811Z)}"]; "0x00158d000273e2 [... cut ...] ", fillcolor="#fff8ce", fontcolor="#000000", label="{pir1|0x35fa |Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online (2019-09-17T15:57:05.013Z)}"]; "0x00158d00027315e8" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{temp1|0x2d50 |Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online (2019-09-17T16:03:15.690Z)}"]; "0x00158d00027315e8" -> "0x000b57fffe900c86" [penwidth=1, weight=0, color="#994444", label="255"] }
```

james-fry, 2019-09-18:
Your map looks great!
For me almost none of the leaf nodes (end devices/non routers) are shown as connected.
They (mostly) all work, but not showing in the diagram.
My problem is that for the rare but annoying times when there is a problem, the lack of consistency in the map makes it v hard to diagnose and fix the issues.

Koenkk, 2019-09-19:
It is known that OSRAM devices kick other our of the network, to confirm, can you check if everything works well when you dont have the osram devices in the network?

odx, 2019-09-20:
@james-fry i noticed that whenever there is a CC2530 as router in the network the end devices are nearly never connected in the graph. after removing the CC2530 (this was when I added the orsam plugs to take over the routing responsibility) the graph started to connect the end devices.

@Koenkk Network seems to have stabilized since two days now without removing the OSRAM plugs (I need them as routers. Otherwise I will have to add the CC2530 as router again) I will observe for further abnormalities and try removing the plugs if this re-occurs.

closed 2019-09-20 by odx

james-fry, 2019-09-20:
> @james-fry i noticed that whenever there is a CC2530 as router in the network the end devices are nearly never connected in the graph. after removing the CC2530 (this was when I added the orsam plugs to take over the routing responsibility) the graph started to connect the end devices.

Interesting theory. I will try removing the last cc2531 router I have.
Just hope that the other routers I have will give me enough coverage :)
Although I appreciate the author for sharing his CC2531 router firmware. its a real shame that the source is not available :(

odx, 2019-10-01:
I narrowed this down to interference with my WiFi. The coordinator was mounted directly next to my WiFi access point. I noticed enormous trouble when I was streaming media content to my TV. When I moved the coordinator about 50 cm away from the access point everything went back to normal...