Koenkk/zigbee2mqtt #1987: 1.6.0 not working on Ubuntu 18.04 LTS

untitledlt, 2019-09-17:
# Bug Report
## What happened
v1.5.1 worked fine but after upgrade to v1.6.0 it stopped working.
I did `rm -rf ./node_module; npm i; npm start`.

Log with `DEBUG=*` is here: https://pastebin.com/ahmmGrHa

I downgraded to 1.5.1, did clean install of dependencies and everything works fine again.

## Debug Info
zigbee2mqtt version: 1.6.1
CC2531 firmware version: `not sure how to check?`

Koenkk, 2019-09-20:
Can you provide the same logs when running with zigbee2mqtt 1.5.1?

untitledlt, 2019-09-23:
Here is a log for v1.6.1:
https://pastebin.com/mhW9A3d9

Koenkk, 2019-09-23:
In 1.5.1 I see the same error as in your 1.6 log; the difference is that 1.5.1 suppresses while 1.6.0 doesn't.

What firmware version is flashed on your stick? Could you try updating to the latest one? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing After this zigbee2mqtt 1.6.0 will probably start fine.

Fabiancrg, 2019-09-28:
I am running Z2M on  Ubuntu 18.04.3 LTS since version 1.3, now I am on 1.6.0 and it's still working without any issue.

closed 2019-12-04 by stale[bot]