Koenkk/zigbee2mqtt #1989: Last_seen in devices/get not in format from configuration.yaml

bochelork, 2019-09-17:
## What happened

Last seen after mqtt request "zigbee2mqtt/bridge/config/devices/get" is:

  "lastSeen":1568733798835

## What did you expect to happen

 "last_seen":"2019-09-17T20:03:23.125Z"

## Debug Info
zigbee2mqtt version: latest dev
CC253X firmware version: latest source routing.

Koenkk, 2019-09-18:
I'm wondering if this should follow the last seen configured by the user. It is meant to be used by external applications and it can be tricky if the format varies per setup. 

@carldebilly initially requested this feature, what do you think?

bochelork, 2019-09-18:
Before I used the state.json file te get the last_seen time.
The last seen in state.json seems a bit obsolete since it's not valid anymore. But it used to be a good source because it was remember between start ups.
The last_seen from the herdsman is updated when zigbee2mqtt is restarted so we can't see if a device is lost before restarting.

carldebilly, 2019-11-01:
@Koenkk  I really don't care... both will work with Z2MA.

But I agree with you, it's not designed to be consumed by a human.

(sorry for the delay to answer, this mention got silenced in my octobox)

closed 2019-11-24 by stale[bot]