Koenkk/zigbee2mqtt #1991: 1.6.0 Herdsman with Z-Stack 3.0 on CC2531

wixoff, 2019-09-18:
Overall, this seems like a very good combination - my system is SOLID for the first time in months.  No drop-offs (knock on wood), and responsiveness is through and FAST.

I am having a couple of issues, though, with several devices .  Hopefully 1.6 + Z-Stack 3.0 won't mess too much up,.

(1) The Tradfri Control Outlet just won't join. database.db picks up some information when going into jointing mode, but it is not nearly enough to complete the interview, so it doesn't work.

(2) I can't pair my MCT-340E door sensors at all,.  They dont even show up in database.db after putting them (and Z2M) into join nmode.

(3) I also can't pair Cree Connected bulbs.  It looks like a complete interview is present in database,db. but modelId isn't present (and adding manually it doesn't help).  Who knows what else is wrong

(4) Finally, In  Herdsman-Covernters, many items previously had timing set up for reporting and min/max values, etc.  Many of those are gone, and I see my Peanut Plug updaing current/power etc. VERY frequently - once every send or two.  Is this something I shoud resubmit as a PR to that branch?  There are a LOT of messages flying around in my system.

I can provide Z-H joining logs and Z2M usage logs over the next few days.

Koenkk, 2019-09-20:
Is this with zigbee2mqtt 1.6 or zigbee2mqtt 1.6-dev? 

If it is with dev please provide the `DEBUG=* npm start` log for the various issues.

wixoff, 2019-09-20:
It's with dev.  I'll provide the logs, etc. -- haven't had a chance yet but the weekend is nearly here.

closed 2019-11-26 by stale[bot]