Koenkk/zigbee2mqtt #1992: New Install on a new Pi, with copy database.db and configuration.yaml  PM Error while starting zigbee-shepherd, attempting to fix

jossie67, 2019-09-18:
I did a new fresh install on a Pi3 , I did this other installation did started after an upgrade
So i took a Pi3, with a fresh SD with "2019-07-10-raspbian-buster-lite"

I followed the install on https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

Before I did number 4, I copied the database.db and configuration.yaml in to the folder /opt/zigbee2mqtt/data.
The owner is still pi!
Then I started 

```
> cd /opt/zigbee2mqtt
> npm start
```
```

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 9/18/2019, 5:08:41 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-09-18.17-08-39'
  zigbee2mqtt:info 9/18/2019, 5:08:41 PM Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
  zigbee2mqtt:info 9/18/2019, 5:08:41 PM Starting zigbee-shepherd
  zigbee2mqtt:info 9/18/2019, 5:08:54 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: SRSP - SYS - osalNvRead after 6000ms)

```
My USB is there

```
ls -l /dev/serial/by-id

lrwxrwxrwx 1 root root 13 Sep 18 16:48 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018ED3A46-if00 -> ../../ttyACM0
```

lolorc, 2019-09-18:
is /dev/ttyACM0 usable by pi ? it should have rw on it as user or as a group pi is member of

jossie67, 2019-09-18:
yes I have access

pi@zigbee2mqtt:~ $ ls -la /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Sep 18 17:19 /dev/ttyACM0

jossie67, 2019-09-18:
I don't know how but it's fixed

I did :
DEBUG=* npm start

lolorc, 2019-09-18:
ok :-) i guess you can close it then :)

jossie67, 2019-09-18:
close

closed 2019-09-18 by jossie67