Koenkk/zigbee2mqtt #1994: Trying to add a new device -> problem detected, please create an issue.

wisekki, 2019-09-18:
I'm trying to add a new device from Legrand Netatmos -series. It's a light switch. It joins the network and I've added configuration to the devices.js and some basic console.log to fromZigbee.js. When I press the button from the switch I get "PROBLEM DETECTED, please create an issue." Here's the complete log from that: https://pastebin.com/fXfiJr9e

Could it be that these Legrand Netatmos devices are Zigbee3.0?
I'm using CC2530 with Z-Stack_Home_1.2

Koenkk, 2019-09-20:
Can you provide the complete log? Also with what happens after the error.

wisekki, 2019-09-20:
I'm not sure if this is what you're looking for but here it goes: https://pastebin.com/haZLqjtx
After the error it doesn't register the clicks from the remote switch.

Koenkk, 2019-09-22:
Can you provide an url to the product page? What kind of device is this, does it just have 1 single push button?

wisekki, 2019-09-22:
I'm sorry I don't have an url to provide. This is a rather new product series from Legrand. This particular device is a switch with dimmer option. On the back it says: Legrand AG08761. On the package it says Valena Life/Allure with Netatmo. This kit also includes power sockets, roller shutter switches, lighting micromodules. Power sockets already work with a bit of modification so I can control them on and off and I suspect micromodules have the same working principle.

Koenkk, 2019-09-23:
- Please move to the dev branch (adding support for this device will be easier there; https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html).
- Add the following configure section for your device: https://github.com/Koenkk/zigbee-shepherd-converters/blob/zigbee-herdsman-converters/devices.js#L898
- Check if it now reports on/off when pressing the switch.

wisekki, 2019-09-24:
I  moved my testing platform to dev branch and cleared out everything. Then I added the section like you suggested to the device. It seems to work a bit better. The logs will tell you more than me :) https://pastebin.com/FTWTR0Sd

Also now when I press on or off I get this which suggests it works:
`zigbee2mqtt:debug 9/24/2019, 5:59:38 PM Device '0x0004740000991173' announced itself
  zigbee2mqtt:debug 9/24/2019, 5:59:39 PM Received Zigbee message from '0x0004740000991173' of type 'commandOn' with data '{}' from endpoint 1 with groupID 0
  zigbee2mqtt:warn 9/24/2019, 5:59:39 PM No converter available for 'LegrandSwitch1' with cluster 'genOnOff' and type 'commandOn' and data '{}'
  zigbee2mqtt:warn 9/24/2019, 5:59:39 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
`

Koenkk, 2019-09-25:
If you use the following `fromZigbee` converters it will probably work: `fromZigbee: [fz.genOnOff_cmdOn, fz.genOnOff_cmdOff]`.

Can you also add `genLevelCtrl` to the bind (like: https://github.com/Koenkk/zigbee-shepherd-converters/blob/zigbee-herdsman-converters/devices.js#L2753). Do you also get events when rotating the dimmer switch now?

wisekki, 2019-09-29:
Sorry for the delayed answer and thank you for your help! The remote light switch works now as expected. Still not reporting battery levels for some reason.

Rotating dimmer switch is not working. It joins the network, but it just keeps repeating the message when I press the buttons:
`zigbee2mqtt:debug 9/29/2019, 12:17:10 PM Device '0x00047400007b162e' announced itself
  zigbee2mqtt:debug 9/29/2019, 12:19:58 PM Device '0x00047400007b162e' announced itself`

Maybe zigbee2mqtt is not registering commands after the announce... This is the join log: https://pastebin.com/gL6X4kmb

romainalca, 2019-09-30:
Is this device ? https://www.legrand.fr/pro/catalogue/42559-version-celiane-with-netatmo/commande-sans-fil-celiane-with-netatmo-pour-eclairage-ou-prise-connectee-ou-micromodule-titane
Or this device ? https://www.legrand.fr/pro/catalogue/42566-version-celiane-with-netatmo/interrupteur-filaire-connecte-celiane-with-netatmo-pour-volet-roulant-titane

I am very interested and i found your question here. All Celiane Netatmo solutions with zigbee2mqtt would be perfect !

Sorry for my (very) poor english !

wisekki, 2019-09-30:
Yes it seems to be the same. I have the mosaic versions, but that's just
the looks of the devices. Check out this
https://www.netatmo.com/en-us/partners/legrand ..you can see all the
"looks" :)

Koenkk, 2019-10-02:
Lets first complete the switch integration afterwards the dimmer; what do you currently have in `devices.js` for the switch?

closed 2019-12-08 by stale[bot]