Koenkk/zigbee2mqtt #1995: Gledopto bulbs ALSO don't handle switch from CCT to RGB correctly

wixoff, 2019-09-18:
I'm seeing a phenomenon that appears to closely reflect issue #1731 - except in bulbs, not strip controllers.

I have several Gledopto GL-B-008Z bulbs (12W RGB+CCT) and one GL-B-001Z bulb (4W RGB+CCT candelabra).  All of the bulbs seem to experience what's described in #1731 .

The bulbs work OK when switching from OFF to ON - if a color temperature is specified, it will be used.  If an RGB (XY) color is specified, it will be used.

But if the light is ON in CCT mode (white), giving it an RGB color won't turn off the white LEDs.  The shade of white seems to change very slightly, suggesting to me that the white LEDs are staying on while the colors are turned on too.

If you turn the bulbs OFF from that state, the white fades out first, and you can then see the colors underneath before they finish fading too.

However, if you're in RGB mode and switch to CCT, it appears to work correctly.  You get the white you want.  And when you turn it off, you don't see any color.

I'm thinking a fix for #1731 might also fix these bulbs, if it is in fact the same problem.

For the record, I'm running zigbee2mqtt 1.6.0+latest-dev in Docker, and have Z-Stack 3.0 firmware on my CC2531 coordinator.  But I honestly think this is a bulb quirk and not a software issue.

bochelork, 2019-09-19:
Same with GL-S-007Z. Not with GL-FL-004TZ. Latest dev, 1.2 source routing.

Koenkk, 2019-09-20:
I will close it as it is exactly the same problem as #1731 (a fix for that issue will also fix this).

closed 2019-09-20 by Koenkk

wixoff, 2020-06-11:
Also, the problem exists with my GL-B-001Z bulb (4W RGB+CCT candelabra) too, and as expected the quickie patch didn't fix it.

andreypolyak, 2020-06-11:
Hi @Koenkk,
I found that this fix works with individual light but doesn't work if two bulbs (both are GL-B-008Z) are in a group (created in zigbee2mqtt). When I set a color for a group of lights that are in CCT mode, it will not be applied.
Do you think it can be fixed?
Thanks!

Koenkk, 2020-06-11:
@wixoff is it fixed when using: `if ((meta.mapped.model === 'GL-C-007-1ID' || meta.mapped.model === 'GL-B-001Z' || meta.mapped.model === 'GL-B-008Z' ) && key === 'color' && !meta.message.transition) {` 

@andreypolyak for groups this cannot be solved from zigbee2mqtt, as there the default converters are used which do not have this fix. As a workaround setting a `transition` for your group should fix it. This can be done by setting the following in `configuration.yaml`.

```yaml
groups:
  '1':
    friendly_name: group_1
    retain: false
    transition: 0.4
```

wixoff, 2020-06-15:
@Koenkk yes, the GL-B-001Z works with that change.

Koenkk, 2020-06-15:
Thanks, merge the fix.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

wixoff, 2020-06-16:
Awesome, thanks.