Koenkk/zigbee2mqtt #1996: Cannot read property 'modelID' of null with dev at #e9a336f

sjorge, 2019-09-18:
```
  zigbee2mqtt:error 9/18/2019, 8:23:52 PM Failed to call 'Groups' 'onZigbeeStarted' (TypeError: Cannot read property 'modelID' of null
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:154:82))
```

I hit these with the current dev branch (#e9a336f)

sjorge, 2019-09-18:
Looks like it also happens on set state!
```
  zigbee2mqtt:error 9/18/2019, 8:24:56 PM Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'modelID' of null
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:154:82))
```

sjorge, 2019-09-18:
Interesting, looks my database.db got wiped whant I pulled in the changes just now :? Will repair and close

closed 2019-09-18 by sjorge

sjorge, 2019-09-18:
OK, that didn't seem to do much... I removed everything, after the first device I pair... i get teh same error again :/

Will look at it more tomorrow and unplug the stick for a bit

juanjux, 2019-09-19:
Reading my logs I've noticed I've almost the same entry with the -dev version that uses herdsman v0.6.11:

```
9/18/2019, 11:09:35 AM - error: Failed to call 'Groups' 'onZigbeeStarted' (TypeError: Cannot read property 'modelID' of null at Zigbee.resolveEntity (/app/lib/zigbee.js:154:82)) 
```

But in my case, everything works (except for a couple Mijia door sensors that like to frequently lose connection, but also happened with 1.6.0 and 1.5.2).

tb-killa, 2019-09-19:
See this on different devices too.
Konke and bitron devices doesn't seems to work directly. 
See in log that the device is or power source comes as value but doesn't write to database on this time.

sjorge, 2019-09-19:
In got most of these from Aqara sensors, so that does sound plausible