Koenkk/zigbee2mqtt #1997: UnhandledPromiseRejectionWarning Device already exists

sjorge, 2019-09-19:
# Bug Report
## What happened
Some how my database got corrupt and I need to repair everything, but this does not seem to work. I hit whenever Itry to add a device:

```
(node:55784) UnhandledPromiseRejectionWarning: Error: Device '0x000d6ffffe8e8d4f' already exists
    at Object.addDevice (/opt/zigbee2mqtt/lib/util/settings.js:262:15)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:167:40)
(node:55784) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:55784) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

Weird that it is a deprecation warning though, so I don;t think this impacts my repairing of all the devices?

## What did you expect to happen
Not get a DeprecationWarning (and the device to be able to repair)

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.6.0 (commit #e9a336f)
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}`

sjorge, 2019-09-19:
Not sure what broke when I update last night but it looks pretty bad, I can also no longer forcefully remove devices it seems.

```
  zigbee2mqtt:debug 9/19/2019, 6:05:37 AM Received MQTT message on 'zigbee2mqtt/bridge/config/remove' with data '0x00158d0004148f89'
  zigbee2mqtt:info 9/19/2019, 6:05:37 AM Removing '0x00158d0004148f89'
  zigbee2mqtt:error 9/19/2019, 6:05:47 AM Failed to remove 0x00158d0004148f89 (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)
```

Which might be the problem why I cannot repair, about half of them show up again after enabling pairing but are obviously incomplete, I also changed the panID and tried again starting fresh but that had no effect.

Before it would forcefully remove it after a few more seconds of no response from the device.

Koenkk, 2019-09-19:
I just also got this error, need to investigate why (can't reproduce it again).

coolguymatt, 2019-09-22:
I got the error today while trying to add a device.

Koenkk, 2019-09-23:
Found and fixed the issue (was a race condition).

closed 2019-09-23 by Koenkk

dzungpv, 2019-11-29:
@Koenkk  I have this issue on 1.7.1:

```
(node:57) UnhandledPromiseRejectionWarning: Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:57) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:57) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```
So this must be still open still we can fix it

Koenkk, 2019-11-29:
@dzungpv this is expected, see https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove