Koenkk/zigbee2mqtt #1998: Adding Zemismart zigbee downlight device

jordy55, 2019-09-19:
## What happened
I have installed Zigbee2mqtt Hass.io Add-on and added the following to the configuration:
“zigbee_shepherd_devices”: true

I downloaded the following [file](https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L376) and put it under share/zigbee2mqtt/.
Then I put the following in the configuration of the file share/zigbee2mqtt/devices.js :
```
{
    zigbeeModel: ['NUET56-DL27LX1.1'], 
    model: 'LXZB-12A', 
    vendor: 'Zemismart', 
    description: 'RGB LED Downlight', 
    supports: 'RGB, Dimming',
    extend: gledopto.light_onoff_brightness_colortemp_colorxy
},
```
After this I tried to connect the lamp with home assistant but unfortunately it says the following:
`zigbee2mqtt:warn 9/18/2019, 9:20:46 PM Message without device!`

I flashed the stick again and then tried to connect the lamp and I get the following message:
`zigbee2mqtt:warn 9/18/2019, 9:30:33 PM Message without device!`

After this I emptied the database.db and flashed the stick again and then I get no message during the first connection.
After this I pressed the connect button again and I get the following message:

```
  zigbee2mqtt:info 9/18/2019, 9:36:10 PM Device incoming...
  zigbee2mqtt:info 9/18/2019, 9:36:10 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 9/18/2019, 9:36:55 PM Device incoming...
  zigbee2mqtt:info 9/18/2019, 9:36:55 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
```

I don’t know what else I could do. I also tried to set zigbee shepherd to false, but then it says it’s an unknown device.

## What did you expect to happen
I thought adding the device and then connecting with zigbee was enough

## How to reproduce it (minimal and precise)
Keep connecting gives the same message

## Debug Info
zigbee2mqtt version: latest (1.6.0)
CC253X firmware version: latest (20190608)

jordy55, 2019-09-19:
Some extra info,
I have read that you also have to empty the devices.yaml file.
I did this. In addition, I also emptied the database. After this I restarted zigbee2MQTT and tried to connect the lamp again. Hereby I get the following log:
```
'{"version":"1.6.0","commit":"unknown","coordinator":20190608,"log_level":"info","permit_join":true}'
  zigbee2mqtt:info 9/19/2019, 9:45:49 PM New device 'NUET56-DL27LX1.1' with address 0x00124b001c85f438 connected!
  zigbee2mqtt:info 9/19/2019, 9:45:49 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00124b001c85f438","meta":{"modelID":"NUET56-DL27LX1.1"}}'
  zigbee2mqtt:info 9/19/2019, 9:46:20 PM Device incoming...
  zigbee2mqtt:info 9/19/2019, 9:46:20 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 9/19/2019, 9:47:34 PM Successfully reenabled joining
  zigbee2mqtt:info 9/19/2019, 9:50:14 PM Successfully reenabled joining
```
But I don't get any data and it doesn't come into the MQTT discovery like the other sensors and actuators do

Koenkk, 2019-09-20:
If you want the device to be discovered it needs to be added to: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L630.

jordy55, 2019-09-21:
Okay! 
I have searched for where it has to be placed but I can't find it. I'm running zigbee2mqtt as addon in HassIO. 
Do I have create it or is it already somewhere?

Koenkk, 2019-09-21:
Can you try to control the device by sending to `zigbee2mqtt/DEVICEID/set` with payload `ON` and `OFF`. if this works I will take care of the further integration.

jordy55, 2019-09-21:
Yes that works!
That would be great, will this come in the form of an update or will I have to adjust my devices.js later?

Koenkk, 2019-09-22:
Added, device is supported in the latest dev branch now. (-edge for hassio).

closed 2019-09-22 by Koenkk