Koenkk/zigbee2mqtt #1999: version 1.6.0 not starting

marcgarciamarti, 2019-09-19:
# Bug Report
## What happened
After upgrading from 1.5.x to 1.6, I can't seem to be able to start zigbee2mqtt even if I press the reset button, unplug and replug the dongle or reboot the pi

## What did you expect to happen
upgrade to 1.6 to work effortlessly

## How to reproduce it (minimal and precise)
by simply starting zigbee2mqtt

## Debug Info
zigbee2mqtt version: 1.6
CC253X firmware version: one of the latest (I can't seem to be able to find it out

this is the output that I see

`[Info] Configuration file found in data path, but no backup file found in data path. Backing up existing configuration to /share/zigbee2mqtt/.configuration.yaml.bk
2019-09-19T22:26:28: PM2 log: Launching in no daemon mode
2019-09-19T22:26:29: PM2 log: App [npm:0] starting in -fork mode-
2019-09-19T22:26:29: PM2 log: App [npm:0] online
> zigbee2mqtt@1.6.0 start /zigbee2mqtt-1.6.0
> node index.js
  zigbee2mqtt:info 9/19/2019, 10:26:35 PM Logging to directory: '/share/zigbee2mqtt/log/2019-09-19.22-26-33'
  zigbee2mqtt:info 9/19/2019, 10:26:35 PM Starting zigbee2mqtt version 1.6.0 (commit #unknown)
  zigbee2mqtt:info 9/19/2019, 10:26:35 PM Starting zigbee-shepherd
  zigbee2mqtt:info 9/19/2019, 10:26:37 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: Cluster with key '64516' does not exist)
  zigbee2mqtt:info 9/19/2019, 10:27:37 PM Starting zigbee-shepherd
  zigbee2mqtt:error 9/19/2019, 10:27:38 PM Error while starting zigbee-shepherd! (Error: dev exists, unregister it first.)
  zigbee2mqtt:error 9/19/2019, 10:27:38 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 9/19/2019, 10:27:38 PM Failed to start
	{"message":"dev exists, unregister it first.","stack":"Error: dev exists, unregister it first.\n    at /zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/lib/shepherd.js:449:19\n    at Promise.apply (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/q/q.js:1185:26)\n    at Promise.promise.promiseDispatch (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/q/q.js:808:41)\n    at /zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/q/q.js:1411:14\n    at runSingle (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/q/q.js:137:13)\n    at flush (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/q/q.js:125:13)\n    at process._tickCallback (internal/process/next_tick.js:61:11)"}
  zigbee2mqtt:error 9/19/2019, 10:27:38 PM Exiting...
npm
 ERR!
 code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm 
ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-09-19T20_27_38_942Z-debug.log
2019-09-19T22:27:39: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-09-19T22:27:39: PM2 log: App [npm:0] starting in -fork mode-
2019-09-19T22:27:39: PM2 log: App [npm:0] online
> zigbee2mqtt@1.6.0 start /zigbee2mqtt-1.6.0
> node index.js
  zigbee2mqtt:info 9/19/2019, 10:27:44 PM Logging to directory: '/share/zigbee2mqtt/log/2019-09-19.22-27-43'
  zigbee2mqtt:info 9/19/2019, 10:27:44 PM Starting zigbee2mqtt version 1.6.0 (commit #unknown)
  zigbee2mqtt:info 9/19/2019, 10:27:44 PM Starting zigbee-shepherd
  zigbee2mqtt:info 9/19/2019, 10:27:46 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: Cluster with key '64516' does not exist)`

Koenkk, 2019-09-19:
Can you check with the dev branch? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

marcgarciamarti, 2019-09-19:
Sure. Could you point me to some guidance on what to do with folders and so on, or can I just disable normal version and copy the add-on configuration over to edge while leaving the folders untouched?

piet206, 2019-09-20:
Same issue with restart loop. Switched over to the dev branch. Working as of now.

Koenkk, 2019-09-20:
@marcgarciamarti nothing special needed, just switch to edge.

marcgarciamarti, 2019-09-20:
@piet206 Could you share the process you followed? Is it as simple as disabling the standard add-on and simply copying and pasting the configuration onto the dev version?

marcgarciamarti, 2019-09-20:
@koenkk will try it in a while

marcgarciamarti, 2019-09-20:
Copied the configuration of the stable release onto the dev version, restarted and everything is working back. Thanks!

Koenkk, 2019-09-20:
awesome!

closed 2019-09-20 by Koenkk