Koenkk/zigbee2mqtt #2000: Failed to call 'Groups' 'onZigbeeStarted'

way2many, 2019-09-19:
# Bug Report
## What happened
```
9/19/2019, 8:51:40 PM - error: Failed to call 'Groups' 'onZigbeeStarted' (TypeError: Cannot read property 'getEndpoint' of null
    at Function.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/group.js:60:41)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/group.js:5:58))
```

## Debug Info
zigbee2mqtt version: Docker 1.6.0 dev
CC253X firmware version: 20190619

Groups are working but with bugs:
- Old groups from 1.5.1 are working
- New groups in 1.6.0 are not working

Koenkk, 2019-09-20:
The error should be fixed now.

> New groups in 1.6.0 are not working
Can you provide more context? (preferably with logs).

way2many, 2019-09-21:
Hi Koen,

it helped but still getting this error:

```
9/21/2019, 7:06:15 AM - error: Failed to call 'Groups' 'onZigbeeStarted' (TypeError: Cannot read property 'modelID' of null
    at Zigbee.resolveEntity (/app/lib/zigbee.js:154:82))
```

closed 2019-11-27 by stale[bot]

VladDrakul1986, 2019-12-19:
Hello I get the same error message when add a group to my config.
The error message appears right after the start up of the docker container i use latest so should be 1.8.0 the names of the devices got changed but i copy the addresses from paired devices section

`error 2019-12-19 12:28:43: Failed to call 'Groups' 'onZigbeeStarted' (Error: Timeout - 62166 - 1 - 2 - 4 - 0 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))`

```yaml
groups:
  '1':
    friendly_name: WoZiThermos
    # Required: Retain messages (true/false) (default: false)
    retain: false
    # Optional: Default transition to be used when e.g. changing brightness (in seconds) (default: 0)
    transition: 2
    # Optional: Change group state when one of the devices in it changes state, see 'State changes' below (default: true)
    optimistic: true
    # Optional: Devices of this group,
    # Note: this has to be the ieeeAddr of the device, not the friendly_name! (default: empty)
    devices:
      - '0xfffffffffffffffffffffffffffffff'
      - '0xeeeeeeeeeeeeeeee'
```

spider7611, 2020-01-09:
Ihave the same problem withikea bulb and switch. What can I do with this?

`error 2020-01-09 14:33:49: Failed to call 'Groups' 'onZigbeeStarted' (Error: Data request failed with error: 'MAC transaction expired' (240)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:578:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:5:58))`

docbobo, 2020-01-12:
Same here:

> zigbee2mqtt:error 2020-01-12 12:01:08: Failed to call 'Groups' 'onZigbeeStarted' (Error: Timeout - 0 - 1 - 2 - 4 - 0 after 10000ms
>      at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
>      at ontimeout (timers.js:436:11)
>      at tryOnTimeout (timers.js:300:5)
>      at listOnTimeout (timers.js:263:5)
>      at Timer.processTimers (timers.js:223:10))

evgrs, 2020-02-13:
I have the same trouble:

log.txt
```
error 2020-02-13 11:30:42: Failed to call 'Groups' 'onZigbeeStarted' (Error: Command 0x00158d000355937e/1 genGroups.add({"groupid":1,"groupname":""}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 18425 - 1 - 2 - 4 - 0 after 10000ms)
    at Endpoint.<anonymous> (C:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\model\endpoint.js:347:23)
    at Generator.throw (<anonymous>)
    at rejected (C:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\model\endpoint.js:6:65))
```

groups.yaml
```yaml
'1':
  friendly_name: mk_zigbee_switches
  retain: false
  optimistic: true
  devices:
    - '0x00158d000355937e'
    - '0x00158d00034cfc18'
    - '0x00158d0003469a86'
```

kalsan, 2020-03-07:
Same issue here with latest-dev docker image:

```
zigbee2mqtt_1    | zigbee2mqtt:error 2020-03-07 10:17:51: Failed to call 'Groups' 'onZigbeeStarted' (Error: Command 0x14b457fffe4dc869/1 genGroups.add({"groupid":2,"groupname":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Data request failed with error: 'No network route' (205))
zigbee2mqtt_1    |     at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:339:23)
zigbee2mqtt_1    |     at Generator.throw (<anonymous>)
zigbee2mqtt_1    |     at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

jdtsmith, 2020-03-19:
I also have group add `205` errors, for devices that formerly I was able to add together into a group (had to reset/rejoin them):

```
Failed to call 'Groups' 'onMQTTMessage' (Error: Command 0xd0<snip>/1 genGroups.add({"groupid":0,"groupname":""}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Data request failed with error: 'No network route' (205))
```

Why would you need a network route to a group?

Koenkk, 2020-03-19:
@jdtsmith the add to group command is send to a device, not to a group. Read more about how zigbee groups work here: https://www.zigbee2mqtt.io/information/groups.html#how-do-groups-work

jdtsmith, 2020-03-20:
OK, understood.  I had partially updated dev branch without full `npm ci`, and it was dropping route on any group command.  I've corrected that, and now route is preserved for my Fyrtur blinds, but binding the up/down remote to group no longer works at all :(.  Continuing on #2496.