Koenkk/zigbee2mqtt #2001: New KS-SM001 - Ksentry Electronics not pairing in 1.6.0

way2many, 2019-09-19:
# Bug Report
## What happened

Tried to add a new KS-SM001 - Ksentry Electronics device and I'm getting a interview_failed message.

## Debug Info
zigbee2mqtt version: Docker 1.6.0 dev
CC253X firmware version: 20190619

```
9/19/2019, 7:57:05 PM - info: 0x01124b001bae7942 (0x01124b001bae7942): Not supported (undefined)
9/19/2019, 7:58:44 PM - info: Device '0x01124b001bae7942' joined
9/19/2019, 7:58:44 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x01124b001bae7942"}}'
9/19/2019, 7:58:44 PM - info: Starting interview of '0x01124b001bae7942'
9/19/2019, 7:58:44 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x01124b001bae7942"}}'
9/19/2019, 7:58:44 PM - debug: Device '0x01124b001bae7942' announced itself
9/19/2019, 7:59:04 PM - error: Failed to interview '0x01124b001bae7942', device has not succesfully been paired
9/19/2019, 7:59:04 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x01124b001bae7942"}}'
```

Koenkk, 2019-09-20:
Can you provide the log when pairing and running with `DEBUG=*`? (set `DEBUG=*` as the environment variable).

way2many, 2019-09-21:
Unfortunately this is all I have. I'm running docker with -e DEBUG=*.
I bought two of those devices at the same time and decided to check the other one (should have done directly but work schedule yada yada yada) and it paired instantly. I opened both enclosures and the version on both circuit board is 2.0 so they should be identical. 

Faulty device you think?

Koenkk, 2019-09-21:
Dont put a dot after * in order to fix it I need the debug logging. 

Did you also try paring it again btw?

way2many, 2019-09-21:
There is no "." This is my docker command:

`docker run -d --name zigbee2mqtt-dev --network host -e PUID=1000 -e PGID=1000 -v /home/osmc/zigbee2mqtt-dev:/app/data --device=/dev/ttyACM0 -e DEBUG=* --restart unless-stopped koenkk/zigbee2mqtt:arm32v6-dev`

I actually managed to make it work by copying the data in the database from one of the other KS-SM001 and replacing the ieeeAddr and _id values with the "now-pairing" one. I actually recommended to do this to another user here a few months ago but forgot this "trick".

Do you think this "trick" might be useful and needs to be added in the documentation?

Koenkk, 2019-09-22:
Nice that it works for you know, I will try to get my hands on this device and check if I can fix the issue (so that the trick isn't necessary).

closed 2019-09-24 by way2many