Koenkk/zigbee2mqtt #2002: 1.6.0 : Not starting

MastaaK, 2019-09-20:
# Bug Report
## What happened
Z2M won't start with the latest version, either with mater or dev branch.
## What did you expect to happen
Z2M to start.
## How to reproduce it (minimal and precise)
Start Z2M (docker image) with a CC2531 flashed with the last firmware
## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: [latest fw](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip). Could it be a probleme with the flashing ? 
## LOG
Here is what I got from docker with master: https://pastebin.com/VBrXhMbj
Here is what I got from docker with master:  https://pastebin.com/bNtRNdwc

Koenkk, 2019-09-20:
Did you start zigbee2mqtt before with this stick?

MastaaK, 2019-09-21:
Yes it worked with 1.5.0 I think

Just flashed the stick with CC2531ZNP-with-SBL.hex and still the same errors.

MastaaK, 2019-09-22:
Okay,

Error was on my VM side.

Just start from a fresh install install and all good.

Thanks

closed 2019-09-22 by MastaaK