Koenkk/zigbee2mqtt #2004: Removed hue lamp doesn't stay removed

hvxl, 2019-09-20:
# Bug Report
## What happened
I removed a hue lamp. Within a few seconds it is automatically reconnected again.
## What did you expect to happen
Once removed, a device should stay removed until explicitly rejoining it again.
## How to reproduce it (minimal and precise)
Issue a zigbee2mqtt/bridge/config/remove message for a hue lamp. Then observe the log and see that it is successfully removed, but then automatically connected again.
Note: This is with permit_join set to false.
I don't know if this problem is limited to hue lamps, but that's all the zigbee devices I have at the moment.
## Debug Info
zigbee2mqtt version:1.6.0
CC253X firmware version:20190425

Debug log showing the issue: [log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3635529/log.txt)

tb-killa, 2019-09-22:
Hi this happened because the end-device as self ones doesn't reset their connectivity states. The result of this would be that Z2M removed them from their side but if you wouldn't allow a join back the workaround would be to blacklist them.

gdave321, 2019-09-24:
Same issue here with an IKEA switch, device is added back within seconds after being removed:
[rem-add.txt](https://github.com/Koenkk/zigbee2mqtt/files/3646185/rem-add.txt)

Koenkk, 2019-09-25:
I cannot reproduce this on the dev branch. Can you try this?

As also mentioned in the OP, make sure that `permit_join: false`, otherwise I see that the Hue bulb immediately rejoins indeed.

gdave321, 2019-09-26:
@Koenkk   tried the dev branch, but could get it to startup, looks I hit issue #1956 
(and yes, permit_join is set to false)

Koenkk, 2019-09-27:
Multiple issues are mentioned there. In case you have the `Error: Cannot find module 'zigbee-herdsman/dist/deprecated/ziee'` issue, see https://github.com/Koenkk/zigbee2mqtt/issues/1892#issuecomment-527837064.

gdave321, 2019-09-28:
Ok, got the dev branch to run, not entirely sure what was wrong but re-installed the modules and got it to successfully run.
However, had some difficulties to get a light to join in order to test the subsequent remove.
Noticed a lot of device reporting issues and timeouts, so blame that for these joining troubles.  As I needed my lights again I went back to release branch.  Will re-test once the reporting issues are settled.

 error 2019-09-28T09:28:57: Failed to setup reporting for '0x0017880110453b2e' - Error: Timeout - 26339 - 11 - 13 - 7 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)

hvxl, 2019-11-29:
Considering the troubles reported by gdave123, I was a bit hesitant to try the dev branch. But I have now tried with zigbee2mqtt version 1.7.1 and can confirm that a removed device no longer immediately rejoins. (I do also see the "Failed to setup reporting" errors.)

closed 2020-02-04 by stale[bot]