Koenkk/zigbee2mqtt #2005: "Failed to mount the cieApp" in 1.6

Erelen-Laiquendi, 2019-09-20:
After upgrading to 1.6, I see "Failed to mount the cieApp" in logs just after start (in ~8 seconds after "'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"unknown","coordinator":20190619,"log_level":"info","permit_join":false}'".

And then, about 10 minutes of "Failed to bind for reporting" and "Failed to setup reporting". In this period, turtitg on/off lights doesn't work, or work very slow (about minute lag). After ~10 minutes everything works ok.

Before upgrading there was no any error, and everything works just after start on zigbee2mqtt.

Dongle: GBAN GB2530
Firmware: 20190619 source_routing
Network: 10 routers, 30 endpoints

Koenkk, 2019-09-20:
Can you try with zigbee2mqtt 1.6 dev? (edge for hassio, -dev for docker users or bare metal: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Erelen-Laiquendi, 2019-09-21:
Trying to start "edge" for 3 times.

Every start I see this just after block of `MQTT publish: topic 'zigbee2mqtt/<every my device>/availability', payload 'online'`:
```
(node:38) UnhandledPromiseRejectionWarning: Error: SRSP - UTIL - ledControl after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:38) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:38) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

Only first start of "edge" there was more errors at start:
```
Failed to setup reporting for '0x000b57fffe494981' - TypeError: Cannot read property 'get' of undefined
Failed to setup reporting for '0x000d6ffffef3158a' - TypeError: Cannot read property 'get' of undefined
Failed to setup reporting for '0x000d6ffffef33241' - TypeError: Cannot read property 'get' of undefined
Failed to setup reporting for '0x000d6ffffef818e7' - TypeError: Cannot read property 'get' of undefined
Failed to setup reporting for '0x01124b001bae7a79' - Error: Data request failed with error: 'MAC no ack' (233)
```
First 4 ids - it's IKEA LED1623G12. This 4 lamps - it's one zigbee group.
Last id - KS-SM001.

And now everything works (I can lamps on/off) right after start.

Koenkk, 2019-09-22:
Good to see that everything is working now, I've added extra logging in the latest dev branch. Do you still see errors with that?

Erelen-Laiquendi, 2019-09-26:
Just now (26.09.2019, ~0:00 UTC+0) reinstalled Hassio addon zigbee2mqtt-edge.

New version require retain parameter fro every device & gorup - ok. But after this fix, behavior was the same: just after block of MQTT publish: topic 'zigbee2mqtt/<every my device>/availability', payload 'online' and first message from device:
```
zigbee2mqtt:info  2019-09-26T00:00:17: MQTT publish: topic 'zigbee2mqtt/Room Interior Motion', payload '{"battery":74,"linkquality":72,"occupancy":true,"last_seen":"2019-09-26T00:00:17.304Z"}'
(node:38) UnhandledPromiseRejectionWarning: Error: SRSP - UTIL - ledControl after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:38) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:38) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

Koenkk, 2019-09-27:
I'm not sure what is going on, could you remove the `disable_led` from your `configuration.yaml`?

Erelen-Laiquendi, 2019-09-27:
Yes, that's it! There is no error after removing `disable_led`.
And `disable_led` has no meaning with my GBAN dongle.

On PC (Win10, Python virtualenv) with CC2531 - everything ok with any state of `disable_led`.

closed 2019-09-27 by Erelen-Laiquendi